openwrt/openwrt #35: ag71xx_main.c : netdev : tx timeout and crash eth0 when WNR2200 boot with cable plugged in

carl-, 2016-07-16:
https://github.com/openwrt/openwrt/blob/master/target/linux/ar71xx/files/drivers/net/ethernet/atheros/ag71xx/ag71xx_main.c#L466
change this line back to CC (15.05.1)version would solve this problem

psyborg55, 2016-10-03:
i've discovered same issue on AR9531 target. DHCP did not assign an address. booting without cable and attaching it later works as expected. kernel log:

>     [    0.000000] Linux version 4.4.14 (ubuntu@ubuntu) (gcc version 5.3.0 (OpenWrt GCC 5.3.0 49934) ) #17 Mon Sep 26 21:11:33 UTC 2016
    [    0.000000] MyLoader: sysp=eb171b51, boardp=48c6c569, parts=8bb656e2
    [    0.000000] bootconsole [early0] enabled
    [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
    [    0.000000] SoC: Qualcomm Atheros QCA9533 ver 1 rev 1
    [    0.000000] Determined physical RAM map:
    [    0.000000]  memory: 02000000 @ 00000000 (usable)
    [    0.000000] Initrd not found or empty - disabling initrd
    [    0.000000] No valid device tree found, continuing without
    [    0.000000] Zone ranges:
    [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000001ffffff]
    [    0.000000] Movable zone start for each node
    [    0.000000] Early memory node ranges
    [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000001ffffff]
    [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000001ffffff]
    [    0.000000] On node 0 totalpages: 8192
    [    0.000000] free_area_init_node: node 0, pgdat 803bd710, node_mem_map 81000000
    [    0.000000]   Normal zone: 64 pages used for memmap
    [    0.000000]   Normal zone: 0 pages reserved
    [    0.000000]   Normal zone: 8192 pages, LIFO batch:0
    [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
    [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
    [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
    [    0.000000] pcpu-alloc: [0] 0 
    [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
    [    0.000000] Kernel command line:  board=TL-MR22U  console=ttyATH0,115200 rootfstype=squashfs,jffs2 noinitrd
    [    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
    [    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
    [    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
    [    0.000000] Writing ErrCtl register=00000000
    [    0.000000] Readback ErrCtl register=00000000
    [    0.000000] Memory: 28004K/32768K available (2703K kernel code, 141K rwdata, 712K rodata, 276K init, 193K bss, 4764K reserved, 0K cma-reserved)
    [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
    [    0.000000] NR_IRQS:51
    [    0.000000] Clocks: CPU:550.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:25.000MHz
    [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6950037990 ns
    [    0.000010] sched_clock: 32 bits at 275MHz, resolution 3ns, wraps every 7809031678ns
    [    0.008229] Calibrating delay loop... 366.18 BogoMIPS (lpj=1830912)
    [    0.071059] pid_max: default: 32768 minimum: 301
    [    0.076072] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
    [    0.083053] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
    [    0.093501] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
    [    0.104767] NET: Registered protocol family 16
    [    0.111051] MIPS: machine is TP-LINK TL-MR22U v1
    [    0.348014] clocksource: Switched to clocksource MIPS
    [    0.354841] NET: Registered protocol family 2
    [    0.360435] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
    [    0.367804] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
    [    0.374575] TCP: Hash tables configured (established 1024 bind 1024)
    [    0.381409] UDP hash table entries: 256 (order: 0, 4096 bytes)
    [    0.387584] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
    [    0.394550] NET: Registered protocol family 1
    [    0.399256] PCI: CLS 0 bytes, default 32
    [    0.400369] futex hash table entries: 256 (order: -1, 3072 bytes)
    [    0.424855] squashfs: version 4.0 (2009/01/31) Phillip Lougher
    [    0.431087] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
    [    0.444312] io scheduler noop registered
    [    0.448532] io scheduler deadline registered (default)
    [    0.454198] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
    [    0.481595] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 1562500) is a 16550A
    [    0.494091] m25p80 spi0.0: found s25fl064k, expected m25p80
    [    0.500053] m25p80 spi0.0: s25fl064k (8192 Kbytes)
    [    0.506347] 5 tp-link partitions found on MTD device spi0.0
    [    0.512282] Creating 5 MTD partitions on "spi0.0":
    [    0.517336] 0x000000000000-0x000000020000 : "u-boot"
    [    0.524265] 0x000000020000-0x00000014d484 : "kernel"
    [    0.531408] 0x00000014d484-0x0000007f0000 : "rootfs"
    [    0.538494] mtd: device 2 (rootfs) set to be root filesystem
[... 79 lines cut ...]
    [   46.058519] [<80081014>] warn_slowpath_fmt+0x2c/0x38
    [   46.058536] [<8027fb68>] dev_watchdog+0x1dc/0x260
    [   46.058566] [<800af2d8>] call_timer_fn.isra.5+0x24/0x80
    [   46.058585] [<800af554>] run_timer_softirq+0x1ac/0x1f4
    [   46.058608] [<80083414>] __do_softirq+0x164/0x290
    [   46.058644] [<8006a008>] plat_irq_dispatch+0xbc/0xf4
    [   46.058654] 
    [   46.058666] ---[ end trace 8d90bdf73e155ec0 ]---
    [   46.058678] eth0: tx timeout
[... cut ...]
    [  128.471712] br-lan: port 1(eth0) entered forwarding state
    [  130.468025] br-lan: port 1(eth0) entered forwarding state
    [  162.470103] eth0: link down
    [  162.470218] br-lan: port 1(eth0) entered disabled state
    [  163.471547] eth0: link up (100Mbps/Full duplex)
    [  163.471628] br-lan: port 1(eth0) entered forwarding state
    [  163.471705] br-lan: port 1(eth0) entered forwarding state
    [  165.468026] br-lan: port 1(eth0) entered forwarding state
    [  194.058026] eth0: tx timeout
    [  209.058045] eth0: tx timeout

psyborg55, 2016-10-15:
replacing tplink u-boot with self-compiled one from pepe2k sources makes eth working fine again.

closed 2022-02-09 by aparcar