openwrt/openwrt #45: No image files are generated for the BCM27xx targets

rmanders, 2016-07-28:
**Git commit**: `558b38f4b8e615fb5262c94dad7db2071e394ac1`
**Target**: `Broadcom BCM27xx`
**Sub Target**: `BCM2710 based boards` or `BCM2709 based boards`

When building OpenWRT from trunk, there appears to be no OS image output in the bin/brcm2708/ directory. The only contents are: "md5sums  packages  sha256sums" as opposed to the expected "openwrt-brcm2708-bcm2710-rpi-3-ext4-sdcard.img  openwrt-brcm2708-bcm2710-rpi-3-ext4-sdcard.img.gz" files

syoder, 2016-07-29:
I'm also having this issue.  I suspect it's due to some reorganization that was done in this file: https://github.com/openwrt/openwrt/blob/master/target/linux/brcm2708/image/Makefile

Currently, you can select the target (BRCM27xx) and the subtarget (BRCM2710, for example), but when you go to select the device profile, it only shows "Default".  I suspect it's not possible to select an actual target device because the Makefile is messed up, and the build system doesn't know what "Default" is, so it's not building the image.

I'm trying to figure out how to fix it, but haven't had any luck so far.  I have verified that it works if you build from here: `8b0324262436e17beada3ea3fece85aedd46a35a`

devran01, 2016-08-08:
@syoder @Noltari 
It seems like the commit `a194e83f16b05240184f58b709152fb840cf00f7`is causing this issue. This commit removed the profile files for raspberry pi boards, because of this the build system only shows "Default" in device profile. I can confirm that image for BCM27XX targets are generated with previous commit `e95f2e2f7217340cff5f8cc37fbd69ffba367c80`.

Noltari, 2016-08-08:
@rmanders @syoder @devran01 
This is the result of several untested backports from LEDE to OpenWrt.
Please consider using LEDE instead.

devran01, 2016-08-09:
@Noltari 
As a temporary solution, I've created a [PR](https://github.com/openwrt/openwrt/pull/51) reverting the changes done in `a194e83f16b05240184f58b709152fb840cf00f7`. With this PR, OpenWrt build system generates image for Raspberry Pi based boards.

Seems like LEDE is being actively developed. What will happen to OpenWrt? Will it still be developed/supported in future?

wigyori, 2016-08-11:
@Noltari - thanks for the comments, this has always proven to be helpful.

@devran01 - I've merged the commit rollback, thanks for working this out. On the "what will happen" question, please refer to https://lists.openwrt.org/pipermail/openwrt-devel/2016-June/041880.html . Above that, for the time being, we're working to maintain a healthy trunk until things are worked out between the two groups.

closed 2016-08-11 by wigyori

henrickveloso, 2017-01-02:
Hello! There is something working about OpenWRT for Raspberry Pi 3 (bcm2720)? I can´t find images or packages on repository. Thanks.