openwrt/openwrt #49: Backport mwlwifi fixes to Chaos Calmer

uzytkownik, 2016-08-01:
There is known issue with mwlwifi (https://github.com/kaloz/mwlwifi/issues/63) which causes it to reduce performance with time. It would be great if it was backported to Chaos Calmer if  OpenWRT 16 is not released soon.

thagabe, 2016-08-01:
Patch already merged

thagabe, 2016-08-01:
@wigyori I agree the patch (which was updating to the lastest upstream) should be backported to CC as most people will want to stick with the stable rather than trunk. As for me I am willing to bet the update will not break anything with CC and should yield great benefit to the users who are wishing to build CC. I will launch a new build with the patch in place and will report back if the build was successful.

thagabe, 2016-08-02:
Just built with new driver: works
Flashed: works
tested wifi: works :100: 

I'm confident we should go ahead and push patch to CC :+1:

wongsyrone, 2016-08-08:
In general, only security and severe bug fixes should be backported, and it is not advice to add new features to stable branches.

uzytkownik, 2016-08-08:
Well, not having to restart router a few times per day would be closer to
severe bug fix then new feature, I'd say.

wongsyrone, 2016-08-08:
Well, what I said above is just best practice. All depends on developers.

uzytkownik, 2016-08-08:
Yeah. Though if performance bug makes WiFi connections from clients time
out I think it might fall under said best practices. Without fix user needs
to reboot router to even connect to WiFi.

thagabe, 2016-08-08:
Agreed. The whole wifi driver of that time was one huge bug. The re-written driver works better and has support for all kernels.

Again @wigyori please do commit the latest driver to CC

wigyori, 2016-08-19:
@thagabe - Kaloz pushed an update to the driver both to trunk and CC.

https://github.com/openwrt/openwrt/commit/d175c09bea24b0a52d15350f025ca418f5f066c5
https://dev.openwrt.org/changeset/49404

Thanks,
-w-

uzytkownik, 2016-08-21:
@wigyori Thanks. When would it be available for opkg update (sorry - I don't know OWRT dev workflow)?

wigyori, 2016-08-24:
@uzytkownik - not sure at this moment, sorry. Probably as part of a 15.05.2 bugfix release, but cannot confirm at this stage. Until then, only as manual builds.

closed 2022-02-09 by aparcar