openwrt/openwrt #50: Missing commits from git.openwrt.org / different recent history?

nwf, 2016-08-08:
I am not sure how this repository was generated, but a little bit of shell hacking, specifically,

    (paste <(git rev-list --reverse oldrepo/master) <(git rev-list --reverse github/master)) \
      | while read old new; do echo -n $old $new; git diff --quiet $old $new; echo " $?"; done

with .git/config containing

    [remote "github"]
    	url = git@github.com:openwrt/openwrt.git
    	fetch = +refs/heads/*:refs/remotes/github/*
    [remote "oldrepo"]
    	url = git://git.openwrt.org/openwrt.git
    	fetch = +refs/heads/*:refs/remotes/oldrepo/*

reveals that the two trees are identical modulo commit messages up through and including 79d21a4158421bf27b6d4cf9bf1bb55a528b0b74 in the old repository, respectively 88d875f604b62c3db36d6c3ab6dc69349f808501 in this repository.

The two repositories then disagree about the successor commits, with the old repository claiming f06425ef611085a791bcd0399c9b1e1a27525eca and this one claiming 3907a8c0b28ba7fce5e72919cb49bd2cb4c408b9.  Several of the commits in the old tree are apparently (looking solely at commit messages) replicated in the new tree under different identifiers in differing commit order.  It looks as though two commits in particular were fully superseded by another, but if not these are missing in the new history:

    f06425ef611085a791bcd0399c9b1e1a27525eca ncurses: install a dummy libtinfo.a for packages that try to link it
    9b4650b3b92e6246b986ac9e3d7c2a80d66b805b rpcd: iwinfo plugin fixes

I presume that this is the official new repository, as per the wiki, and that the above is just a mistake?

hnyman, 2016-08-22:
I updated my Openwrt trunk build system to use the Github server, fetched sources, tried to build Openwrt and noticed that htop fails to build from the Github sources. However, htop built just fine one hour earlier with previous sources from git.openwrt.org. Strange. Quick investigation led me to this issue.

Discrepancy between git.openwrt.org and github sources for ncurses can be easily noticed here:
https://github.com/openwrt/openwrt/commits/master/package/libs/ncurses
http://git.openwrt.org/?p=openwrt.git;a=history;f=package/libs/ncurses;hb=HEAD

One commit for ncurses is missing from Github:
``` ncurses: install a dummy libtinfo.a for packages that try to link it ```
http://git.openwrt.org/?p=openwrt.git;a=commit;h=f06425ef611085a791bcd0399c9b1e1a27525eca  

Manually replicating the missing commit fixes htop. Probably something else gets fixed, too.

The missing commit breaks htop also at the Openwrt buildbot:  http://buildbot.openwrt.org:8010/broken_packages/ar71xx/htop/compile.txt

(Note: the rpcd discrepancy also reported above has apparently been fixed by https://github.com/openwrt/openwrt/commit/cc48d5b3724237a8de31cb26b53bdb2f15125207 )

nwf, 2016-11-18:
While I'd love an explanation for the issue, I suspect it's long enough in the past to no longer matter.  Should this issue be closed?

closed 2017-08-13 by nwf