openwrt/openwrt #74: OPENWRT ON RASPBERRY PI 3

thonguyendinh, 2016-08-25:
Dear all,
I have compiled openwrt for RP3.It work but cannot connect network
I build with options 
         Target system BCM27XX
         Subtarget  BCM27010
What is my fault?
Help me
Thank

scropion86, 2016-09-19:
@thonguyendinh hi did you have your problem solved as i have the same problem.
if yes can you shared the compiled file with me.

thanks

thonguyendinh, 2016-10-02:
Let's re-configure openwrt imge file follow its version for raspberry pi2
Reply later,hope to help you

dade80vr, 2017-01-11:
Hi.
Did you solve your problem? Same issue here.
Where is OpenWRT image right for bcm2710 ? I cannot found it here: https://wiki.openwrt.org/toh/raspberry_pi_foundation/raspberry_pi

thonguyendinh, 2017-01-12:
You should re-configure network in custom file img,default values is
not correct, before build for raspberry

Best regards