openwrt/openwrt #80: Hard-coded IPv4 addresses of ds-lite package...

tahiro86j, 2016-08-29:
I think the package _de-lite_ should be renamed to _ipip6_ and its hard-coded IPv4 addresses (192.0.0.1, 192.0.0.2) should be turned into user-defined options.

An alternative approach is to keep the name _ds-lite_ while adding _ipip6_ and _ds-lite_ now behaving like a wrapper.

I do have a working code (entirely based on the shell-script supplied by the _ds-lite_ package) so if folks are here to accept a pull-request, I'm ready to open one.

tahiro86j, 2016-09-07:
Some might ask why this is necessary, and/or if you are curious about the backgrounds, please continue reading this additional post.

I am a happy user of OpenWrt in East Japan, whose internet connectivity is widely based on the service called "Flets's Hikari Next" by NTT-East.

The infrastructure provided by the service is rather interesting, in that the service alone merely provides the access-line, hence the L1 and L2. Although it does provide L3 via IPv6, you are given no connectivity to the WAN/Internet. By default, it's a local-WAN or Wide-Area-LAN based on IPv6. 

Getting access to the web on this service is possible by either one of the following "commercial" options;
a.)IPv4 via PPPoE
b.)DS-Lite

Primary use of such system (I imagine) is to replace the copper-line based PSTN (both analog and ISDN) by VoIP and imaginably, the monthly fee of its "Lite" service is now getting as low as those copper-line based services. There is another "interesting" use of the service, however, and this will be the main focus in this post (will be mentioned soon).

Some interesting parts of  this "Lite" service are;
a.)They are cheaper than the regular service, priced around $30/mo (around JPY3,000/mo).
***For comparison, the regular service is priced around $50/mo (around JPY5,000/mo) with unlimited download/upload.
b.)Traffic in excess of 500MBytes/mo involving PPPoE will be charged for on the pay-as-you-go basis.
c-1.)Traffic via IPv6 with no involvement of PPPoE counts only 40MBytes/mo towards the monthly use.
c-2.)Subscribers are able to establish connections with other subscribers of both "Lite" and regular via IPv6, and DDNS mechanism is made available for easing (Prefixes are "semi-static", they explain). 

Now, that I knew about the service and my parents has a semi-empty house some 40km outside of central Tokyo (and we used to have an ADSL line there, effectively working at Down:1.3Mbps/Up:700kbps and this was too slow), I somehow wanted to get that house online at speeds exceeding 30Mbps Up/Down. The idea of getting the two houses linked up via IPv4/IPv6 tunnel was the way to go.

tahiro86j, 2016-09-14:
Since the issue seems to gather no attention, I'm silently making the modified script available under the repository [openwrt_v6tunnel](https://github.com/tahiro86j/openwrt_v6tunnel). As long as the rights of the script are claimed by any of the OpenWrt developers, I'll be more than happy to move the ownership of the script to the official repository of OpenWrt.
Cheers.

KanjiMonster, 2016-09-14:
Well, DS-Lite is more than just an ip4-in-ip6 tunnel, RFC6333 defines the endpoints as 192.0.0.1 and 192.0.0.2 (although with a SHOULD, not a MUST), and also has additional implications like NAT being done on the AFTR. It might make sense to allow user configuration of the AFTR/B4 addresses, but RFC6333 / RFC7335 limit it to 192.0.0.0/29. There does not seem to be any DHCP option to allow auto-configuration of these addresses.

So having a separate protocol is the way to go IMHO.

closed 2022-02-09 by aparcar