openwrt/openwrt #84: NCM with Huawei E3372

halmi, 2016-09-02:
I have been trying to get my E3372 to work in NCM-Mode in the current trunk.
This has been working quite well for a while now, when manually sending the AT^NDISDUP,1,1,"apn" to /dev/ttyUSB0, and setting the wwan0 to proto dhcp.
(provided that all the necessary packages, esp. kmod-usb-net-huawei-cdc-ncm are installed)

But using the builtin (i.e. setting the proto for wwan0 to wwan or ncm) tools, it did not work at all.

1) proto wwan
There the issue is, that the Stick has USB Vendor/ProductID [12d1:1506](https://github.com/openwrt/openwrt/blob/master/package/network/utils/wwan/files/data/12d1:1506) and is therefore recognized as something else, which defaults to using ppp instead of ncm

I don't know if the E367/E398 support NCM, but if yes then it would probably be a good idea to change that.

2) proto ncm
When forcing the protocol to ncm this all gets handled by [ncm.sh](https://github.com/openwrt/openwrt/blob/master/package/network/utils/comgt/files/ncm.sh) and [ncm.json](https://github.com/openwrt/openwrt/blob/master/package/network/utils/comgt/files/ncm.json)

And there are a few issues at hand:
ncm.sh uses [getcardinfo.gcom](https://github.com/openwrt/openwrt/blob/master/package/network/utils/comgt/files/getcardinfo.gcom) to select the device manufacturer, but my stick returns the line just "Manufacturer:" without anything which leads to an empty string. This then leads to all the sent commands being empty which does obviously not work.

When fixing the manufacturer string to "huawei" in ncm.sh, it basically works, but then the initialization fails at the AT+CGDCONT=1,... command, which the stick does not support.
From what my tests have shown, the stick does not need any initialization at all besides the AT^NDISDUP,1,1,...

In order to create a clean solution, I would recommend:
Somehow reworking the ncm.sh in order how to decide which commandset to use. Maybe a similar way as in [wwan.sh](https://github.com/openwrt/openwrt/blob/master/package/network/utils/wwan/files/wwan.sh) could be possible? (selection based on USB Vendor&DeviceID).

And as it is clear now that not all devices by one manufacturer need the same commands, a bit more differentiation will be needed in ncm.json.

I'm definitely willing to help to resolve this issue, but I'm not yet familiar enough with the codebase to come up with a solution all on my own.

adrianmihalko, 2017-10-17:
Any update on this? I can't use my Huawei modem with OpenWRT/LEDE.

The guide is useless:

https://wiki.openwrt.org/doc/recipes/ethernetoverusb_ncm

halmi, 2017-10-18:
I digged into this a bit back then copied a bit together a "new" protocol.
It is basically still just writing the AT^NDISDUP to a device file, but automates this a bit better.

As before, the packages `kmod-usb-net-huawei-cdc-ncm` and probably `usb-modeswitch` have to be installed.

The following files need to go under `/lib/netifd/proto`
https://gist.github.com/halmi/70ac29a31878f8675366459a3a4b9fe2
There are two variants, but I mainly used the `huawei_dhcp` one.

I then configured the interface as
```bash
config interface 'lte'
  option ifname 'wwan0'
  option proto 'huawei_dhcp'
  option device '/dev/cdc-wdm0'
  option apn 'drei.at'
```

Took me quite a while to figure all this out, but it has been running very nicely for over a year now.

adrianmihalko, 2017-10-18:
Hello Halmi,

I have kmod-usb-net-huawei-cdc-ncm installed, but I have no device /dev/cdc-wdm0.

Issuing command:

echo -e "AT^NDISDUP=1,1,\"internet\"\r" > /dev/ttyUSB0

...the LED on the modem constantly lit, which means it is connected to the network.

I have no idea why I don't have /dev/cdc-wdm0 device, do you have any idea?

[   16.049007] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB0
[   16.050224] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB1

lsusb:

Bus 002 Device 002: ID 12d1:1506 Huawei Technologies Co., Ltd. E398 LTE/UMTS/GSM Modem/Networkcard

usbmode -l is empty, interesting. But If I am not wrong it must works, because 12d1:1506 is a modified indentifier.

In rPi where I have this modem working I have modeswitch config:

pi@raspberrypi3:~ $ cat /etc/usb_modeswitch.d/12d1\:14fe 
DefaultVendor=0x12d1
DefaultProduct=0x14fe
# T-Mobile NL (Huawei E352)
TargetVendor=0x12d1
TargetProductList="1506,150f,151d"
HuaweiNewMode=1

adrianmihalko, 2017-10-18:
Edit:

Ok I reinstalled everything, including Lede on rPi. I installed kmod-usb-net-huawei-cdc-ncm as you suggested.

Now, /dev/cdc-wdm0 is recognized correctly.

```
[   14.316866] usb 1-1.4: new high-speed USB device number 5 using dwc_otg
[   14.438028] usb 1-1.4: New USB device found, idVendor=12d1, idProduct=1506
[   14.447211] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   14.456787] usb 1-1.4: Product: HUAWEI_MOBILE
[   14.463313] usb 1-1.4: Manufacturer: HUAWEI_MOBILE
[   14.645538] huawei_cdc_ncm 1-1.4:1.2: MAC-Address: 00:1e:10:1f:00:00
[   14.654281] huawei_cdc_ncm 1-1.4:1.2: setting rx_max = 16384
[   14.670922] huawei_cdc_ncm 1-1.4:1.2: NDP will be placed at end of frame for this device.
[   14.684033] huawei_cdc_ncm 1-1.4:1.2: cdc-wdm0: USB WDM device
[   14.693309] huawei_cdc_ncm 1-1.4:1.2 wwan0: register 'huawei_cdc_ncm' at usb-3f980000.usb-1.4, Huawei CDC NCM device, 00:1e:10:1f:00:00
[   14.710825] usb-storage 1-1.4:1.3: USB Mass Storage device detected
[   14.719755] scsi host1: usb-storage 1-1.4:1.3
[   14.727916] usb-storage 1-1.4:1.4: USB Mass Storage device detected
[   14.737258] scsi host2: usb-storage 1-1.4:1.4
[   15.732457] scsi 1:0:0:0: CD-ROM            HUAWEI   Mass Storage     2.31 PQ: 0 ANSI: 2
[   15.751262] scsi 2:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
[   15.766790] sd 2:0:0:0: [sda] Attached SCSI removable disk
```

Installed /lib/netifd/proto/huawei_dhcp.sh, chmod +x…

Created network interface in /etc/config/network as you wrote.

Result: nothing. If I do “ifup lte” I see the interface in ifconfig, but the modem is not connected (blue led on modem is blinking) + no dhcp address.

I tried manually connect:

```
#!/bin/sh
echo -e "AT^NDISDUP=1,1,\”internet\”\ > /dev/cdc-wdm0
udhcpc -i wwan0
```

…and this is working.

Do you have any idea what I am missing with your script? :)

nixnuex, 2017-11-11:
Tried the `huawei_dhcp` with my Huawei 3372 non-HiLink USB LTE Adapter and it worked out of the box! Thanks a lot! Started up automatically on boot, will see if it always stays online.

closed 2022-02-09 by aparcar