openwrt/openwrt #106: CC: squashfs4 fails to compile

wararjey, 2016-09-29:
makeing a clean build of the latest CC branch fails with the following error messages:

make[3]: Entering directory '/home/sabine/devel/openwrt/openwrt_git/tools/lzma'
make[3]: Leaving directory '/home/sabine/devel/openwrt/openwrt_git/tools/lzma'
make[3]: Entering directory '/home/sabine/devel/openwrt/openwrt_git/tools/lzma'
make[3]: Leaving directory '/home/sabine/devel/openwrt/openwrt_git/tools/lzma'
make[3]: Entering directory '/home/sabine/devel/openwrt/openwrt_git/tools/squashfs4'
make -C /home/sabine/devel/openwrt/openwrt_git/build_dir/host/squashfs4.2/squashfs-tools CC="ccache gcc" XZ_SUPPORT=1 LZMA_XZ_SUPPORT=1 XATTR_SUPPORT= LZMA_LIB="/home/sabine/devel/openwrt/openwrt_git/staging_dir/host/lib/liblzma.a" EXTRA_CFLAGS="-I/home/sabine/devel/openwrt/openwrt_git/staging_dir/host/include" mksquashfs unsquashfs
make[4]: Entering directory '/home/sabine/devel/openwrt/openwrt_git/build_dir/host/squashfs4.2/squashfs-tools'
ccache gcc   mksquashfs.o read_fs.o sort.o swap.o pseudo.o compressor.o gzip_wrapper.o lzma_xz_wrapper.o xz_wrapper.o lzma_xz_options.o /home/sabine/devel/openwrt/openwrt_git/staging_dir/host/lib/liblzma.a -lpthread -lm -lz -o mksquashfs
/home/sabine/devel/openwrt/openwrt_git/staging_dir/host/lib/liblzma.a(liblzma_la-check.o): In function `lzma_check_init':
check.c:(.text+0x65): undefined reference to `SHA256Init'
/home/sabine/devel/openwrt/openwrt_git/staging_dir/host/lib/liblzma.a(liblzma_la-check.o): In function `lzma_check_update':
check.c:(.text+0xbf): undefined reference to `SHA256Update'
/home/sabine/devel/openwrt/openwrt_git/staging_dir/host/lib/liblzma.a(liblzma_la-check.o): In function `lzma_check_finish':
check.c:(.text+0x105): undefined reference to `SHA256Final'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:211: mksquashfs] Error 1
make[4]: Leaving directory '/home/sabine/devel/openwrt/openwrt_git/build_dir/host/squashfs4.2/squashfs-tools'
make[3]: *** [Makefile:43: /home/sabine/devel/openwrt/openwrt_git/build_dir/host/squashfs4.2/.built] Error 2
make[3]: Leaving directory '/home/sabine/devel/openwrt/openwrt_git/tools/squashfs4'
make[2]: *** [tools/Makefile:123: tools/squashfs4/compile] Error 2
make[2]: Leaving directory '/home/sabine/devel/openwrt/openwrt_git'
make[1]: *** [tools/Makefile:121: /home/sabine/devel/openwrt/openwrt_git/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2_wt3020/stamp/.tools_install_yynyyynynynyyyyyyyyynyyyyyyyyynyyyyynnyyynnyynnnyy] Error 2
make[1]: Leaving directory '/home/sabine/devel/openwrt/openwrt_git'
make: *** [/home/sabine/devel/openwrt/openwrt_git/include/toplevel.mk:183: world] Fehler 2

seems to be some header mismatch between liblzma and squashfs4 that I fail to interpret.

Regards

wigyori, 2016-10-05:
Can you please do a make distclean and try again.

wararjey, 2016-10-06:
Tried building from a completely fresh tree, got the same error messages. Could there be a mismatch with xz installed on the host? I'm on Arch Linux that has xz 5.2.2 installed....

Regards,

closed 2022-02-09 by aparcar