openwrt/openwrt #128: MT7620N WIFI not work

Edwardzhangyuhang, 2016-10-12:
I have build the image for mt7620n
And wifi driver do not work
Through the dmesg, I found 
[    8.929788] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 5390, rev 0500 detected
[    8.937810] ieee80211 phy0: rt2800_init_eeprom: Error - Invalid RF chipset 0xffff detected
[    8.946271] ieee80211 phy0: rt2x00lib_probe_dev: Error - Failed to allocate device

Did the trunk do not support this chip or some configeration I do not enable?

gmzhuo, 2016-10-12:
Looks like factory data is invalid.

Edwardzhangyuhang, 2016-10-12:
So what I can do to fix it?

gmzhuo, 2016-10-12:
Reflash factory mtd with correct factory data,and adjust wifi parameter.

Edwardzhangyuhang, 2016-10-12:
Thanks
so where can i get the correct factory data, I have no idea of that.

gmzhuo, 2016-10-12:
You can copy it from another machine, but the RF parameter is not right,you need a device to test and correct it.

closed 2022-02-09 by aparcar