openwrt/openwrt #130: Tplink TL-740N wan port

alex-eri, 2016-10-12:
DESIGNATED DRIVER (Bleeding Edge, 49966)

On configuring network
```
[   39.063426] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:306 dev_watchdog+0x1dc/0x260()
[   39.072153] NETDEV WATCHDOG: eth1 (ag71xx): transmit queue 0 timed out
[   39.078902] Modules linked in: ath9k ath9k_common pppoe ppp_async iptable_nat ath9k_hw ath pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_ecn xt_dscp xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[   39.156004] CPU: 0 PID: 0 Comm: swapper Not tainted 4.4.14 #25
[   39.162036] Stack : 803ba164 00000000 00000001 80410000 804010d8 80400d63 8039b898 00000000
[   39.162036] 	  80463790 803ff3fc 00000200 00100000 00000100 800a678c 803a0edc 80400000
[   39.162036] 	  00000003 803ff3fc 8039f2e8 803fbc2c 00000100 800a4758 00000006 00000000
[   39.162036] 	  00000000 801ef200 00000000 00000000 00000000 00000000 00000000 00000000
[   39.162036] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   39.162036] 	  ...
[   39.199066] Call Trace:
[   39.201604] [<8007197c>] show_stack+0x50/0x84
[   39.206115] [<80081580>] warn_slowpath_common+0xa0/0xd0
[   39.211528] [<800815dc>] warn_slowpath_fmt+0x2c/0x38
[   39.216664] [<80280418>] dev_watchdog+0x1dc/0x260
[   39.221557] [<800af8a0>] call_timer_fn.isra.5+0x24/0x80
[   39.226956] [<800afb1c>] run_timer_softirq+0x1ac/0x1f4
[   39.232282] [<800839dc>] __do_softirq+0x164/0x290
[   39.237164] [<8006a5d0>] plat_irq_dispatch+0xbc/0xf4
[   39.242299] 

```

and after
```
[  277.058615] eth1: tx timeout
[  302.018602] eth1: tx timeout
[  327.018614] eth1: tx timeout
[  357.018613] eth1: tx timeout
[  387.018606] eth1: tx timeout
[  417.018604] eth1: tx timeout
[  447.018601] eth1: tx timeout
[  477.018604] eth1: tx timeout
```

and no taffic from device later. up/down interface did not help.

alex-eri, 2016-10-12:
related #35

wigyori, 2016-10-13:
Can you try the last snapshot or build one from a new checkout? The flow-control patch has been merged in https://github.com/openwrt/openwrt/pull/126

alex-eri, 2016-10-13:
@wigyori
seems I built with #126 and #127 merged. Will checkout last and rebuild to sure...

alex-eri, 2016-10-13:
Seems working!

closed 2016-10-13 by alex-eri

psyborg55, 2016-10-15:
had the same issue, after managed to get modified uboot running on my tplink the timeout bug no longer happens. so the tplink uboot was causing it.

mkresin, 2016-10-16:
Seams to me tp-link bootloader misses to initialise some switch/soc register which are required for flow-control.

@psyborg55 since you are able to reproduce the good and the bad state, would you please revert f16bdf493ff6fa0ced7ad80807befc08676665ad (aka enable flow control) and compile the kernel with debugging enabled. This way the driver should print at least [some register values](https://github.com/openwrt/openwrt/blob/master/target/linux/ar71xx/files/drivers/net/ethernet/atheros/ag71xx/ag71xx_main.c#L1356) on boot. By comparing the values from the good and bad state might help nbd to fix the underlying issue.

To make sure that you found something relevant, you can just overwrite the register value using ```ag71xx_wr()``` with the value form the good state while using the tp-link bootloader. The best place to do this, should be before the call to ```ag71xx_hw_init(ag)```.

It might be possible that the relevant register is an AR7240 specific one and not a common ag71xx. You can find the AR7240 specific register at the first lines of [ar7240.c](https://github.com/openwrt/openwrt/blob/master/target/linux/ar71xx/files/drivers/net/ethernet/atheros/ag71xx/ag71xx_ar7240.c).

Use ```printk()``` and ```ar7240sw_reg_read()``` to dump the AR7240 register. I only had a brief look at the source but ```ag71xx_ar7240_init``` seams to be good place for adding the ar7240 specific register dumping.

I don't have an affected device, but I was able to fix atheros switch configuration issues on other boards by dumping and comparing the register values. Just wanted to show a way how the issue can be properly fixed.