openwrt/openwrt #147: feeds/management/libmicroxml compile error

topeqin, 2016-10-24:
trunk/tmp/.libmicroxml-2012-06-11-72965423184f24cc0b963d91c2d1863cdb01b6aa.tar.gz.flock -c '     echo "Checking out files from the git repository..."; 
mkdir -p /home/gqin/openwrt_trunk/tmp/dl && cd /home/gqin/openwrt_trunk/tmp/dl && rm -rf libmicroxml-2012-06-11 && [ \! -d libmicroxml-2012-06-11 ] && git clone git://dev.freecwmp.org/microxml libmicroxml-2012-06-11 && (cd libmicroxml-2012-06-11 && git checkout 72965423184f24cc0b963d91c2d1863cdb01b6aa && git submodule update --init --recursive) && echo "Packing checkout..." && rm -rf libmicroxml-2012-06-11/.git &&        tar czf /home/gqin/openwrt_trunk/tmp/dl/libmicroxml-2012-06-11-72965423184f24cc0b963d91c2d1863cdb01b6aa.tar.gz libmicroxml-2012-06-11 && mv /home/gqin/openwrt_trunk/tmp/dl/libmicroxml-2012-06-11-72965423184f24cc0b963d91c2d1863cdb01b6aa.tar.gz /home/gqin/openwrt_trunk/dl/ && rm -rf libmicroxml-2012-06-11; '
Checking out files from the git repository...
Cloning into 'libmicroxml-2012-06-11'...
fatal: unable to connect to dev.freecwmp.org:
dev.freecwmp.org[0: 217.115.15.24]: errno=Connection timed out

make[3]: *** [/home/gqin/openwrt_trunk/dl/libmicroxml-2012-06-11-72965423184f24cc0b963d91c2d1863cdb01b6aa.tar.gz] Error 128
make[3]: Leaving directory `/home/gqin/openwrt_trunk/feeds/management/libmicroxml'
make[2]: *** [package/feeds/management/libmicroxml/compile] Error 2
make[2]: Leaving directory `/home/gqin/openwrt_trunk'
make[1]: *** [/home/gqin/openwrt_trunk/staging_dir/target-arm_cortex-a9_musl-1.1.14_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/gqin/openwrt_trunk'
make: *** [world] Error 2
gqin@gqin-virtual-machine:~/openwrt_trunk$

>>>>>>>>>>>>>>>>>>>

after check the log, the root cause is for the PKG_SOURCE_URL:=git://dev.freecwmp.org/microxml issue, maybe the server is down or other issue.

i only can download by mysefl first to pass the compile or change this URL and VERSION to another available git source would you please change this to a good one.
My fix using one source in github of pirvasoftware
PKG_VERSION:=2015-03-18 
PKG_SOURCE_URL:=https://github.com/pivasoftware/microxml.git
PKG_SOURCE_VERSION:=caa8d3e6887f5c70e54df555dd78e4e45cfa74cc 
>>>>>>>>>>>>>>>

topeqin, 2016-10-24:
if can I provide this fix then create a pull request to trunk branch. look forward to your response.

Thanks
Qin Guang

joker5bb, 2017-05-15:
the repo is still done and there is another package that uses that server

closed 2022-02-09 by aparcar