openwrt/openwrt #159: ar71xx-nand SDK broken in 15.05.1

brianjmurrell, 2016-10-29:
I'm trying to use the SDK to build some packages for CC 15.05.1 for the ar71xx-nand using the [instructions on the wiki](https://wiki.openwrt.org/doc/howto/obtain.firmware.sdk) but it doesn't seem to work the way that wiki page describes.

```
$ ./scripts/feeds install tmux
Installing package 'tmux'
Installing package 'toolchain'
Installing package 'ncurses'
Installing package 'libevent2'
Installing package 'openssl'
Installing package 'zlib'
Installing package 'ocf-crypto-headers'

$ make package/tmux/download
make[3]: *** No rule to make target `package/prereq'.  Stop.
make[2]: *** [/net/server/src/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_prereq] Error 2
make[1]: *** [prereq] Error 2
make: *** [package/tmux/download] Error 2
```

SrividhyaMH, 2017-05-31:
I followed the instructions on the wiki  and pulled the sdk from Stable SDK "Chaos Calmer (15.05.1)" → Platform → OpenWrt-SDK-<Platform>-for-linux-x86_64-gcc-<version>-linaro-uClibc-<version>.tar.bz2 and was able to build the package. 

After successful installation using "./scripts/feeds install tmux" following steps were done.

Step 1: make package/tmux/download
         I also faced some issues during make package/tmux/download, so i ran it again with "make -j1 V=s package/tmux/download" option [to get complete details of error] and found the below are the dependent packages which should be downloaded in dl/ directory from https://sources.lede-project.org/,
libevent-2.0.22-stable.tar.gz
ncurses-5.9.tar.gz
openssl-1.0.2j.tar.gz
tmux-1.9a.tar.gz
zlib-1.2.8.tar.gz

Step 2: make package/tmux/prepare - no issues

Step 3:make package/tmux/compile, 
               Found ccache issues while compiling zlib -1.2.8 configure file as ccache was not installed on my ubuntu machine. So installed it with "apt-get install ccache" and compilation was successful.

Finally, I could find that tmux_1.9a-1_ar71xx.ipk package installed in following directory
./bin/ar71xx/packages/packages/tmux_1.9a-1_ar71xx.ipk

On top of these steps if you are still facing issues please run make with -j1 V=s options enabled to get detailed logs.

brianjmurrell, 2017-05-31:
> found the below are the dependent packages which should be downloaded in dl/ directory

Is this not something that should happen implicitly/automatically?

If not, one at least, one should not have to (even know one has to) resort to `make -j1 V=s` to discover that there are missing dependencies.  If dependencies are needed, they should be installed automatically, or in the least, they should be listed clearly as required instead of a cryptic error message such as:

```
make[3]: *** No rule to make target `package/prereq'.  Stop.
make[2]: *** [/net/server/src/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_prereq] Error 2
make[1]: *** [prereq] Error 2
make: *** [package/tmux/download] Error 2
```

> On top of these steps if you are still facing issues please run make with -j1 V=s options enabled to get detailed logs.

This is probably not necessary as I am quite confident that what you discovered is what I was facing.

So this issue of dependencies needs to be fixed.  Either installing them automatically (preferred) or displaying a clear message that they need to be installed first.

closed 2022-02-09 by aparcar