openwrt/openwrt #162: build error e2fsprogs

iddq, 2016-10-31:
```
CONFIG_TARGET_ar71xx_generic_TLWDR4300=y

MD5 sum of the downloaded file does not match (file: 24d346327cbbbb9d8865517a12cb3857, requested: 1775f3f0eed9dee1c5f39e08d1964a97) - deleting download.
converted 'http://downloads.sourceforge.net/e2fsprogs/e2fsprogs-1.43.1.tar.gz' (ANSI_X3.4-1968) -> 'http://downloads.sourceforge.net/e2fsprogs/e2fsprogs-1.43.1.tar.gz' (UTF-8)
--2016-10-31 10:19:58--  http://downloads.sourceforge.net/e2fsprogs/e2fsprogs-1.43.1.tar.gz
Resolving downloads.sourceforge.net (downloads.sourceforge.net)... 216.34.181.59
Connecting to downloads.sourceforge.net (downloads.sourceforge.net)|216.34.181.59|:80... connected.
HTTP request sent, awaiting response... 307 Temporary Redirect
Location: http://sourceforge.net/projects/e2fsprogs/files/e2fsprogs/v1.43.1/e2fsprogs-1.43.1.tar.gz/download?use_mirror=kent [following]
converted 'http://sourceforge.net/projects/e2fsprogs/files/e2fsprogs/v1.43.1/e2fsprogs-1.43.1.tar.gz/download?use_mirror=kent' (ANSI_X3.4-1968) -> 'http://sourceforge.net/projects/e2fsprogs/files/e2fsprogs/v1.43.1/e2fsprogs-1.43.1.tar.gz/download?use_mirror=kent' (UTF-8)
--2016-10-31 10:19:59--  http://sourceforge.net/projects/e2fsprogs/files/e2fsprogs/v1.43.1/e2fsprogs-1.43.1.tar.gz/download?use_mirror=kent
Resolving sourceforge.net (sourceforge.net)... 216.34.181.60
Connecting to sourceforge.net (sourceforge.net)|216.34.181.60|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sourceforge.net/projects/e2fsprogs/files/e2fsprogs/v1.43.1/e2fsprogs-1.43.1.tar.gz/download?use_mirror=kent [following]
converted 'https://sourceforge.net/projects/e2fsprogs/files/e2fsprogs/v1.43.1/e2fsprogs-1.43.1.tar.gz/download?use_mirror=kent' (ANSI_X3.4-1968) -> 'https://sourceforge.net/projects/e2fsprogs/files/e2fsprogs/v1.43.1/e2fsprogs-1.43.1.tar.gz/download?use_mirror=kent' (UTF-8)
--2016-10-31 10:19:59--  https://sourceforge.net/projects/e2fsprogs/files/e2fsprogs/v1.43.1/e2fsprogs-1.43.1.tar.gz/download?use_mirror=kent
Connecting to sourceforge.net (sourceforge.net)|216.34.181.60|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 69814 (68K) [text/html]
Saving to: 'STDOUT'

blablabla

Download failed.
converted 'http://downloads.openwrt.org/sources/e2fsprogs-1.43.1.tar.gz' (ANSI_X3.4-1968) -> 'http://downloads.openwrt.org/sources/e2fsprogs-1.43.1.tar.gz' (UTF-8)
--2016-10-31 10:20:07--  http://downloads.openwrt.org/sources/e2fsprogs-1.43.1.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-10-31 10:20:08 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
Makefile:49: recipe for target '/usr/src/openwrt/dl/e2fsprogs-1.43.1.tar.gz' failed
make[3]: *** [/usr/src/openwrt/dl/e2fsprogs-1.43.1.tar.gz] Error 2
make[3]: Leaving directory '/usr/src/openwrt/tools/e2fsprogs'
tools/Makefile:134: recipe for target 'tools/e2fsprogs/compile' failed
make[2]: *** [tools/e2fsprogs/compile] Error 2
make[2]: Leaving directory '/usr/src/openwrt'
tools/Makefile:133: recipe for target '/usr/src/openwrt/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.tools_install_yynyynynynyyyyyyynyyyyyyyyyyyyynyyyyynnyynynyyyynnnyy' failed
make[1]: *** [/usr/src/openwrt/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.tools_install_yynyynynynyyyyyyynyyyyyyyyyyyyynyyyyynnyynynyyyynnnyy] Error 2
make[1]: Leaving directory '/usr/src/openwrt'
/usr/src/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2

```

SrividhyaMH, 2017-05-29:
You can download the package e2fsprogs-1.43.1.tar.gz from https://sources.lede-project.org/ and place it in the dl (openwrt/dl/) and build again using make -j1 V=s.

ktjr, 2017-06-02:
Hello guys,
I dunno weather I could rise a request here or not. Am sorry for that.
Could some one help me to install OpenWRT on my Raspberry Pi 3. In fact am
new to this field itself.
Will installing this Open WRT allo multiple VLANS available on my Pi?
Thanks for any suggestions in advance,
--
Tejaswini Kokkanti (M.Tech)

closed 2022-02-09 by aparcar