openwrt/openwrt #187: dropbear exit bug

psyborg55, 2016-11-04:
running trunk with dropbear 2016.74-1 and pressing ctrl-c to exit for example from airodump console sometimes kills ssh connection completely. the bug has been present with previous versions of dropbear but i didn't pay much attention to it since i'm running outdated 10.04. today noticed the same thing happen on 15.04 too. since i don't remember now if it ever happened while editing some files or doing something else, anyone interested please try to reproduce the issue to see if it is really related to dropbear or airodump.

CodeFetch, 2016-11-04:
As you state it seems to be an Dropbear or Airodump related issue. You might have better luck asking for help there...
https://www.aircrack-ng.org/doku.php?id=Main#support
https://github.com/mkj/dropbear

psyborg55, 2016-11-08:
no way,. just followed a suggestion "For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses"

CodeFetch, 2016-11-08:
You could try to reproduce this on an „ordinary“ linux computer with the latest versions of aircrack and dropbear. If the same issue occurs, it's not an OpenWrt related problem. Personally I doubt any OpenWrt developer will do this for you any time soon as this bug is minor.

closed 2022-02-09 by aparcar