openwrt/openwrt #243: mt7620a rt2800mmio_txstatus_is_spurious

psyborg55, 2016-11-25:
r49946, doing a file transfer from pc results in logs filled with these messages. affected throughput would drop by more than 70% every once in a while and it seems to correspond the timings between these occurrences.

>     [ 8849.424094] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8863.460859] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8863.473622] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8886.828341] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8886.838917] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8913.848401] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8913.858821] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8950.572351] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8950.584290] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8962.556817] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8962.570305] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8965.706955] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8965.719024] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8975.292700] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8975.303387] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8982.510360] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8982.522476] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8994.245717] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8994.256742] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 8995.410230] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 8995.420808] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9000.638877] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9000.650976] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9027.367010] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9027.377953] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9039.905590] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9039.916544] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9094.038935] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9094.051367] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9095.104431] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9095.116925] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9113.346387] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9113.358873] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9124.735628] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9124.746132] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9124.871074] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9124.881587] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9134.270597] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9134.281732] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9140.339837] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9140.351944] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9149.168867] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9149.179811] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9152.976021] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9152.987029] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9179.659501] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9179.670512] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9199.615620] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9199.628212] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9210.357671] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9210.368163] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9223.991434] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9224.003823] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9224.711192] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9224.722502] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9265.217593] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9265.229761] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9297.348242] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9297.360891] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9301.035442] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9301.047933] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9358.380095] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9358.390571] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9368.588399] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9368.598726] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9374.832269] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [ 9374.844653] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [ 9422.112215] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
[... 746 lines cut ...]
>     [22406.902559] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [22451.211326] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [22451.221850] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
[... cut ...]
>     [23372.861793] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [23389.963568] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [23389.976022] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping
>     [23430.708043] ieee80211 phy1: rt2800mmio_txstatus_is_spurious: Warning - 4 spurious TX_FIFO_STATUS interrupt(s)
>     [23430.720052] ieee80211 phy1: rt2800mmio_txdone: Warning - Got TX status for an empty queue 2, dropping

psyborg55, 2016-11-26:
update: throughput drop is related to some other problem as it manifests quite less frequent than these warnings, the number of warnings per minute increased when set to upload more files at the same time

bobdenardd, 2016-12-20:
Hey there. I experienced the exact same problem with this CPU (on a mt300a device). The throughput drops and even the wifi drops, when under heavy load (when using wifi n). Switching to wifi g solves the problem, but I'd like to benefit from 300mbps. Did you get any news on this? (I feel like a lot of users actually experience this problem)

psyborg55, 2017-01-13:
can you check if this patch solves the problem? https://github.com/dangowrt/linux/commit/76d0bee74e334d26065df30a0855bd5a4d60d47a

psyborg55, 2017-05-08:
do you have any device attached to usb port?

psyborg55, 2017-07-14:
with updated mt7620 patch this problem manifests a bit different:

```
[118607.360336] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118634.309727] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118634.835240] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118640.069995] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118651.735466] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118681.232420] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118686.027445] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118698.155563] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118705.512591] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118713.026111] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118723.828708] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118756.967596] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118775.394380] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118776.265478] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118777.297109] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118778.978200] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118782.188762] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118788.891415] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118792.873331] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118798.133936] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118798.471836] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118799.768388] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118817.317664] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118822.364042] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118826.563119] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118835.517934] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118838.454480] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118847.011520] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118848.307916] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118851.886946] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118852.059923] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118874.678601] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118880.673456] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[118890.452591] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118891.928415] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010e, type=4
[118900.237013] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
[... cut ...]
[121806.618674] ieee80211 phy1: rt2x00lib_rxdone_read_signal: Warning - Frame received with unrecognized signal, mode=0x0001, signal=0x010f, type=4
```

running iperf with one or two streams (-P option) doesn't make these warnings printed to logs, but increasing stream count to 3, 4 or more seems to trigger them more frequently.

because of this i think it might be CPU related (processing power/optimization) problem. avg load in luci was also between 2 and 3. unfortunately our chinese friend hackpascal hasn't made any breed update that would allow to set higher clocks than 620MHz

closed 2022-02-09 by aparcar