openwrt/openwrt #257: Trunk ath9k is very unstable on TP-Link TL-WR1043N/ND v1

iandall, 2016-12-04:
In trunk, my ath9k wifi won't stay up, typically for more than a few minutes. This is somewhat volume related. When there are only mobile devices doing not much, then the wifi can stay up for a few hours at least, but as soon as you do some heavy traffic, like streaming a YouTube video, the wifi fails within about 5 minutes.

The ath9k wifi is quite stable in Chaos-Calmer 15.05.1 on the same hardware and network configuration.

I see nothing in the logs with looks at all interesting. It seem as though power goes to zero. The wlan layer reports stations disassociated due to lack of activity, but presumably that is an effect, not a cause.\

I spent quite a bit of time on this but didn't get anywhere. I did wonder if the GPIO pins, which are meant to implement manual wifi on/off, on this platform are floating or connected to something else so that the wifi is turned off spuriously. I don't have much to base this hypothesis on except that quite a few of the changes to this driver since Chaos Calmer are to do with gpio pins.

asferreira, 2017-01-14:
I've been suffering the same problem since I've installed trunk on my router. Clients are disconnected, the router stops to publish SSID. All problems are temporarily solved after a router reboot, but after a while the problems appear again. My solution was to got back my router to Chaos Calmer.

Snotmann, 2017-02-22:
Got the same here ... seems to be a driver problem ... will go back to Chaos Calmer

nbd168, 2017-02-23:
This is fixed in current LEDE versions, OpenWrt needs a backport of the mac80211 package.

Snotmann, 2017-02-23:
Hm i hope the issue will get fixed ... i got not enought linux know-how to do that and Chaos Calmer got a few things that annoy me ;)

closed 2022-02-09 by aparcar