openwrt/openwrt #261: WDR3600 CC 15.05.1

himuracarter, 2016-12-05:
[  350.024000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 3 is up
[16700.128000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 3 is down
[26204.200000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 3 is up
[38910.284000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 3 is down
[49326.676000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 4 is up
[54637.140000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 4 is down
[54639.140000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 4 is up
[74255.300000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 3 is up
[74299.300000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 4 is down
[76805.308000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 3 is down
[111147.752000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 3 is up

How to fix\solve this?

oliv3r, 2017-03-02:
I see the same, only on port 4 however. When this happens, I also loose wifi on my device for a few seconds. I was running a 2 year old version of openwrt before and never had this problem.

himuracarter, 2017-03-02:
It's fixed in LEDE 17.01.1

closed 2017-03-02 by himuracarter

rpodric, 2017-03-30:
Mind pointing to some details about the fix?  It's been an issue that's been around for years, and if there something that can be done about it short of moving to unreleased firmware, I'd like to know.