openwrt/openwrt #263: ftp kernel.org down ?

keulu, 2016-12-07:
since yesterday (after upgrading to the latested Chaos Calmer release), i can't compile

`
--2016-12-07 09:50:12--  ftp://ftp.all.kernel.org/pub/linux/kernel/v3.x/linux-3.18.44.tar.xz
           => '-'
Resolving ftp.all.kernel.org (ftp.all.kernel.org)... 198.145.20.140, 149.20.4.69, 199.204.44.194
Connecting to ftp.all.kernel.org (ftp.all.kernel.org)|198.145.20.140|:21... failed: Connection timed out.
Connecting to ftp.all.kernel.org (ftp.all.kernel.org)|149.20.4.69|:21... failed: Connection timed out.
Connecting to ftp.all.kernel.org (ftp.all.kernel.org)|199.204.44.194|:21... failed: Connection timed out.
Retrying.

--2016-12-07 09:51:13--  ftp://ftp.all.kernel.org/pub/linux/kernel/v3.x/linux-3.18.44.tar.xz
  (try: 2) => '-'
Connecting to ftp.all.kernel.org (ftp.all.kernel.org)|198.145.20.140|:21... failed: Connection timed out.
Connecting to ftp.all.kernel.org (ftp.all.kernel.org)|149.20.4.69|:21... failed: Connection timed out.
Connecting to ftp.all.kernel.org (ftp.all.kernel.org)|199.204.44.194|:21... failed: Connection timed out.
Retrying.

`

Thynix, 2017-01-02:
This isn't a problem OpenWRT's developers can fix.

irongomme, 2017-01-23:
This is a French (or european ?) issue, you can use https://ftp.all.kernel.org instead !

murix, 2017-03-07:
To workaround it without changing the code, is possible pre-download require the kernel to dl folder.
Example:
wget https://www.kernel.org/pub/linux/kernel/v3.0/linux-3.18.45.tar.xz -O dl/linux-3.18.45.tar.xz

murix, 2017-03-07:
Kernel.org shut down FTP services on March 1, 2017
https://kernel.org/shutting-down-ftp-services.html

closed 2022-02-09 by aparcar