openwrt/openwrt #267: mt7620 ap and adhoc mode could not working at same time.

chaokw, 2016-12-09:
modify /etc/config/wireless to enable two interface ap0 and adhoc0, but could not working at same time.
only using ap0 or adhoc0 is fine.

config wifi-device radio0
option type mac80211
option channel 10
option hwmode 11n
option path '10180000.wmac'
option htmode HT40
option txpower 20
option country 00
option beacon_int 3000

config wifi-iface
option device radio0
option ifname ap0
option network lan
option mode ap
option ssid openwrt_test
option encryption none
option wmm 0

config wifi-iface
option device radio0
option ifname adhoc0
option encryption none
option network mesh
option mode adhoc
option ssid 'mesh'

bittu072, 2016-12-13:
Hi,

I am interested in contributing. As I am new to opensource please let me know how to contribute to this issue?
And is station mode same as adhoc mode?

chaokw, 2016-12-13:
Hello,

As i know, sta mode also could not work simultaneously with ap mode by VIF. 
Maybe adhoc belong to sta mode. Because MTK original driver only support AP and STA compiled separately. STA driver like RT2860STA.ko.After insmod RT2860STA.ko, we can change it to adhoc mode by using cmd "iwpriv ra0 set NetworkType=Adhoc".
openwrt not use MTK original driver,  but use mac80211.
Atheros 9344 also using mac80211 driver, but could let ap and adhoc mode working at same time.

bittu072, 2016-12-17:
Hi again,
I am not sure about mediatek(MTK), but I am using atheros and I can use both STA mode and AP mode simultaneously without any additional cmd

chaokw, 2016-12-20:
Hi bittu072
The same phenomenon compare with you. atheros9344 can use STA and AP simultaneously. Hower, MTK can not. So, maybe some adapt bug in mt7620 wifi driver.

brunovianna, 2017-06-10:
did you solve this? apparently there is a guy working on the mt7620: https://www.kickstarter.com/projects/1327597961/better-support-for-mt7620a-n-in-openwrt-lede/updates
did you try his patches?

closed 2022-02-09 by aparcar