openwrt/openwrt #276: build error on Ubuntu 16.04.01 LTS

mmitch, 2016-12-17:
(sorry for the generic bug title, please update as needed)

I currently (did not try for some time, last working build is from 2016-07-31) can't build OpenWRT on my machine. I can reproduce the error with the following steps:

1. do a pristine checkout of the git repository (last try with commit 851a890647205e58363f391664696104a6608464)
2. call `make menuconfig`
3. change "Target_ Profile" to "Buffalo WHR-HP-G300N", this unsets `CONFIG_TARGET_ar71xx_generic_Default` and sets `CONFIG_TARGET_ar71xx_generic_WHRHPG300N=y` in `.config`
4. exit menuconfig and save configuration
5. run `make`
6. this takes some time, the last lines of the build are these:
```
[…]
 make[3] -C toolchain/fortify-headers install
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] package/cleanup
 make[2] package/compile
 make[3] -C package/libs/toolchain compile
make -r world: build failed. Please re-run make with -j1 V=s to see what's going on
/home/mitch/OpenWRT/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 1
```
7. run `make -j1 V=s` as advised, output looks like this:
```
mitch@zecora:~/OpenWRT/openwrt$ make -j1 V=s
fatal: No names found, cannot describe anything.
fatal: No names found, cannot describe anything.
fatal: No names found, cannot describe anything.
make[1]: Entering directory '/home/mitch/OpenWRT/openwrt'
make[2]: Entering directory '/home/mitch/OpenWRT/openwrt'
make[3]: Entering directory '/home/mitch/OpenWRT/openwrt/target/linux'
make[4]: Entering directory '/home/mitch/OpenWRT/openwrt/target/linux/ar71xx'
/home/mitch/OpenWRT/openwrt/scripts/kconfig.pl  + + /home/mitch/OpenWRT/openwrt/target/linux/generic/config-4.4 /home/mitch/OpenWRT/openwrt/target/linux/ar71xx/config-4.4 /home/mitch/OpenWRT/openwrt/target/linux/ar71xx/generic/config-default > /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/mitch/OpenWRT/openwrt/.config >> /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.target
/home/mitch/OpenWRT/openwrt/scripts/metadata.pl kconfig /home/mitch/OpenWRT/openwrt/tmp/.packageinfo /home/mitch/OpenWRT/openwrt/.config 4.4 > /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.override
/home/mitch/OpenWRT/openwrt/scripts/kconfig.pl 'm+' '+' /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.target /dev/null /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.override > /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.set
mv /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.set /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.old
grep -v INITRAMFS /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.old > /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.set
rm -rf /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/modules
cmp -s /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.set /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.prev || { cp /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.set /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config; cp /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.set /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.prev; }
export MAKEFLAGS= ; [ -d /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/user_headers ] || make -C /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14 HOSTCFLAGS="-O2 -I/home/mitch/OpenWRT/openwrt/staging_dir/host/include -I/home/mitch/OpenWRT/openwrt/staging_dir/host/usr/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" CONFIG_SHELL="bash" V=''  CC="mips-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/user_headers headers_install
. /home/mitch/OpenWRT/openwrt/include/shell.sh; grep '=[ym]' /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.config.set | LC_ALL=C sort | md5s > /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.vermagic
touch /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.configured
rm -f /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/vmlinux /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/System.map
make -C /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14 HOSTCFLAGS="-O2 -I/home/mitch/OpenWRT/openwrt/staging_dir/host/include -I/home/mitch/OpenWRT/openwrt/staging_dir/host/usr/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" CONFIG_SHELL="bash" V=''  CC="mips-openwrt-linux-musl-gcc" modules
make[5]: Entering directory '/home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14'
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  Building modules, stage 2.
  MODPOST 48 modules
make[5]: Leaving directory '/home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14'
touch /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/linux-ar71xx_generic/linux-4.4.14/.modules
make -C image compile TARGET_BUILD=
make[5]: Entering directory '/home/mitch/OpenWRT/openwrt/target/linux/ar71xx/image'
make[5]: Nothing to be done for 'compile'.
make[5]: Leaving directory '/home/mitch/OpenWRT/openwrt/target/linux/ar71xx/image'
make[4]: Leaving directory '/home/mitch/OpenWRT/openwrt/target/linux/ar71xx'
make[3]: Leaving directory '/home/mitch/OpenWRT/openwrt/target/linux'
make[2]: Leaving directory '/home/mitch/OpenWRT/openwrt'
make[2]: Entering directory '/home/mitch/OpenWRT/openwrt'
rm -rf /home/mitch/OpenWRT/openwrt/staging_dir/target-mips_34kc_musl-1.1.15/root-ar71xx
[... 3 lines cut ...]
mkdir -p /home/mitch/OpenWRT/openwrt/bin/ar71xx/packages /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/toolchain/ipkg-ar71xx/libgcc/CONTROL /home/mitch/OpenWRT/openwrt/staging_dir/target-mips_34kc_musl-1.1.15/pkginfo
install -d -m0755 /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/toolchain/ipkg-ar71xx/libgcc/lib
[... cut ...]
package/Makefile:193: recipe for target '/home/mitch/OpenWRT/openwrt/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/mitch/OpenWRT/openwrt/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mitch/OpenWRT/openwrt'
/home/mitch/OpenWRT/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2
```

What causes this error?
Can I fix this by changing my build configuration?

mauron85, 2016-12-22:
I remember same experience openwrt failing on Ubuntu. I've sucessfuly bould openwrt chaos calmer under docker container running latest debian image.

mauron85, 2016-12-22:
I've found my Dockerfile. In case of chaos calmer many source packages were not available (dead links). So it was painful process of searching alternative links on internet and putting them into download folder. Finally I've created `openwrt.src.tar` with all library sources included.

Here is the Dockerfile I've used

```
FROM debian

# Add aditional debian sources (contrib,...) into sources.d
COPY conf_sources.sh /tmp/
RUN /tmp/conf_sources.sh

# Install and set-up the current image - run only when creating the image
RUN apt-get update && apt-get install -y \
  build-essential \
  libncurses5-dev \
  gawk \
  git \
  subversion \
  libssl-dev \
  gettext \
  unzip \
  zlib1g-dev \
  file python \
  mercurial \
  flex \
  quilt \
  xsltproc \
  libxml-parser-perl \
  bzr \
  ecj \
  cvs \
  wget \
  b43-fwcutter

# Default working directory of all containers from this image
WORKDIR .

# Extract openwrt sources
ADD openwrt.src.tar /openwrt
```

mmitch, 2016-12-27:
I don't do Docker, but I have tried a build inside a Debian 8.0 chroot: It worked fine.  So that is kind of a workaround that only takes up some space on my harddisk.

What makes me wonder is that in issue #284 @thagabe says that Ubuntu 16.04 LTS is supported.
The packages listed there (`build-essential flex gettext git gawk libncurses5-dev make pkg-config python unzip zlib1g-dev subversion libssl-dev`) are installed, but the original error given in this bug report still remains when I try to build under Ubuntu.

xcb567, 2016-12-29:
This bug looks a lot like this libgcc bug, https://github.com/openwrt/packages/issues/3343

mmitch, 2016-12-29:
@xcb567 I can try that on the next reboot, but it does not look like a segmentation fault to me, but rather like some paths or variables getting messed up. The 'buggy' part of the log seems to be this:
```
install -d -m0755 /home/mitch/OpenWRT/openwrt/bin/ar71xx/packages/base
/home/mitch/OpenWRT/openwrt/scripts/ipkg-build -c -o 0 -g 0 /home/mitch/OpenWRT/openwrt/build_dir/target-mips_34kc_musl-1.1.15/toolchain/ipkg-ar71xx/libgcc /home/mitch/OpenWRT/openwrt/bin/ar71xx/packages/base
gzip: /bin/gzip.gz already exists;      not overwritten
```

...which raises the question why `/bin/gzip.gz` exists on my host system. It is indeed a leftover of an earlier failure, where I thought "hey, my user is not allowed to write to /bin so I'll just place that file there and see what happens":
```
$ ls -lh /bin/gzip.gz 
-rw-r--r-- 1 mitch mitch 0 Okt 24 00:13 /bin/gzip.gz
```

I will remove the file and see what happens. (I expect the error message to change into something like "can't write to /bin/gzip.gz")

thagabe, 2016-12-30:
@mmitch as mentioned on my last post using ubuntu 16.04.1 server the built finished and ran fine, maybe it is your target that something broke, try building a generic x86 if not try a virtual machine to see maybe it is your host environment that isnt working. Thanks

PolynomialDivision, 2017-04-21:
@thagabe I'm experiencing the same bug under Xubuntu 16.04.2 LTS

mmitch, 2017-05-23:
I have made some progress while playing around with the LEDE SDK packages and running into the same error:
```
#
# configuration written to .config
#
make[1]: Entering directory '/tmp/lede-sdk-17.01.1-lantiq-xrx200_gcc-5.4.0_musl-1.1.16.Linux-x86_64'
make[2]: Entering directory '/tmp/lede-sdk-17.01.1-lantiq-xrx200_gcc-5.4.0_musl-1.1.16.Linux-x86_64'
make[3]: Entering directory '/tmp/lede-sdk-17.01.1-lantiq-xrx200_gcc-5.4.0_musl-1.1.16.Linux-x86_64/package/linux'
gzip: /bin/gzip.gz: Permission denied
Makefile:63: recipe for target '/tmp/lede-sdk-17.01.1-lantiq-xrx200_gcc-5.4.0_musl-1.1.16.Linux-x86_64/bin/targets/lantiq/xrx200/packages/kernel_4.4.61-1-207df999b35f2d3e27085fc93000a358_mips_24kc.ipk' failed
```
Running `make` inside a clean environment fixed it:
`env -i PATH=/usr/sbin:/usr/bin:/sbin:/bin make -j1 V=s`

I don't know what exactly is the problem with my environment, but this seems to be a valid workaround.
(I have not yet tested the workaround with an OpenWRT build, but I am confident ;-)

Machine-Hum, 2018-01-11:
Any update on this? Technique mentioned above does not work for me.

Machine-Hum, 2018-01-11:
svn: E170013: Unable to connect to a repository at URL 'http://repo.ham.fi/svn/aprx/trunk'

This is the error that proceeds

commodo, 2018-01-11:
what if you retry with a clean checkout ?
does it still happen ?
fwiw: the openwrt tree has reset to what used to be LEDE [ i.e. LEDE & OpenWrt re-merged back ]

Machine-Hum, 2018-01-11:
I just pulled **aprx-2.00_r421.tar.bz2** from the internet and dropped it in the dl folder. There was an SVN error proceeding.

closed 2022-02-09 by aparcar