openwrt/openwrt #284: compile error on ubuntu 15.10

qianguozheng, 2016-12-23:
weeds@R:~$ uname -a
Linux R 4.2.0-42-generic #49-Ubuntu SMP Tue Jun 28 21:26:26 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
weeds@R:~$ cat /etc/issue
Ubuntu 15.10 \n \l

Compile error follows:
```
weeds@R:~/magicwifi/oschina/mt7628/openwrt$ make V=s
Collecting package info: done
Collecting target info: done
make[1]: Entering directory '/home/weeds/magicwifi/oschina/mt7628/openwrt'
make[2]: Entering directory '/home/weeds/magicwifi/oschina/mt7628/openwrt'
+ mkdir -p /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15
+ cd /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15
+ mkdir -p bin lib include stamp
mkdir -p /home/weeds/magicwifi/oschina/mt7628/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/stamp
touch /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/.prepared
+ mkdir -p /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/host
+ cd /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/host
+ mkdir -p bin lib include stamp
mkdir -p /home/weeds/magicwifi/oschina/mt7628/openwrt/build_dir/host/stamp /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/host/include/sys
install -m0644 /home/weeds/magicwifi/oschina/mt7628/openwrt/tools/include/*.h /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/host/include/
install -m0644 /home/weeds/magicwifi/oschina/mt7628/openwrt/tools/include/sys/*.h /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/host/include/sys/
ln -sf lib /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/host/lib64
touch /home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/host/.prepared
make[3]: Entering directory '/home/weeds/magicwifi/oschina/mt7628/openwrt/tools/flock'
/home/weeds/magicwifi/oschina/mt7628/openwrt/include/shell.sh: line 35: md5: command not found
/home/weeds/magicwifi/oschina/mt7628/openwrt/include/shell.sh: line 35: md5: command not found
/home/weeds/magicwifi/oschina/mt7628/openwrt/include/shell.sh: line 35: md5: command not found
/home/weeds/magicwifi/oschina/mt7628/openwrt/include/shell.sh: line 35: md5: command not found
Makefile:25: *** target file '/home/weeds/magicwifi/oschina/mt7628/openwrt/build_dir/host/flock/.preparedprint' has both : and :: entries.  Stop.
make[3]: Leaving directory '/home/weeds/magicwifi/oschina/mt7628/openwrt/tools/flock'
tools/Makefile:134: recipe for target 'tools/flock/compile' failed
make[2]: *** [tools/flock/compile] Error 2
make[2]: Leaving directory '/home/weeds/magicwifi/oschina/mt7628/openwrt'
tools/Makefile:133: recipe for target '/home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/stamp/.tools_install_yynyynynynyyyyyyynyyynyyyyyyyyynyyyyynnyynynnyyynnnyy' failed
make[1]: *** [/home/weeds/magicwifi/oschina/mt7628/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/stamp/.tools_install_yynyynynynyyyyyyynyyynyyyyyyyyynyyyyynnyynynnyyynnnyy] Error 2
make[1]: Leaving directory '/home/weeds/magicwifi/oschina/mt7628/openwrt'
/home/weeds/magicwifi/oschina/mt7628/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2
weeds@R:~/magicwifi/oschina/mt7628/openwrt$ 

```

qianguozheng, 2016-12-23:
not matter download source code from current branch nor 15.04 branch, still comes such problem

thagabe, 2016-12-23:
Ubuntu 15.10 and 15.04 are no longer supported, please update the LTS 16.04 or 16.10

qianguozheng, 2016-12-25:
what a pity, can ubuntu support autoupgrade by itself ? or not destroy all my desktop work environment? I install a lot of tools and use it as work environment.

qianguozheng, 2016-12-25:
what a pity, can ubuntu support autoupgrade by itself ? or not destroy all my desktop work environment? I install a lot of tools and use it as work environment. @thagabe

thagabe, 2016-12-25:
I believe it can, however there might me some cleaning up you might have to do manually as well as some packages/repos that need adjusting or wont work at all. This is the main reason why work environments should stick to LTS (14.04 and 16.04 and the future 18.04) so you only need to upgrade every 2 years the the support is for 4 years i believe

thagabe, 2016-12-25:
That being said, I doubt the distro (even ubuntu 15.04) matters as long as the packages have been installed. 
try this:
'''
$ sudo apt-get update
$ sudo apt-get -y upgrade
$ sudo apt-get -y dist-upgrade
$ sudo apt-get -y autoremove
$ sudo apt-get autoclean
$ sudo apt-get -y install build-essential flex gettext git gawk libncurses5-dev make pkg-config python unzip zlib1g-dev subversion libssl-dev 
''' 
This should get you to the latest packages and system then installed all the packages openwrt needs to build an image. 

Hope this helps

qianguozheng, 2016-12-26:
@thagabe  what a pity, I upgrade system to 16.04 and download a new system, and it still not work right.

```
Collecting package info: done
Collecting target info: done
make[1]: Entering directory '/home/weeds/magicwifi/oschina/mt7628/openwrt-1'
make[2]: Entering directory '/home/weeds/magicwifi/oschina/mt7628/openwrt-1'
+ mkdir -p /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/target-mips_34kc_musl-1.1.15
+ cd /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/target-mips_34kc_musl-1.1.15
+ mkdir -p bin lib include stamp
mkdir -p /home/weeds/magicwifi/oschina/mt7628/openwrt-1/build_dir/target-mips_34kc_musl-1.1.15/stamp
touch /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/target-mips_34kc_musl-1.1.15/.prepared
+ mkdir -p /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/host
+ cd /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/host
+ mkdir -p bin lib include stamp
mkdir -p /home/weeds/magicwifi/oschina/mt7628/openwrt-1/build_dir/host/stamp /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/host/include/sys
install -m0644 /home/weeds/magicwifi/oschina/mt7628/openwrt-1/tools/include/*.h /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/host/include/
install -m0644 /home/weeds/magicwifi/oschina/mt7628/openwrt-1/tools/include/sys/*.h /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/host/include/sys/
ln -sf lib /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/host/lib64
touch /home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/host/.prepared
make[3]: Entering directory '/home/weeds/magicwifi/oschina/mt7628/openwrt-1/tools/flock'
/home/weeds/magicwifi/oschina/mt7628/openwrt-1/include/shell.sh: line 35: md5: command not found
/home/weeds/magicwifi/oschina/mt7628/openwrt-1/include/shell.sh: line 35: md5: command not found
/home/weeds/magicwifi/oschina/mt7628/openwrt-1/include/shell.sh: line 35: md5: command not found
/home/weeds/magicwifi/oschina/mt7628/openwrt-1/include/shell.sh: line 35: md5: command not found
Makefile:25: *** target file '/home/weeds/magicwifi/oschina/mt7628/openwrt-1/build_dir/host/flock/.preparedprint' has both : and :: entries.  Stop.
make[3]: Leaving directory '/home/weeds/magicwifi/oschina/mt7628/openwrt-1/tools/flock'
tools/Makefile:134: recipe for target 'tools/flock/compile' failed
make[2]: *** [tools/flock/compile] Error 2
make[2]: Leaving directory '/home/weeds/magicwifi/oschina/mt7628/openwrt-1'
tools/Makefile:133: recipe for target '/home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.tools_install_yynyynynynyyyyyyynyyyyyyyyyyyyynyyyyynnyynynyyyynnnyy' failed
make[1]: *** [/home/weeds/magicwifi/oschina/mt7628/openwrt-1/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.tools_install_yynyynynynyyyyyyynyyyyyyyyyyyyynyyyyynnyynynyyyynnnyy] Error 2
make[1]: Leaving directory '/home/weeds/magicwifi/oschina/mt7628/openwrt-1'
/home/weeds/magicwifi/oschina/mt7628/openwrt-1/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2

```

mmitch, 2016-12-27:
Just a wild guess: Do you have the `md5sum` binary installed?
It's in the `coreutils` package which is marked `essential` in Ubuntu 16.04, so it whould be rather strange if it was missing.

The relevant part in `include/shell.sh` seems to be this:
```bash
md5s() {
        cat "$@" | (
                md5sum 2>/dev/null ||
                md5
        ) | awk '{print $1}'
}
```
`md5` is an uncommon name on Linux systems, normally the binary is called `md5sum`.
In my build `staging_dir/host/bin/md5sum` is a symlink to the `/usr/bin/md5sum` binary on my host system, so it looks as if `md5sum` is taken from the host system.

thagabe, 2016-12-28:
@mmitch I do remember running into issues on Ubuntu from way back when. I have since settled on Debian as it seems the stability is something desired. The "fix" was for upstream to fix up the toolchain scripts and suddenly it would work again. Seems like a rather disappointing answer but Ubuntu seems to like to change names and packages willy-nilly. Right now Debian seems to be the best option.

thagabe, 2016-12-28:
@mmitch @qianguozheng Downloaded Ubuntu 16.04.1 LTS Server, install the packages, pulled the latest source, ./scripts/feeds {update, install} -a, make defconfig, make menuconfig, make -j# V=s, then it built just fine with my target mvebu.

thagabe, 2016-12-28:
can you guys download vm-player or virtualbox? Test with a fresh install maybe it's a system problem

thagabe, 2016-12-28:
#276

closed 2022-02-09 by aparcar