openwrt/openwrt #288: compile cmake error on ubuntu 16.04

fjr115, 2016-12-25:
kw@kw-PC:~$ uname -a
Linux kw-PC 4.4.0-22-generic #40-Ubuntu SMP Thu May 12 22:03:46 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
kw@kw-PC:~$ cat /etc/issue
Ubuntu 16.04 LTS \n \l

`
make[6]: Entering directory '/home/kw/OpenWrt/trunk/build_dir/host/cmake-3.6.1'
[  0%] Building C object Source/kwsys/CMakeFiles/cmsys.dir/ProcessWin32.c.obj
/home/kw/OpenWrt/trunk/build_dir/host/cmake-3.6.1/Source/kwsys/ProcessWin32.c:36:39: fatal error: windows.h: No such file or directory
compilation terminated.
Source/kwsys/CMakeFiles/cmsys.dir/build.make:62: recipe for target 'Source/kwsys/CMakeFiles/cmsys.dir/ProcessWin32.c.obj' failed
make[6]: *** [Source/kwsys/CMakeFiles/cmsys.dir/ProcessWin32.c.obj] Error 1
make[6]: Leaving directory '/home/kw/OpenWrt/trunk/build_dir/host/cmake-3.6.1'
CMakeFiles/Makefile2:1025: recipe for target 'Source/kwsys/CMakeFiles/cmsys.dir/all' failed
make[5]: *** [Source/kwsys/CMakeFiles/cmsys.dir/all] Error 2
make[5]: Leaving directory '/home/kw/OpenWrt/trunk/build_dir/host/cmake-3.6.1'
Makefile:160: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/kw/OpenWrt/trunk/build_dir/host/cmake-3.6.1'
Makefile:29: recipe for target '/home/kw/OpenWrt/trunk/build_dir/host/cmake-3.6.1/.built' failed
make[3]: *** [/home/kw/OpenWrt/trunk/build_dir/host/cmake-3.6.1/.built] Error 2
make[3]: Leaving directory '/home/kw/OpenWrt/trunk/tools/cmake'
tools/Makefile:134: recipe for target 'tools/cmake/compile' failed
make[2]: *** [tools/cmake/compile] Error 2
make[2]: Leaving directory '/home/kw/OpenWrt/trunk'
tools/Makefile:133: recipe for target '/home/kw/OpenWrt/trunk/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.tools_install_yynyynynynyyyyyyynyyyyyyyyyyyyynyyyyynnyynynyyyynnnyy' failed
make[1]: *** [/home/kw/OpenWrt/trunk/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.tools_install_yynyynynynyyyyyyynyyyyyyyyyyyyynyyyyynnyynynyyyynnnyy] Error 2
make[1]: Leaving directory '/home/kw/OpenWrt/trunk'
/home/kw/OpenWrt/trunk/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2
`

huzhifeng, 2017-01-14:
@fjr115 
I think it is related to GCC 5
You can try [200-gcc5_compat.patch](https://dev.openwrt.org/browser/trunk/tools/mkimage/patches/200-gcc5_compat.patch?rev=45703&order=name)
Go [Cannot build with gcc 5](https://forum.openwrt.org/viewtopic.php?id=56853) for detail

wzonelayer, 2017-01-26:
I have also encountered this problem. Then solve the problem by installing the wine-dev. 
Follow the steps below,

```
sudo apt-get update
sudo apt-get install wine1.6-dev
cd openwrt
make clean
make dirclean
make defconfig
make prereq
make V=s
```

closed 2022-02-09 by aparcar