openwrt/openwrt #308: compile error

callnothing, 2017-01-06:
install -m0755 /home/xxx/workspaces/xiaomi/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/net-snmp-5.7.3/agent/snmpd-static /home/xxx/workspaces/xiaomi/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/net-snmp-5.7.3/ipkg-ramips/snmpd-static/usr/sbin/snmpd
find /home/xxx/workspaces/xiaomi/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/net-snmp-5.7.3/ipkg-ramips/snmpd-static -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package snmpd-static is missing dependencies for the following libraries:
libnl-3.so.200

mirko, 2017-01-06:
In your case snmpd is linking against libnl, however libnl is not a stated dependency of snmpd.
According source:
`DEPENDS:=+PACKAGE_libnl-core:libnl-core`
it should be a dependency, if the package is selected.

My guess: You're having the library somewhere in your staging_dir (from a previous build), the libnl-package is not selected (so not stated as dependency) but snmpd automagically finds the libnl and links against it.

If that's the case, my proposal would to make snmpd (not) link against libnl explicitly.

What does `find /home/xxx/workspaces/xiaomi/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15 -name  libnl-3.so.200` show and does it solve the issue if you (temporarily) (re)move it?

stintel, 2017-01-15:
This is a problem in the packages feed and should be solved by 42aeb0630858f60f9d82f7b982af823d17ba45dc. Please close.

mirko, 2017-01-15:
Most likely fixed by above commit - please reopen if it still occurs.

closed 2017-01-15 by mirko