openwrt/openwrt #311: Change/wrong zone_wan_prerouting rules. in chaos calmer

WRTHacker, 2017-01-07:
Dear experts,

I'm using OpenWRT for years and try to migrate from Attitude Adjustment to Chaos Calmer (last version). After the first try was greatly failing, I setup from scratch in a test environment. However, there seems to be sometjing wrong with the zone_wan_prerouting rules.

Setup: WAN, LAN, DMZ

config interface 'dmz'
        option proto 'static'
        option ifname 'eth0.100'
        option ipaddr '192.168.100.1'
        option netmask '255.255.255.0'

For testing haproxy statistic interface is bount to 192.168.100.1:8888
Checked with lsof.

Port forwarding rule:

config redirect
        option target 'DNAT'
        option src 'wan'
        option proto 'tcp'
        option dest_port '8888'
        option dest 'dmz'
        option dest_ip '192.168.100.1'
        option name 'test port forward dmz'
        option src_dport '4488'

Resulting ipfilter rule (iptables-save)

A zone_wan_prerouting -p tcp -m tcp --dport 4488 -m comment --comment "test port forward dmz" -j REDIRECT --to-ports 8888

Same setup in attitude adjustment for port 443:

A zone_wan_prerouting -p tcp -m tcp --dport 443 -j DNAT --to-destination 192.168.5.1:443

So somethings goes really wrong in chaos calmer! There is no destination IP in the prerouting rule specified in the port forwarding ...

Or did I miss something, even debugging for days?

Any help would be appreciated?

Best regards

JW0914, 2017-01-28:
There were changes made to the firewall, but I can't recall when it affected CC (I use DD).... however since I use my router as a smart switch and Sophos as my wan facing router.  Have you tried bypassing the firewall config all together and inputting the rule in command form with the `firewall.user` script?

WRTHacker, 2017-02-05:
Hi, thanks for the reply. No, I don't check by firewall user. I#M not really an iptables expert ... but expect, that the rules correctly saved in /etc/config/firewall are implemented in the right way.
So the test should be really easy when saving an rule silimar to mine with a second DMZ subnet. Nowhere in the iptables rules the destination IP could be found ... so it could no work.
It would be very nict, it someone could have a look on it.
May thanks!

SrividhyaMH, 2017-06-06:
Hi, this issue looks interesting. Can you please tell me more on the test environment on which you observed the issue. Is it on a virtual machine you configured the virtual interfaces and configured port forwarding or on a real device. I am trying to simulate this on a openwrt virtual machine with virtual interfaces created on eth0.

WRTHacker, 2017-06-06:
Hello SrividhyaMH,

I use an identical hardware, like the production system TP-Link 4300. If it helps, I can zip the configuration out of /etc/config and send it to you. If I can support you, don't hesitate to contact me.

However, It could take some days, to make my test system accessible, cause I'm a little busy this time.

I'll be glad to hear from you. This issue is the one which prevents migration from V12.09 ...

Here comes the ifconfig output from the production system:

br-dmz    Link encap:Ethernet  HWaddr 90:F6:52:F6:0E:26
          inet addr:192.168.5.1  Bcast:192.168.5.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:16992428 errors:0 dropped:12904 overruns:0 frame:0
          TX packets:26386179 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:7950873047 (7.4 GiB)  TX bytes:28412328097 (26.4 GiB)

br-lan    Link encap:Ethernet  HWaddr 90:F6:52:F6:0E:26
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:35158425 errors:0 dropped:143669 overruns:0 frame:0
          TX packets:54405048 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:25246157870 (23.5 GiB)  TX bytes:52865776255 (49.2 GiB)

eth0      Link encap:Ethernet  HWaddr 90:F6:52:F6:0E:26
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:121044882 errors:0 dropped:8 overruns:49 frame:0
          TX packets:114467938 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2541011505 (2.3 GiB)  TX bytes:76383236 (72.8 MiB)
          Interrupt:4

eth0.1    Link encap:Ethernet  HWaddr 90:F6:52:F6:0E:26
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:33034070 errors:0 dropped:16265 overruns:0 frame:0
          TX packets:47601877 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:27074433877 (25.2 GiB)  TX bytes:42179283789 (39.2 GiB)

eth0.2    Link encap:Ethernet  HWaddr 90:F6:52:F6:0E:26
          inet addr:xx.xx.xxx.xx  Bcast:xx.xx.xxx.xx  Mask:255.255.255.252
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:72004286 errors:0 dropped:3252 overruns:0 frame:0
          TX packets:42074852 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:5
          RX bytes:72903099173 (67.8 GiB)  TX bytes:25579948037 (23.8 GiB)

eth0.5    Link encap:Ethernet  HWaddr 90:F6:52:F6:0E:26
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:16006464 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24791209 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:7758843663 (7.2 GiB)  TX bytes:26348560170 (24.5 GiB)

haprxy for example is bound to 192.168.5.1

Then the port forward looks like

config redirect
        option target 'DNAT'
        option src 'wan'
        option proto 'tcp'
        option src_dport '443'
        option dest_port '443'
        option dest 'dmz'
        option dest_ip '192.168.5.1'
        option name 'haProxy_https'

This gives in 12.09 the prerouting line mentioned in the first post.

SrividhyaMH, 2017-06-07:
Hi, Thanks for the reply. I will also try to repro the issue on virtual machine or any local setup available. It will be very helpful if you could also zip the configuration in /etc/config and send it to me.

SrividhyaMH, 2017-06-09:
Hi, From my analysis, I found that after Attitude Adjustment, under iptables new nat target extensions has been added which includes REDIRECT as one target. And the already existing DNAT target option is also still existing. I could find this commit when I checked the file history of openwrt/package/network/utils/iptables
in May 21, 2013. Link:[]( https://github.com/openwrt-mirror/openwrt/commit)/212b5e0e85576b41bde869da1203504d3fc6ec74

I checked in both Attitude Adjustment and Chaos Calmer. Please find the logs below.

**Chaos_Calmer**

root@OpenWrt:~# 
root@OpenWrt:~# iptables-save | grep DNAT
-A zone_wan_prerouting -p tcp -m tcp --dport 8888 -m comment --comment "!fw3: test port forward dmz" -j DNAT --to-destination 192.168.20.1:4488
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-ports 4488
iptables: No chain/target/match by that name.
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-destination 192.168.20.1:4448
iptables v1.4.21: unknown option "--to-destination"
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-ports 4488
iptables v1.4.21: unknown option "--to-ports"
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4448
iptables: No chain/target/match by that name.
root@OpenWrt:~# 

_root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT -help
iptables v1.4.21

REDIRECT target options:
 --to-ports <port>[-<port>]
				Port (range) to map to.
 [--random]

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT -help
iptables v1.4.21

DNAT target options:
 --to-destination [<ipaddr>[-<ipaddr>]][:port[-port]]
				Address to map destination to.
[--random] [--persistent]_

**Attitude_Adjustment**

root@OpenWrt:~# 
root@OpenWrt:~# iptables-save | grep DNAT
-A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4488 
root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-ports 4488
iptables v1.4.10: unknown option `--to-ports'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-destination 192.168.20.1:4448
iptables v1.4.10: unknown option `--to-destination'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-ports 4488
iptables v1.4.10: unknown option `--to-ports'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4448
iptables: No chain/target/match by that name.

_root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT -help
iptables v1.4.10
 //***No REDIRECT target option is displayed as we dont have that target ***//

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT -help
iptables v1.4.10

DNAT target options:
 --to-destination <ipaddr>[-<ipaddr>][:port-port]
				Address to map destination to.
[--random] [--persistent]_

So, when you specify with -j DNAT target option you can specify destination IP and when you specify with -j REDIRECT target option you need to specify the port. You can also verify the same in the man page of iptables-extensions. Link: [](http://ipset.netfilter.org/iptables-extensions.man.html)

SrividhyaMH, 2017-06-09:
Hi, From my analysis, I found that after Attitude Adjustment, under iptables new nat target extensions has been added which includes REDIRECT as one target. And the already existing DNAT target option is also still existing. I could find this commit when I checked the file history of openwrt/package/network/utils/iptables
in May 21, 2013. Link:[]( https://github.com/openwrt-mirror/openwrt/commit)/212b5e0e85576b41bde869da1203504d3fc6ec74

I checked in both Attitude Adjustment and Chaos Calmer. Please find the logs below.

**Chaos_Calmer**

root@OpenWrt:~# 
root@OpenWrt:~# iptables-save | grep DNAT
-A zone_wan_prerouting -p tcp -m tcp --dport 8888 -m comment --comment "!fw3: test port forward dmz" -j DNAT --to-destination 192.168.20.1:4488
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-ports 4488
iptables: No chain/target/match by that name.
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-destination 192.168.20.1:4448
iptables v1.4.21: unknown option "--to-destination"
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-ports 4488
iptables v1.4.21: unknown option "--to-ports"
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4448
iptables: No chain/target/match by that name.
root@OpenWrt:~# 

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT -help
iptables v1.4.21

REDIRECT target options:
 --to-ports <port>[-<port>]
				Port (range) to map to.
 [--random]

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT -help
iptables v1.4.21

DNAT target options:
 --to-destination [<ipaddr>[-<ipaddr>]][:port[-port]]
				Address to map destination to.
[--random] [--persistent]

**Attitude_Adjustment**

root@OpenWrt:~# 
root@OpenWrt:~# iptables-save | grep DNAT
-A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4488 
root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-ports 4488
iptables v1.4.10: unknown option `--to-ports'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-destination 192.168.20.1:4448
iptables v1.4.10: unknown option `--to-destination'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-ports 4488
iptables v1.4.10: unknown option `--to-ports'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4448
iptables: No chain/target/match by that name.

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT -help
iptables v1.4.10
 //***No REDIRECT target option is displayed as we dont have that target ***//

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT -help
iptables v1.4.10

DNAT target options:
 --to-destination <ipaddr>[-<ipaddr>][:port-port]
				Address to map destination to.
[--random] [--persistent]

So, when you specify with -j DNAT target option you can specify destination IP and when you specify with -j REDIRECT target option you need to specify the port. You can also verify the same in the man page of iptables-extensions. Link: [](http://ipset.netfilter.org/iptables-extensions.man.html)

SrividhyaMH, 2017-06-09:
Hi, From my analysis, I found that after Attitude Adjustment, under iptables new nat target extensions has been added which includes REDIRECT as one target. And the already existing DNAT target option is also still existing. I could find this commit when I checked the file history of openwrt/package/network/utils/iptables
in May 21, 2013. Link:[]( https://github.com/openwrt-mirror/openwrt/commit)/212b5e0e85576b41bde869da1203504d3fc6ec74

I checked in both Attitude Adjustment and Chaos Calmer. Please find the logs below.

**Chaos_Calmer**

root@OpenWrt:~# 
root@OpenWrt:~# iptables-save | grep DNAT
-A zone_wan_prerouting -p tcp -m tcp --dport 8888 -m comment --comment "!fw3: test port forward dmz" -j DNAT --to-destination 192.168.20.1:4488
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-ports 4488
iptables: No chain/target/match by that name.
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-destination 192.168.20.1:4448
iptables v1.4.21: unknown option "--to-destination"
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-ports 4488
iptables v1.4.21: unknown option "--to-ports"
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4448
iptables: No chain/target/match by that name.
root@OpenWrt:~# 

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT -help
iptables v1.4.21

REDIRECT target options:
 --to-ports <port>[-<port>]
				Port (range) to map to.
 [--random]

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT -help
iptables v1.4.21

DNAT target options:
 --to-destination [<ipaddr>[-<ipaddr>]][:port[-port]]
				Address to map destination to.
[--random] [--persistent]

**Attitude_Adjustment**

root@OpenWrt:~# 
root@OpenWrt:~# iptables-save | grep DNAT
-A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4488 
root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-ports 4488
iptables v1.4.10: unknown option `--to-ports'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT --to-destination 192.168.20.1:4448
iptables v1.4.10: unknown option `--to-destination'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-ports 4488
iptables v1.4.10: unknown option `--to-ports'
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# 
root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT --to-destination 192.168.20.1:4448
iptables: No chain/target/match by that name.

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j REDIRECT -help
iptables v1.4.10
 //***No REDIRECT target option is displayed as we dont have that target ***//

root@OpenWrt:~# iptables -A zone_wan_prerouting -p tcp -m tcp --dport 8888 -j DNAT -help
iptables v1.4.10

DNAT target options:
 --to-destination <ipaddr>[-<ipaddr>][:port-port]
				Address to map destination to.
[--random] [--persistent]

So, when you specify with -j DNAT target option you can specify destination IP and when you specify with -j REDIRECT target option you need to specify the port. You can also verify the same in the man page of iptables-extensions. Link: [](http://ipset.netfilter.org/iptables-extensions.man.html)

Hope, this resolves the issue you mentioned in this ticket.

SrividhyaMH, 2017-06-09:
Hi, From my analysis, I found that after Attitude Adjustment, under iptables new nat target extensions has been added which includes REDIRECT as one target. And the already existing DNAT target option is also still existing. I could find this commit when I checked the file history of openwrt/package/network/utils/iptables
in May 21, 2013. Link: https://github.com/openwrt-mirror/openwrt/commit)/212b5e0e85576b41bde869da1203504d3fc6ec74

I checked in both Attitude Adjustment and Chaos Calmer. 

In Chaos Calmer, you can append to a chain with -j DNAT target option which takes destination IP:port as parameter and as well you append to a chain with -j REDIRECT target option which takes port as parameter.

In Attitude Adjustment , you can append to a chain only with -j DNAT target option with destination IP:Port as REDIRECT target extension was added in later release.

You can check these in the log file attached. 
You can also verify the parameters taken by each target extension in the man page of iptables-extensions. Link: http://ipset.netfilter.org/iptables-extensions.man.html.

Hope, this resolves the issue you mentioned in this ticket.

[log .txt](https://github.com/openwrt/openwrt/files/1063654/log.txt)

WRTHacker, 2017-06-12:
Hi, thanks for the analysis. The problem is, that I configure the rules with luci GUI. And it seems that the ipfilter changes you mention are not correctly ported to luci. So when I enter exact the same parameters in the luci gui page, the port forward didn't work cause luci builds a wron rule out of it. So there should be an fix for it in luci, to prevent other users to run into this issue.

Are you able to reproduce this?

Many thanks

SrividhyaMH, 2017-06-13:
Basically, I analysed on a virtual machine, with only one interface eth0 configured as bridged adapter. So i created br-wan, br-lan and br-dmz on eth0 and configured port forwarding (via luci).

I observed two cases in Chaos Calmer, 

when I configured port forwarding (via luci) from wan->dmz the corresponding config populated in /etc/config/firewall is,
config redirect                                 
        option target 'DNAT'                    
        option src 'wan'                        
        option dest 'dmz'                       
        option proto 'tcp'                      
        option src_dport '4488'                 
        option dest_ip '192.168.5.1'            
        option dest_port '8888'                 
        option name 'testdmz'  

Case 1: When I created dmz interface without including any available network(via luci), the config populated in /etc/config/firewall is, 
config zone                                     
        option forward 'REJECT'                 
        option output 'ACCEPT'                  
        option name 'dmz'                       
        option input 'REJECT'                   
        option network ' '

The resulting ipfilter rule (using iptables-save) is,
-A zone_wan_prerouting -p tcp -m tcp --dport 4488 -m comment --comment testdmz -j DNAT --to-destination 192.168.5.1:8888

Case 2: When I created dmz interface including dmz network in covered networks(via luci), the config populated in /etc/config/firewall is, 
config zone                                     
        option forward 'REJECT'                 
        option output 'ACCEPT'                  
        option name 'dmz'                       
        option input 'REJECT'                   
        option network 'dmz'

The resulting ipfilter rule (using iptables-save) is,
-A zone_wan_prerouting -p tcp -m tcp --dport 4488 -m comment --comment testdmz -j REDIRECT --to-ports 8888

Also, for verification purpose, i configured  port forwarding (via luci) from wan->lan the corresponding config populated in /etc/config/firewall is , 
config redirect                                 
        option target 'DNAT'                    
        option src 'wan'                        
        option dest 'lan'                       
        option proto 'tcp'                      
        option src_dport '4488'                 
        option dest_ip '192.168.2.1'            
        option dest_port '8888'                 
        option name 'testdmznew'                
                                    
lan zone config in /etc/config/firewall is, 
config zone                                     
        option name 'lan'                       
        option input 'ACCEPT'                   
        option output 'ACCEPT'                  
        option forward 'ACCEPT'                 
        option network 'lan' 

The resulting ipfilter rule (using iptables-save) is,
-A zone_wan_prerouting -p tcp -m tcp --dport 4488 -m comment --comment testdmznew -j REDIRECT --to-ports 8888

In Openwrt wiki its specified, For DNAT if the dest_ip value matches the local ip addresses of the router, as shown in the ifconfig, then the rule is translated in a DNAT + input 'accept' rule. Otherwise it is a DNAT + forward rule.

Redirection is a specialized case of destination NAT. REDIRECT is equivalent to doing DNAT to the incoming interface.

Attached file contains the configs i used.

[CCConfigIpfilterRule.txt](https://github.com/openwrt/openwrt/files/1071169/CCConfigIpfilterRule.txt)
[image]

WRTHacker, 2017-06-15:
Hi SrividhyaMH,

thanks for the clarification. I will have a look into my test setup ... 

On the 12.09 system is was indeed 

config zone
        option name 'dmz'
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'REJECT'

However, it seems something in the setup of the chaos calmer when migriting goes wrong. At the moment, I can't identify why this happens, cause my first try was "migration" of the config. The second one "new setup" from scratch with luci ...

Will come back with further results ...

Best regards!

WRTHacker, 2017-06-15:
Hi,
just checked it. You got it!!! Great.

The problem was the "network" line ind the "dmz" zone definition:

zone
...
network "dmz" 

No the forward works as expected. However, it's absolutely not clear, how the line comes into the configuration. Will check it again, when migration the production system.

Is there an "switch" in luci to enable/disable the config option?

Many many thanks for clarification!

closed 2022-02-09 by aparcar