openwrt/openwrt #317: DNS push support for OpenVPN client

abbradar, 2017-01-08:
DNS servers can be pushed via OpenVPN; my aim is to make dnsmasq use upstream DNS server provided by local OpenVPN client when connection is established.

On OpenVPN side this can be done by using script which runs on interface up/down events -- OpenVPN provides it with pushes from the server. However I'm unsure how to properly add a DNS server.

On a conventional Linux distribution I'd use [resolvconf](https://github.com/masterkorp/openvpn-update-resolv-conf). IIUC DNS servers on OpenWrt are managed by netifd which populates `/tmp/resolv.conf.auto`. Can I ask netifd (possibly via ubus call) to add/remove a DNS server?

ViperRu, 2017-10-06:
I join the issue

msvalina, 2017-10-18:
The same questions we have

closed 2022-02-09 by aparcar