openwrt/openwrt #333: Firewall Skript Issue with 2 diff. zones

ThomasCr, 2017-01-12:
Hi,
today I discovered a bug in the firewall script. I have 2 interfaces which are listen on the same interface.

> config interface 'wan1'
>         option ifname 'eth0'
>         option _orig_ifname 'eth0'
>         option _orig_bridge 'false'
>         option proto 'pppoe'
>         option username '000000000000000000001@t-online.de'
>         option password '00000000000000'
>         option metric '1'
>         option mtu '1492'

and 

> config interface 'wan1_modem'
>         option proto 'static'
>         option ifname 'eth0'
>         option ipaddr '10.10.10.254'
>         option netmask '255.255.255.0'

and this two interfaces are seperated in to zones:

> config zone
>         option name 'wan'
>         option output 'ACCEPT'
>         option masq '1'
>         option mtu_fix '1'
>         option network 'wan1 wan1_v6 wan2 wan2_v6'
>         option input 'DROP'
>         option forward 'DROP'
> 
> config zone
>         option name 'modem'
>         option input 'DROP'
>         option forward 'DROP'
>         option output 'ACCEPT'
>         option network 'wan1_modem wan2_modem'
>         option masq '1'
> 

so... now I want to allow NTP on the zone 'modem':

> config rule
>         option target 'ACCEPT'
>         option proto 'udp'
>         option dest_port '123'
>         option src 'modem'
>         option name 'Allow-NTP'

which works all okay... the only problem is, that zone wan1 should be pppoe interface, but when the interface is offline, the iptables rules is bould to eth0 (not ppp interface) and grap all packets away:

    5     0 zone_wan_input  all  --  eth0 *       0.0.0.0/0            0.0.0.0/0
    0     0 zone_modem_input  all  --  eth0 *       0.0.0.0/0            0.0.0.0/0

The packets are never reach the zone_modem_input rule. Yes, we can say it is as expected (when you look deep inside). But I think it could be better, so when a pppoe interfaces is found in the firewall processing script, it renders the interface name every time to the final name like pppoe-wan1.

jow-, 2017-01-12:
There is no simple solution to that in the firewall rule builder. You have multiple options though:
1) in zone wan, replace `option network "wan1 ..."` with `option device "pppoe-wan1 pppoe-wan2"`
2) in zone modem, replace `option network "wan1_modem ..."` with `option subnet "wan1_modem ..."` and make sure it appears before zone wan in the config

amitprakash0801, 2017-06-05:
The above comment is true, so if in the zone wan if option network is replaced with pppoe-wan1 pppoe-wan2 then it is working, same for  zone modem replacing wan1_modem with wan1_modem as subnet.