openwrt/openwrt #346: updated netifd breaks ipv6 updates over ppp

timcoote, 2017-01-22:
Accessing my isp over pppoe, and using hnetd, I was finding that ipv6 addresses would become deprecated after about an hour. Bisecting trunk, the commit introducing the bug was: 0b394dd2f5133466ac0920d3ab4ca7f66c42ac98, which updated to netifd of 2016-12-09. With this patch included, an ipv6 renew message and reply for the prefix, can be seen on pppoe-<interface>, but ip -6 addr shows the  preferred_lft on interfaces drop to zero, and then the ip address becomes deprecated and unusable.

I haven't dug into trying to work out if the issue's due to the interaction between netifd and hnet or whether it happens with a stock build (as I don't have a suitable working configuration).

fwiw, hw: Linksys AC1200, Caiman build with hnet-full. I'm using an external modem to convert between adsl and pppoe; hnet integration looks like this:

config interface 'e0'
        option ifname 'eth0'
        option proto 'pppoe'
        option username '<isp username'
        option password '<password>'

config interface 'e0ext'
        option ifname '@e0'
        option proto 'hnet'
        option mode 'external'
        option _orig_ifname 'pppoe-e0'

timcoote, 2017-01-23:
After bisecting netifd using a test of  preferred_lft being refreshed for delegated IPv6 addresses after 30 mins, git claims:
git bisect bad
b8ef742bd04ebef324ae11aee56c6e1d2cb7e0ad is the first bad commit
commit b8ef742bd04ebef324ae11aee56c6e1d2cb7e0ad

presumably there's some change in the interaction with hnet (or something else in my system) as otherwise this issue would have occurred elsewhere.  Looking at the code and guessing, is it likely that this patch:
--- a/interface.c
+++ b/interface.c
@@ -693,7 +693,8 @@ interface_proto_event_cb(struct interface_proto_state *state, enum interface_pro
        switch (ev) {
        case IFPEV_UP:
                if (iface->state != IFS_SETUP) {
-                       interface_event(iface, IFEV_UPDATE);
+                       if (iface->state == IFS_UP && iface->updated)
+                               interface_event(iface, IFEV_UPDATE);
                        return;
                }

is incomplete for my configuration as pppoe-<interface> itself is not updated, whereas the prefix managed by hnet is updated, so the event needs to be propagated?

timcoote, 2017-01-24:
If I remove the && iface->updated, then the ipv6 prefix is renewed. But I don't know what other damage such a change makes.

dedeckeh, 2017-03-09:
I've delivered two patches on the LEDE devel mailing list (http://lists.infradead.org/pipermail/lede-dev/2017-March/006604.html and http://lists.infradead.org/pipermail/lede-dev/2017-March/006605.html) which will trigger interface ubus notifaction events when the IPv6 prefix and address lifetime changes.
I hope this fixes the issue; if not an issue needs to be raised on the hnetd package so the maintainer can investigate the issue

closed 2022-02-09 by aparcar