openwrt/openwrt #352: AR71XX: Traffic kernel -> switch0 fails on Routerboard RB493G

forsberg, 2017-01-24:
I recently bought a Routerboard RB493G with the intention of running OpenWRT. However, I'm having severe problems using half of its ethernet ports. 

The device has two switches, both of which it detects as being _AR8316 rev 1_. Switch1, which connects the rightmost 4 ports + WAN ports, and is connected to Linux kernel as eth1, works as a charm. Switch0, for the 4 leftmost ports, however has the following behaviour:

1. Package (dhcp request) sent to switch from computer on one of its ports is forwarded to Linux/OpenWRT.
2. Dhcpd replies. Tcpdump shows that package is sent back to eth0.
3. Reply never reaches computer.

Using swconfig I can see that switch port 0 (the one connected to CPU) has a RxBadByte counter that increases constantly, while the RxGoodByte counter is 0. On port connected to computer, RxBadByte stays 0 while RxGoodByte increase.

The switch in itself works fine, if I connect two different computers to two of its ports, traffic flows as normal.

Also, if I boot RouterOS, the built-in OS that comes with the switch, traffic can be bridged between the two switches. This leads me to believe that there's some minor problem in the kernel driver for the switch.

Attaching a .tar.gz with output from dmesg, swconfig and /etc/config/network (for some reason, github didn't like the files when I tried to upload them as they were)

[routerboard-config.tar.gz](https://github.com/openwrt/openwrt/files/728000/routerboard-config.tar.gz)

I've tried this with Chaos Chalmer (15.05.1) and also with 14.07.

I'm more than willing to help in debugging this by testing patches and building my own openwrt images.

smail84, 2017-02-08:
Confirm i have the same issue
and I dont understand if it worked in any version of openwrt
Same issue was in rb2011 and it was fixed with 
ath79_eth0_pll_data.pll_1000 = 0x6f000000 in mach-rb2011.c
But in mach-rb4xx.c it has now effect

betelgeux1, 2017-08-29:
Same here. This is the most modern iteration of the 493 series. It'd be nice to get this working.

closed 2022-02-09 by aparcar