openwrt/openwrt #363: AR71XX: trunk does not boot on RB493G

forsberg, 2017-01-30:
I'm trying to boot trunk from 2016-01-22  (specifically [openwrt-ar71xx-mikrotik-vmlinux-initramfs.elf](https://downloads.openwrt.org/snapshots/trunk/ar71xx/mikrotik/openwrt-ar71xx-mikrotik-vmlinux-initramfs.elf) on a Routerboard RB493G.

Unfortunately, serial console does not show up. The board replies to ping on its WAN port, but that's all. output on serial port as follows:

> Got IP address: 192.168.1.131
resolved mac address F0:DE:F1:53:61:88
Gateway: 192.168.1.10
transfer started .............................................. transfer ok, time=1.79s
setting up elf image... OK
jumping to kernel code
[    0.000000] Linux version 4.4.14 (buildbot@debian-amd64) (gcc version 5.3.0 (OpenWrt
 GCC 5.3.0 50082) ) #2 Sun Jan 22 21:02:00 UTC 2017
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Atheros AR7161 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] No valid device tree found, continuing without
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 650
24
[    0.000000] Kernel command line: parts=1 boot_part_size=4194304 gpio=4023 HZ=3400000
00 mem=256M kmac=6C:3B:6B:21:73:00 board=493G ver=3.24 hw_opt=00020004 boot=1 mlc=9 roo
tfstype=yaffs noinitrd
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 253232K/262144K available (3045K kernel code, 186K rwdata, 760K 
rodata, 2016K init, 202K bss, 8912K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:680.000MHz, DDR:340.000MHz, AHB:170.000MHz, Ref:40.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns:
 5621354254 ns
[    0.000008] sched_clock: 32 bits at 340MHz, resolution 2ns, wraps every 6316128254ns
[    0.007794] Calibrating delay loop... 452.19 BogoMIPS (lpj=2260992)
[    0.070514] pid_max: default: 32768 minimum: 301
[    0.075245] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.081859] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.091272] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_
ns: 19112604462750000 ns
[    0.102290] NET: Registered protocol family 16
[    0.108174] MIPS: machine is MikroTik RouterBOARD 493G
[    2.749662] registering PCI controller with io_map_base unset
[    2.763851] ------------[ cut here ]------------
[    2.768484] WARNING: CPU: 0 PID: 1 at drivers/clk/clk.c:730 clk_core_enable+0x34/0xb
0()
[    2.776463] Modules linked in:
[    2.779519] CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.14 #2
[    2.785244] Stack : 803d4868 00000000 00000000 80430000 8f830280 80420f63 803b828c 0
0000001
          806436fc 00000000 804482e8 80640000 80640000 800a69d8 803bd828 80420000
          00000003 00000000 803bbc30 8f81fc44 80640000 800a49a4 00400000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[    2.820987] Call Trace:
[    2.823431] [<80071bc8>] show_stack+0x50/0x84
[    2.827788] [<800817cc>] warn_slowpath_common+0xa0/0xd0
[    2.833003] [<80081880>] warn_slowpath_null+0x18/0x24
[    2.838042] [<80245ab4>] clk_core_enable+0x34/0xb0
[    2.842824] [<80246b6c>] clk_enable+0x2c/0x58
[    2.847188] [<80200058>] rb4xx_spi_probe+0xa0/0x1bc
[    2.852060] [<801f6380>] platform_drv_probe+0x40/0x7c
[    2.857100] [<801f49ec>] driver_probe_device+0x114/0x278
[    2.862399] [<801f4bcc>] __driver_attach+0x7c/0xb4
[    2.867197] [<801f2eec>] bus_for_each_dev+0x94/0xa4
[    2.872062] [<801f40f4>] bus_add_driver+0xec/0x204
[    2.876842] [<801f52cc>] driver_register+0xb0/0x104
[    2.881717] [<80069a38>] do_one_initcall+0x1f4/0x21c
[    2.886681] [<80448d60>] kernel_init_freeable+0x164/0x224
[    2.892077] [<8006503c>] kernel_init+0x14/0x118
[    2.896601] [<80060878>] ret_from_kernel_thread+0x14/0x1c
[    2.901981] 
[    2.903489] ---[ end trace 28f1b842c9ef9aef ]---
[    2.908162] rb4xx-spi: probe of rb4xx-spi failed with error -143
[    2.914671] PCI host bridge to bus 0000:00
[    2.918802] pci_bus 0000:00: root bus resource [mem 0x10000000-0x16ffffff]
[    2.925661] pci_bus 0000:00: root bus resource [io  0x0000]
[    2.931229] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    2.938002] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-f
f]
[    2.946848] clocksource: Switched to clocksource MIPS
[    2.952996] NET: Registered protocol family 2
[    2.958242] TCP established hash table entries: 2048 (order: 1, 8192 bytes)
[    2.965233] TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
[    2.971653] TCP: Hash tables configured (established 2048 bind 2048)
[    2.978105] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    2.983941] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    2.990475] NET: Registered protocol family 1
[    5.049755] futex hash table entries: 256 (order: -1, 3072 bytes)
[    5.068400] io scheduler noop registered
[    5.072326] io scheduler deadline registered (default)
[    5.077808] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    5.086600] console [ttyS0] disabled
[    5.110251] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 10625000) 
is a 16550A
`

That base_baud looks a bit suspicious. 

Could this be related to https://dev.openwrt.org/ticket/21773 or https://dev.openwrt.org/ticket/21857, but on Linux 4.4?

forsberg, 2017-01-30:
Here's what a boot looks like on 15.05:

>transfer started .............................................. transfer ok, time=1.79s
setting up elf image... OK
jumping to kernel code
[    0.000000] Linux version 3.18.45 (forsberg@eksem) (gcc version 4.8.3 (OpenWrt/Linar
o GCC 4.8-2014.04 r49389) ) #4 Sat Jan 14 18:53:37 CET 2017
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Atheros AR7161 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x0fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x0fffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x0fffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 650
24
[    0.000000] Kernel command line:  console=ttyS0,115200 parts=1 boot_part_size=419430
4 gpio=4023 HZ=340000000 mem=256M kmac=6C:3B:6B:21:73:00 board=493G ver=3.24 hw_opt=000
20004 boot=1 mlc=9 rootfstype=yaffs noinitrd
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 253308K/262144K available (2554K kernel code, 158K rwdata, 532K 
rodata, 2696K init, 189K bss, 8836K reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:680.000MHz, DDR:340.000MHz, AHB:170.000MHz, Ref:40.000MHz
[    0.000000] Calibrating delay loop... 452.19 BogoMIPS (lpj=2260992)
[    0.060000] pid_max: default: 32768 minimum: 301
[    0.060000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.080000] NET: Registered protocol family 16
[    0.080000] MIPS: machine is MikroTik RouterBOARD 493G
[    2.720000] registering PCI controller with io_map_base unset
[    2.730000] rb4xx-spi rb4xx-spi: master is unqueued, this is deprecated
[    2.740000] PCI host bridge to bus 0000:00
[    2.740000] pci_bus 0000:00: root bus resource [mem 0x10000000-0x16ffffff]
[    2.750000] pci_bus 0000:00: root bus resource [io  0x0000]
[    2.750000] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-f
f]
[    2.760000] Switched to clocksource MIPS
[    2.760000] NET: Registered protocol family 2
[    2.770000] TCP established hash table entries: 2048 (order: 1, 8192 bytes)
[    2.770000] TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
[    2.780000] TCP: Hash tables configured (established 2048 bind 2048)
[    2.790000] TCP: reno registered
[    2.790000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    2.790000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    2.800000] NET: Registered protocol family 1
[    3.160000] futex hash table entries: 256 (order: -1, 3072 bytes)
[    3.170000] msgmni has been set to 494
[    3.180000] io scheduler noop registered
[    3.180000] io scheduler deadline registered (default)
[    3.190000] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    3.200000] console [ttyS0] disabled
[    3.220000] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 10625000) 
is a 16550A
[    3.230000] console [ttyS0] enabled
[    3.230000] console [ttyS0] enabled
[... 59 lines cut ...]
[   11.010000] ip_tables: (C) 2000-2006 Netfilter Core Team
[   11.030000] nf_conntrack version 0.5.0 (4000 buckets, 16000 max)
[   11.060000] xt_time: kernel timezone is -0000
[   11.130000] PPP generic driver version 2.4.2
[   11.140000] NET: Registered protocol family 24
[   17.320000] device eth0 entered promiscuous mode
[   17.350000] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   17.400000] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   17.410000] device eth1.1 entered promiscuous mode
[   17.410000] device eth1 entered promiscuous mode
[   17.440000] IPv6: ADDRCONF(NETDEV_UP): eth1.2: link is not ready
[   18.490000] eth1: link up (1000Mbps/Full duplex)
[   18.490000] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   18.510000] br-lan: port 2(eth1.1) entered forwarding state
[   18.510000] br-lan: port 2(eth1.1) entered forwarding state
[   18.520000] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.2: link becomes ready
[   18.530000] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   19.120000] eth0: link up (1000Mbps/Full duplex)
[   19.120000] br-lan: port 1(eth0) entered forwarding state
[   19.130000] br-lan: port 1(eth0) entered forwarding state
[   20.490000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 5 is up
[   20.510000] br-lan: port 2(eth1.1) entered forwarding state
[   21.130000] br-lan: port 1(eth0) entered forwarding state

That base_baud number is apparently not so suspicious.

AlverGant, 2017-08-29:
Same problem here, tried to compile older openwrt versions to track down what commits may have broken things and what I found is:

- Chaos Calmer 15.05.1 runs fine
- LEDE does not support yaffs2 so no hope of doing a permanent installation on NAND
- OpenWRT trunk has SPI issues during boot ( rb4xx-spi: probe of rb4xx-spi failed with error -143)
- Going back to OpenWRT r49379 solves the SPI issue during boot but no console available
- Going backwards to r48626 and still no console (base_baud = 10625000)
- If we continue going backwards, on r47238 it's not possible to compile anymore as some download links from mediatek are broken.

forsberg, 2017-09-07:
Have not tried myself, but I found [this post](http://lists.infradead.org/pipermail/lede-dev/2017-April/007069.html) where a LEDE developer manages to install LEDE on RB493G by converting a squashfs image to UBI.

closed 2022-02-09 by aparcar