openwrt/openwrt #372: OpenWRT build fails due to OpenSSL

semihalf-majczak-lukasz, 2017-02-03:
OpenWRT build fails due to OpenSSL API changes,

Step to reproduce:
1) git clone https://github.com/openwrt/openwrt; cd openwrt
2) ./scripts/feeds update -a
3) ./scripts/feeds install -a
4) make defconfig
5) make -j4 V=s

.......
/home/lukasz/openwrt/build_dir/host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c: In function 'get_cert_chain':
make[3]: Entering directory '/home/lukasz/openwrt/tools/pkg-config'
/home/lukasz/openwrt/build_dir/host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c:2482:13: error: dereferencing pointer to incomplete type 'X509 {aka struct x509_st}'
     cinf = x->cert_info;
             ^~
/home/lukasz/openwrt/build_dir/host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c:2484:30: error: dereferencing pointer to incomplete type 'X509_CINF {aka struct x509_cinf_st}'
     j = asn1_object_dump(cinf->signature->algorithm, bufp, CERTBUFFERSIZE);
                              ^~
/home/lukasz/openwrt/build_dir/host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c:2510:20: error: dereferencing pointer to incomplete type 'EVP_PKEY {aka struct evp_pkey_st}'
       switch(pubkey->type) {
                    ^~

.....

/home/lukasz/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c: In function 'rsa_get_exponent':
/home/lukasz/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:279:21: error: dereferencing pointer to incomplete type 'RSA {aka struct rsa_st}'
  if (BN_num_bits(key->e) > 64)
                     ^~

hnyman, 2017-02-05:
That looks like the old cmake 3.6.1 is causing problems in Openwrt buildroot (likely depending on the OpenSSL version on the buildhost).

semihalf-majczak-lukasz, 2017-02-06:
Hi @hnyman hnyman, you are right, I have downgraded libopenssl at buildhost and everything moved forward. Thanks :)

closed 2017-02-06 by semihalf-majczak-lukasz

Oliv4945, 2017-04-07:
Hi, @semihalf-majczak-lukasz. I am new to OpenWrt and I have the same error, could you please explain your solution ? Whould I change my host openssl version or an openwrt config file ?
Thank you

semihalf-majczak-lukasz, 2017-04-07:
Hi,

In my case downgrading libssl-dev package solved the issue: sudo apt
install libssl-dev=1.0.1t-1+deb8u6.

Best regards,

Lukasz

BrokenPen, 2017-05-09:
@semihalf-majczak-lukasz 

E: Version '1.0.1t-1+deb8u6' for 'libssl-dev' was not found

I take you advice to down grade the libssl-dev but the 1.0.1version was not found, any other version I could try?  So I look for other version...

apt-cache showpkg libssl-dev

but only 1.1.0e-1 is availble ???

What can I do next?

Oliv4945, 2017-05-09:
Hi @BrokenPen,
I had the same issue so I finally applied [this patch](https://git.lede-project.org/?p=source.git;a=blob;f=tools/mkimage/patches/210-openssl-1.1.x-compat.patch;h=fa7c99f39b0a65f0d784473ca9b8fde836e4fa6e;hb=70b104f98c0657323b28fce140b73a94bf3eb756) to the source.

evasil, 2017-06-05:
Thank you, @Oliv4945 ,

that patch eliminates problem with:
`if (BN_num_bits(key->e) > 64)`

but not with:
```
cinf = x->cert_info;
...
j = asn1_object_dump(cinf->signature->algorithm, bufp, CERTBUFFERSIZE);
...
switch(pubkey->type) {
```

Oliv4945, 2017-06-05:
Sorry @evasil, it worked for me without anything else and I don't remember the erros :/

evasil, 2017-06-05:
Yeah, @Oliv4945, I was a bit wrong in my tests.

Those 3 lines actually thrown when building cmake-3.6.1. Fixed if make it manually with 
./configure --system-ssl

gileri, 2017-06-16:
@evasil  I can't find where you did apply the system-ssl flag.
```
% pwd
/home/gileri/dev/openwrt/build_dir/host/cmake-3.6.1

% ./configure --system-ssl
Unknown option: --system-ssl
```

wtlusvm, 2017-06-18:
Same issue here. But I've downgrade my system from Debian testing to stable, hope it will work.

stopostit, 2017-06-29:
Hi guys, same issues here. 
I'm using Archlinux so I have both openssl (default 1.1) and openssl1.0
I had to add the following lines to the openwrt/include/host-build.mk in order to link the good version: 
`HOST_CFLAGS += -I/usr/include/openssl-1.0/openssl
HOST_LDFLAGS += -L/usr/lib/openssl-1.0/`

@gileri try looking at ./bootstrap --help :) 
I used ./bootstrap --system-libs and the compilation went ok for cmake 3.6.1
However the following error remains (and more due to this first error);

`./md5.c:65:22: warning: 'struct md5_ctx' declared inside parameter list will not be visible outside of this definition or declaration
 md5_init_ctx (struct md5_ctx *ctx)
                      ^~~~~~~
./md5.c: In function 'md5_init_ctx':
./md5.c:67:6: error: dereferencing pointer to incomplete type 'struct md5_ctx'
   ctx->A = (md5_uint32) 0x67452301;
`
Has anyone got any idea on how to fix it ?

Lotus253, 2017-08-22:
with cmake version 3.9.0 the same bug