openwrt/openwrt #376: OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64 - build failed

imKota, 2017-02-05:
```
Installing wpad-mini (2016-06-15-1) to root...
Downloading file:packages/base/wpad-mini_2016-06-15-1_ramips.ipk.
Configuring libubox.
Configuring libuclient.
Configuring uclient-fetch.
Configuring opkg.
Configuring kmod-nls-base.
Configuring kmod-usb-core.
Configuring libubus.
Configuring libjson-c.
Configuring libblobmsg-json.
Configuring ubusd.
Configuring ubus.
Configuring libnl-tiny.
Configuring iw.
Configuring busybox.
Configuring libuci.
Configuring odhcpd.
Configuring kmod-usb-dwc2.
Configuring swconfig.
Configuring kmod-lib-crc-ccitt.
Configuring kmod-ledtrig-usbdev.
Configuring kmod-slhc.
Configuring kmod-ppp.
Configuring kmod-pppox.
Configuring kmod-pppoe.
Configuring kmod-nf-ipt.
Configuring kmod-ipt-core.
Configuring kmod-nf-conntrack.
Configuring kmod-ipt-conntrack.
Configuring jshn.
Configuring netifd.
Configuring libjson-script.
Configuring ubox.
Configuring procd.
Configuring jsonfilter.
Configuring usign.
Configuring fstools.
Configuring base-files.
Configuring kmod-nf-nat.
Configuring kmod-cfg80211.
Configuring hostapd-common.
Configuring kmod-mac80211.
Configuring kmod-lib-crc-itu-t.
Configuring kmod-rt2x00-lib.
Configuring kmod-rt2800-lib.
Configuring dnsmasq.
Configuring libxtables.
Configuring libip4tc.
Configuring libip6tc.
Configuring kmod-ipt-nat.
Configuring firewall.
Configuring kmod-nf-conntrack6.
Configuring kmod-nf-ipt6.
Configuring kmod-ip6tables.
Configuring odhcp6c.
Configuring uci.
Configuring wpad-mini.
Configuring kmod-eeprom-93cx6.
Configuring kmod-rt2x00-mmio.
Configuring kmod-rt2800-mmio.
Configuring dropbear.
Configuring kmod-ledtrig-timer.
Configuring mtd.
Configuring ppp.
Configuring kmod-leds-gpio.
Configuring kmod-gpio-button-hotplug.
Configuring logd.
Configuring iptables.
[... 68 lines cut ...]
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 60
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
dd if=/root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/linux-ramips_rt305x/root.squashfs of=/root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64/bin/ramips/openwrt-ramips-rt305x-root.squashfs bs=128k conv=sync
13+1 records in
14+0 records out
1835008 bytes (1.8 MB) copied, 0.00526095 s, 349 MB/s
cp /root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/linux-ramips_rt305x/vmlinux /root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/linux-ramips_rt305x/vmlinux-3g-6200n
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/linux-ramips_rt305x/linux-4.4.14/arch/mips/boot/dts -I/root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/linux-ramips_rt305x/linux-4.4.14/arch/mips/boot/dts/include -undef -D__DTS__  -o /root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/linux-ramips_rt305x/3G-6200N.dtb.tmp ../dts/3G-6200N.dts
bash: mipsel-openwrt-linux-musl-cpp: command not found
make[3]: *** [install] Error 127
make[3]: Leaving directory `/root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64/target/linux/ramips/image'
make[2]: *** [build_image] Error 2
make[2]: Leaving directory `/root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64'
make[1]: *** [_call_image] Error 2
make[1]: Leaving directory `/root/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64'
make: *** [image] Error 2
root@Ubuntu1404x64:~/OpenWrt-ImageBuilder-ramips-rt305x.Linux-x86_64# 
```

ppaulojr, 2017-03-20:
Same problem here

kissste, 2017-07-11:
the same problem here -"bash: mipsel-openwrt-linux-musl-cpp: command not found"

closed 2017-07-17 by imKota