openwrt/openwrt #394: Can't make with openssl 1.1.0d

zglzone, 2017-02-15:
issue：
`root/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c: In function 'rsa_get_exponent':
/root/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:279:21: error: dereferencing pointer to incomplete type 'RSA {aka struct rsa_st}'
  if (BN_num_bits(key->e) > 64)`

reubot, 2017-04-14:
Apparently this has been fixed in u-Boot 2017.03
https://lists.denx.de/pipermail/u-boot/2017-February/281142.html

simonszu, 2017-10-04:
Same Problem for me. Is it possible to update u-Boot?

skokkk, 2017-12-04:
Experiencing the same error here: 

> HOSTCC  tools/img2srec
> In file included from /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/include/compiler.h:46:0,
>                  from /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/include/libfdt_env.h:12,
>                  from <command-line>:0:
> /home/shock/Documents/Projects/openwrtnode/openwrt/staging_dir/host/include/byteswap.h:2:2: warning: #include_next is a GCC extension
>  #include_next <byteswap.h>
>   ^~~~~~~~~~~~
>   HOSTCC  tools/aisimage.o
>   HOSTCC  tools/atmelimage.o
>   HOSTCC  tools/common/bootm.o
>   HOSTCC  tools/common/image-fit.o
>   HOSTCC  tools/common/image-sig.o
>   HOSTCC  tools/common/image.o
>   HOSTCC  tools/default_image.o
>   HOSTCC  tools/dumpimage.o
>   HOSTCC  tools/fit_check_sign.o
>   HOSTCC  tools/fit_common.o
>   HOSTCC  tools/fit_image.o
>   HOSTCC  tools/fit_info.o
>   HOSTCC  tools/gpimage-common.o
>   HOSTCC  tools/gpimage.o
>   HOSTCC  tools/image-host.o
>   HOSTCC  tools/imagetool.o
>   HOSTCC  tools/imximage.o
>   HOSTCC  tools/kwbimage.o
>   HOSTCC  tools/lib/crc32.o
> In file included from /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/include/compiler.h:46:0,
>                  from /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/include/libfdt_env.h:12,
>                  from <command-line>:0:
> /home/shock/Documents/Projects/openwrtnode/openwrt/staging_dir/host/include/byteswap.h:2:2: warning: #include_next is a GCC extension
>  #include_next <byteswap.h>
>   ^~~~~~~~~~~~
>   HOSTCC  tools/lib/fdtdec.o
>   HOSTCC  tools/lib/fdtdec_common.o
>   HOSTCC  tools/lib/libfdt/fdt.o
>   HOSTCC  tools/lib/libfdt/fdt_ro.o
>   HOSTCC  tools/lib/libfdt/fdt_rw.o
>   HOSTCC  tools/lib/libfdt/fdt_strerror.o
>   HOSTCC  tools/lib/libfdt/fdt_wip.o
>   HOSTCC  tools/lib/md5.o
> In file included from /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/include/compiler.h:46:0,
>                  from /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/include/libfdt_env.h:12,
>                  from <command-line>:0:
> /home/shock/Documents/Projects/openwrtnode/openwrt/staging_dir/host/include/byteswap.h:2:2: warning: #include_next is a GCC extension
>  #include_next <byteswap.h>
>   ^~~~~~~~~~~~
>   HOSTCC  tools/lib/rsa/rsa-checksum.o
>   HOSTCC  tools/lib/rsa/rsa-sign.o
> In file included from tools/lib/rsa/rsa-sign.c:1:0:
> /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c: In function 'rsa_remove':
> /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:156:2: warning: 'ERR_remove_thread_state' is deprecated [-Wdeprecated-declarations]
>   ERR_remove_thread_state(NULL);
>   ^~~~~~~~~~~~~~~~~~~~~~~
> In file included from /usr/include/openssl/x509_vfy.h:20:0,
>                  from /usr/include/openssl/x509.h:309,
>                  from /usr/include/openssl/pem.h:17,
>                  from /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:13,
>                  from tools/lib/rsa/rsa-sign.c:1:
> /usr/include/openssl/err.h:247:1: note: declared here
>  DEPRECATEDIN_1_1_0(void ERR_remove_thread_state(void *))
>  ^
> In file included from tools/lib/rsa/rsa-sign.c:1:0:
> /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c: In function 'rsa_sign_with_key':
> /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:213:2: warning: implicit declaration of function 'EVP_MD_CTX_cleanup'; did you mean 'EVP_MD_CTX_create'? [-Wimplicit-function-declaration]
>   EVP_MD_CTX_cleanup(context);
>   ^~~~~~~~~~~~~~~~~~
>   EVP_MD_CTX_create
> /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c: In function 'rsa_get_exponent':
> /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:279:21: error: dereferencing pointer to incomplete type 'RSA {aka struct rsa_st}'
>   if (BN_num_bits(key->e) > 64)
>                      ^~
> make[5]: *** [scripts/Makefile.host:134: tools/lib/rsa/rsa-sign.o] Error 1
> make[4]: *** [Makefile:1195: tools-only] Error 2
> make[4]: Leaving directory '/home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10'
> make[3]: *** [Makefile:48: /home/shock/Documents/Projects/openwrtnode/openwrt/build_dir/host/u-boot-2014.10/.built] Error 2
> make[3]: Leaving directory '/home/shock/Documents/Projects/openwrtnode/openwrt/tools/mkimage'
> make[2]: *** [tools/Makefile:135: tools/mkimage/compile] Error 2
> make[2]: Leaving directory '/home/shock/Documents/Projects/openwrtnode/openwrt'
> make[1]: *** [tools/Makefile:133: /home/shock/Documents/Projects/openwrtnode/openwrt/staging_dir/target-mips_mips32_musl-1.1.16/stamp/.tools_install_yynyynynynyyyyyyynyyynyyyyyyyyynyyyyynnyynynnyyynnnyy] Error 2
> make[1]: Leaving directory '/home/shock/Documents/Projects/openwrtnode/openwrt'
> make: *** [/home/shock/Documents/Projects/openwrtnode/openwrt/include/toplevel.mk:194: world] Error 2
>

techge, 2018-01-04:
Hi,

this is indeed a pity. Anyway I created a patch-file for u-boot-2014-10 which one can save as tools/mkimage/patches/001-u-boot.patch

This should solve the problem until a more recent u-boot is included in openwrt...

[001-uboot.patch.txt](https://github.com/openwrt/openwrt/files/1603526/001-uboot.patch.txt)

But afterwards I ran into #509 anyway :(

Kind regards
techge

closed 2022-02-09 by aparcar