openwrt/openwrt #408: linux-3.18.45 is missing

Avadesign-David, 2017-03-02:
In branch "Chaos Calmer", the linux kernel is missing in all the download source.
The one solution is change dowload path to "https://kernel.org/pub/$dir" (replace the ftp.all.kernel.org)

Note:
The ftp.all.kernel.org DNS is unresolved, and linux-3.18.45 is not exist in http://mirror2.openwrt.org/sources and http://downloads.openwrt.org/sources

sled, 2017-03-07:
Easiest fix I have found is to just add another mirror i.e open `scripts/download.pl` and add

`push @mirrors, 'https://kernel.org/pub/linux/kernel/v3.x/';`

after this line: https://github.com/openwrt/openwrt/blob/master/scripts/download.pl#L228

riblo, 2017-03-08:
@sled @Avadesign-David @wigyori there's a fix, see #413