openwrt/openwrt #423: Ubiquiti UAP-LR v2 not flashing - Invalid version error??

2SI3NX, 2017-03-15:
When i try and flash openwrt for the first time using syswrapper.sh i am getting Invalid version 'BZ.ar7240.v6.0.0-OpenWrt-r42625'   when i check nohup.out
https://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/openwrt-ar71xx-generic-ubnt-unifi-squashfs-factory.bin

Is this no longer working for version 2 of the UAP-LR ?
Can this be fixed quite easily?

PS:
I also try flash with LEDE project
https://downloads.lede-project.org/releases/17.01.0/targets/ar71xx/generic/lede-17.01.0-r3205-59508e3-ar71xx-generic-ubnt-unifi-squashfs-factory.bin 
But i get the same error.

1lann, 2017-03-18:
You need to first downgrade to an old version of Ubiquiti's firmware, this is because the newer ones check for signatures to verify that it's an official firmware. Ubiquiti has removed the older firmware which does not check for signatures, so you need to search the web for one with a version less than 3.4.14. This website has more information: https://pwassi.privatedns.org/openwrt/unifiac/. If you trust me I've attached an older version of the firmware you can use to downgrade to.
[firmware.bin.zip](https://github.com/openwrt/openwrt/files/852195/firmware.bin.zip)

I've personally used this method on my own UAP-AC-LR and it works, I've successfully installed OpenWrt and LEDE.

2SI3NX, 2017-03-18:
thank you for your help!
I understand what you are saying, but have you seen on https://www.ubnt.com/download/unifi/unifi-ap/uaplr  that my UAP-LR-v2 ie. VERSION 2 have a look it has different firmware from UAP-LR. are you sure your firmware will be compatible with my UAP-LR v2 for downgrade?  because UAP-LR and UAP-LR-v2 are two different devices.
And as far as i can see if you click on 'Past firmware' on the above site, you will see that UAP-LR-v2 was only introduced from firmware version 3.7.29.
Does this mean that ive hit a dead end and that il have to wait for Openwrt developers to rewrite code for this newer device? :(

Tibyon, 2017-05-07:
Have you attempted to flash via tftp? I believe tftp may bypass the verification. Also I cannot tell you the reason, but on my Ubiquiti device, (Nanostation Loco M2) the standard CC builds would not work, but the CC images I compiled from source flashed properly.

2SI3NX, 2017-05-07:
@Tibyon - No i havent tried with tftp because on the site https://wiki.openwrt.org/toh/ubiquiti/unifi
the instructions say that installing the new firmware image should be by using these commands for later 3.x.x versions:
mv /tmp/openwrt-ar71xx-generic-ubnt-unifi-squashfs-factory.bin /tmp/fwupdate.bin
cd /tmp
nohup syswrapper.sh upgrade2

However i do see that under "TFTP Upgrade Process"
it says tftp -i 192.168.1.20 put openwrt-ar71xx-generic-ubnt-unifi-squashfs-factory.bin
Does this mean i can use the "TFTP Upgrade method" to flash the firmware over Ubiquiti firmware without bricking it?

Tibyon, 2017-05-07:
Hm, I am not certain. If it had been more recently that I flashed my device I  might be able to tell you exactly what to do. I can tell you that I did end up having to unbrick my device via serial. If you can access uboot, then you will have no real risk of a permanent brick but you will need a ttl serial cable. Is that something you have and are comfortable with?

Tibyon, 2017-05-09:
@2SI3NX Here's a topic that discusses the procedure to properly flash these devices with the new uboot. https://community.ubnt.com/t5/airOS-SDK-Custom-Development/5-6-2-TFTP-ERROR/td-p/1307614 

TFTP flash probably won't work for you because your device shipped with a bootloader that does firmware authentication (scummy behavior IMO) so you have to access uboot and start urescue with the -e -f parameters which is non authoritative and allows bootloader modification.  Then you TFTP over ethernet. So all you need is a USB TTL / USB Serial adapter and a bit of bravery. Be careful of the 3.3v pins on both the adapter and the device, since you don't want to short a serial pin. And like I said, better to compile your own Chaos Calmer build since the buildbot builds didn't work for me 100%. Contact me if you have any question, I just did this again on my Loco M2 after I accidentally bricked it again so the info is fresh in my mind.

closed 2022-02-09 by aparcar