openwrt/openwrt #435: mwan3 and firewall restart break routing

hunters14473, 2017-03-27:
Hello,
i build my own openwrt firmware for Archer C7 v2 and was getting the issue that an /etc/init.d/firewall restart break the routing and the system is no more able to go out on the internet. Both from the router itself and from clients on the LAN. I was thinking it can be something introduced by my configuration but i tried with a fresh installation with precompiled binaries 15.05.1 .
I installed the mwan3 and luci-mwan3 package.
Configured the network with 2 wan each with their own metric and the self interface with a fake address 10.144.73.144 as the HowTo say. When you go in the shell and perform a /etc/init.d/firewall restart the routing stops to work. I have to restart the mwan3 too to get it back.
It happen if you make modification to the firewall from the web interface too.
Is it possible to fix it in some way?
Thank you all
Bye
Marcello

psyborg55, 2017-05-10:
try to reload firewall instead of restart and see if the routing still works

hunters14473, 2017-05-25:
With reload the routing still work but I think will be better to fix the issue with reload
Thank you
Regards
Marcello

closed 2022-02-09 by aparcar