openwrt/openwrt #442: OpenWrt support for Tp-Link TL-WR1043ND v4

Markovskij, 2017-04-03:
How about adding support for tl-wr1043nd v4?

kishangondaliya, 2017-06-28:
If you can donate the router, we can add the support.

wigyori, 2017-06-28:
https://github.com/openwrt/openwrt/commit/df384cfeb6103dbe0140a60d8894ca8b0a41b6fd

Not sure who is "we".

kishangondaliya, 2017-06-28:
"we" means contributors like you. You are labeled as contributor, I'm not ;)
MI router 3C, tplink eap220 and BBB(stable one) is something I'll be adding soon.

Dlay2017, 2017-08-09:
Support is already there ? I am already using it ?

https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/openwrt-ar71xx-generic-tl-wr1043nd-v4-squashfs-factory.bin
https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/openwrt-ar71xx-generic-tl-wr1043nd-v4-squashfs-sysupgrade.bin

Have fun!

sajtalma, 2017-09-22:
Hello, thank you very much for the links, although i have an issue, maybe you can help me?
I got this file 
https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/openwrt-ar71xx-generic-tl-wr1043nd-v4-squashfs-factory.bin
and updated my router through the stock web interface.
Then over winscp i put the file from:
https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/openwrt-ar71xx-generic-tl-wr1043nd-v4-squashfs-sysupgrade.bin
somewhere on the router, i SSH-ed into it with the basic openwrt credentials and performed a sysupgrade -n with the sysupgrade file but he router is still booting into:
DESIGNATED DRIVER (Bleeding Edge, 50108)
does anyone have an idea on how to fix this?

Sorry i am new to the forums, any help would be appreciated and thanks for your input.

closed 2022-02-09 by aparcar