openwrt/openwrt #451: Build dependency: Please install the Subversion client ?

wds75, 2017-04-16:
ubuntu 16.04 lts
sudo apt-get install git-core build-essential libssl-dev libncurses5-dev unzip gawk subversion wine1.6-dev
make defconfig
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'zlib'... ok.
Checking 'libssl'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'md5sum'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'svn'... failed.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'openssl'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please install the Subversion client

ifding, 2017-04-16:
For Ubuntu16.04, compile OpenWrt:
$ sudo apt-get update
$ sudo apt-get install git-core build-essential libssl-dev libncurses5-dev unzip gawk subversion wine1.6-dev
$ git clone git://git.openwrt.org/openwrt.git
$ cd openwrt
$ make defconfig
$ make prereq
$ make menuconfig

wds75, 2017-04-16:
Thanks.
Why not:    git clone git://github.com/openwrt/openwrt.git

Lin-Buo-Ren, 2017-09-17:
Same problem here.  KDE neon 5.10(Based on Ubuntu 16.04)

The error message is located at `include/prereq-build.mk`, with the matching statement:

```
$(eval $(call SetupHostCommand,svn,Please install the Subversion client, \
	svn --version | grep Subversion))
```

However `svn --version | grep Subversion` does match in my system, with the following output:

```
Subversion is open source software, see http://subversion.apache.org/
```

I have no idea why the command failed, though

Lin-Buo-Ren, 2017-09-17:
The issue can be workarounded by commenting the above section.
Not sure if it affects the build, though.

closed 2022-02-09 by aparcar