openwrt/openwrt #453: WRT1900AC - deauthenticated due to inactivity

GRMrGecko, 2017-04-18:
I have an issue where when I upgrade my phone to lineage-14.1-20170313-nightly-angler-signed.zip the router starts to deauthenticate my phone 10 seconds after connection. My phone then re-connects, and the cycle continues. I have also noticed with the official ROM from Google, so I do not think its necessarily an issue introduced by Lineage changes. Here are the changes made to that version of Lineage https://download.lineageos.org/angler/changes/1489394255/

With all of that being said, I can fix the issue on my phone by going back to lineage-14.1-20170306-nightly-angler-signed.zip. The reason I am opening a ticket here is because I have seen elsewhere with this was reported for the WRT1900AC and people stating that it is a driver bug. Each instance I have seen this reported had people that were not active, so I want to be active regarding this issue.

I have compiled OpenWRT from git commit 871372c so that there is no doubt that I am running the latest. I am also reporting this bug to LineageOS just-in-case they decide to fix it by reverting some changes.

Log entries below:
```
Tue Apr 18 06:00:34 2017 daemon.info hostapd: wlan1: STA ac:cf:85:ca:13:1d IEEE 802.11: authenticated
Tue Apr 18 06:00:34 2017 daemon.info hostapd: wlan1: STA ac:cf:85:ca:13:1d IEEE 802.11: associated (aid 3)
Tue Apr 18 06:00:34 2017 daemon.info hostapd: wlan1: STA ac:cf:85:ca:13:1d WPA: pairwise key handshake completed (RSN)
Tue Apr 18 06:00:42 2017 daemon.info hostapd: wlan1: STA ac:cf:85:ca:13:1d IEEE 802.11: disassociated
Tue Apr 18 06:00:43 2017 daemon.info hostapd: wlan1: STA ac:cf:85:ca:13:1d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

Thank You

drkhosla, 2017-04-21:
I am facing same issue with Android 7 #446 My older phone able to connect but phone with Android 7 is unable to connect.

GRMrGecko, 2017-04-21:
@drkhosla It is good to have multiple confirmations on this. I hope we have someone who can assist us in debugging/finding a solution. I am good at coding, however how to debug an issue like this is something I will need assistance with.

orenre, 2017-04-22:
Getting this also on TP-Link Archer C7 v2, running on a recent OpenWRT snapshot, but after longer periods of time, with a mobile phone as well (Nexus 5x, running Android nougat) 

deauthenticated due to inactivity (timer DEAUTH/REMOVE)

steffanschlein, 2017-04-23:
We are experiencing the same issue on a variety of access points and client devices. The AP models are:
- D-Link DIR-825 rev. B1
- TP-Link TL-WDR4900 v1
- TP-LINK Archer C5
- TP-LINK Archer C7

So far we tried two builds of OpenWRT: r48532 (cc 15.05.1) and r49389. There does not seem to be a difference though.

Devices on the client side include various manufacturers including Apple (both mobile and desktop, probably starting with macOS 10.11), Samsung, LG. The period of time between the disconnects vaires from a few minutes to several hours. 

The dropouts happen both on the 2.4 GHz and 5 GHz network.

After the `deauth` log we usually see an `IEEE 802.11: authenticated` and `IEEE 802.11: associated (aid 2)` a few secconds later. The number after `aid` varies and we did not find the reason yet.

If anyone needs more logs I could provide some.

artturnip, 2017-05-07:
This issue also affects the TP-Link TL-WDR3600 v1. 
Version: Chaos Calmer 15.05.1

Affects Android phones, Windows and Apple laptops as well as an OpenWrt WDS Client. 

No pattern in time between disconnects. It can be many hours between disconnects or there can be 2 in 10 minutes.
 
**Logs:**
```
Sun May  7 17:37:42 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:24:15:ad IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun May  7 17:37:59 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:24:15:ad IEEE 802.11: authenticated
Sun May  7 17:37:59 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:24:15:ad IEEE 802.11: associated (aid 6)
Sun May  7 17:38:00 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:24:15:ad WPA: pairwise key handshake completed (RSN)
```

vedrancek, 2017-05-20:
Hello,

So I am not the only one getting crazy with this problem.

I am having the same issues after upgrading my Nexus 4 stock Android to Pure Nexus - 7.1.2-20170514 ROM. So Android Nougat.
I am getting the following message in the system log:
```
Sat May 20 23:50:59 2017 daemon.info hostapd: wlan0: STA 10:68:3f:7f:21:e0 IEEE 802.11: authenticated
Sat May 20 23:50:59 2017 daemon.info hostapd: wlan0: STA 10:68:3f:7f:21:e0 IEEE 802.11: associated (aid 1)
Sat May 20 23:51:08 2017 daemon.info hostapd: wlan0: STA 10:68:3f:7f:21:e0 IEEE 802.11: deauthenticated due to local deauth request
```
I cannot connect to unless I disable encryption.

My Router is a Buffalo WZR-HP-G300NH with OpenWrt Chaos Calmer 15.05.1 (r48532) installed. Prior to Chaos Calmer I had Barrier Breaker 14.07 installed. I hoped that the upgrade would do the trick.

I have a tablet with Android Marshmallow installed and a Nokia N9 (MeeGo 1.2), few laptops (Win 7, 10, Ubuntu) and have no problems connecting...

So this seems to be a problem with Android Nougat ?

Best regards,
Vedrancek

GRMrGecko, 2017-07-04:
I haven't had an issue since I had my device replaced by Google. So this issue may have been hardware related.

kunickiaj, 2017-07-21:
Been having this same issue on OpenWrt + HooToo TM-06 for awhile. Not sure why.

ptaking, 2017-10-25:
Confirm on TP-LINK 740n ver 4.23 Chaos Calmer 15.05.1
Affects Android phones, Windows computer. iPhone is not affected. Windows computer disconnect after 2 minutes connected and having network activity (not idling, I am playing games).

Log:
`Wed Oct 25 23:33:39 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d IEEE 802.11: disassociated
Wed Oct 25 23:33:40 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Oct 25 23:33:42 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d IEEE 802.11: authenticated
Wed Oct 25 23:33:42 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d IEEE 802.11: associated (aid 2)
Wed Oct 25 23:33:42 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d WPA: pairwise key handshake completed (RSN)
Wed Oct 25 23:33:42 2017 daemon.info dnsmasq-dhcp[1141]: DHCPREQUEST(br-lan) 192.168.0.135 1c:4b:d6:3d:4d:0d 
Wed Oct 25 23:33:42 2017 daemon.info dnsmasq-dhcp[1141]: DHCPACK(br-lan) 192.168.0.135 1c:4b:d6:3d:4d:0d DESKTOP-FGT8JNI
Wed Oct 25 23:33:42 2017 daemon.warn odhcpd[783]: DHCPV6 SOLICIT IA_NA from 00010001207b1bbb7446a03cd751 on br-lan: ok fdaa:5ff8:dd04::3a0/128 
Wed Oct 25 23:33:42 2017 daemon.info dnsmasq[1141]: read /etc/hosts - 1 addresses
Wed Oct 25 23:33:42 2017 daemon.info dnsmasq[1141]: read /tmp/hosts/odhcpd - 0 addresses
Wed Oct 25 23:33:42 2017 daemon.info dnsmasq[1141]: read /tmp/hosts/dhcp - 1 addresses
Wed Oct 25 23:33:42 2017 daemon.info dnsmasq-dhcp[1141]: read /etc/ethers - 0 addresses
Wed Oct 25 23:33:42 2017 daemon.warn odhcpd[783]: DHCPV6 SOLICIT IA_NA from 00010001207b1bbb7446a03cd751 on br-lan: ok fdaa:5ff8:dd04::3a0/128 
Wed Oct 25 23:33:43 2017 daemon.warn odhcpd[783]: DHCPV6 REQUEST IA_NA from 00010001207b1bbb7446a03cd751 on br-lan: ok fdaa:5ff8:dd04::3a0/128 
Wed Oct 25 23:33:43 2017 daemon.info dnsmasq[1141]: read /etc/hosts - 1 addresses
Wed Oct 25 23:33:43 2017 daemon.info dnsmasq[1141]: read /tmp/hosts/odhcpd - 1 addresses
Wed Oct 25 23:33:43 2017 daemon.info dnsmasq[1141]: read /tmp/hosts/dhcp - 1 addresses
Wed Oct 25 23:33:43 2017 daemon.info dnsmasq-dhcp[1141]: read /etc/ethers - 0 addresses
Wed Oct 25 23:37:35 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d IEEE 802.11: disassociated
Wed Oct 25 23:37:36 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Oct 25 23:37:38 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d IEEE 802.11: authenticated
Wed Oct 25 23:37:38 2017 daemon.info hostapd: wlan0: STA 1c:4b:d6:3d:4d:0d IEEE 802.11: associated (aid 2)`

ptaking, 2017-10-26:
You can try to disable this behavior by setting `disassoc_low_ack` to `0`.
`uci set wireless.@wifi-iface[0].disassoc_low_ack=0`
`uci commit wireless`
`wifi`

It's worked for me.

drkhosla, 2017-11-11:
I try to debug hostapd daemon and found that RADIUS server authenticate user but due to EAPOL timeout. The disconnect request is given to client interface.

````
wlo1: STA 8c:f5:a3:36:b5:b4 RADIUS: Received RADIUS packet matched with a pending request, round trip time 0.03 sec
RADIUS packet matching with station 8c:f5:a3:36:b5:b4
wlo1: STA 8c:f5:a3:36:b5:b4 IEEE 802.11: binding station to interface 'wlo1'
nl80211: wlo1[3]: set_sta_vlan(8c:f5:a3:36:b5:b4, ifname=wlo1[3], vlan_id=0)
MS-MPPE-Send-Key - hexdump(len=32): [REMOVED]
MS-MPPE-Recv-Key - hexdump(len=32): [REMOVED]
wlo1: STA 8c:f5:a3:36:b5:b4 IEEE 802.1X: old identity 'admin' updated with User-Name from Access-Accept 'admin'
wlo1: STA 8c:f5:a3:36:b5:b4 IEEE 802.1X: decapsulated EAP packet (code=3 id=107 len=4) from RADIUS server: EAP Success
EAP: EAP entering state SUCCESS2
wlo1: CTRL-EVENT-EAP-SUCCESS2 8c:f5:a3:36:b5:b4
IEEE 802.1X: 8c:f5:a3:36:b5:b4 BE_AUTH entering state SUCCESS
wlo1: STA 8c:f5:a3:36:b5:b4 IEEE 802.1X: Sending EAP Packet (identifier 107)
IEEE 802.1X: 8c:f5:a3:36:b5:b4 BE_AUTH entering state IDLE
WPA: 8c:f5:a3:36:b5:b4 WPA_PTK entering state INITPMK
WPA: PMK from EAPOL state machine (MSK len=64 PMK len=32)
WPA: 8c:f5:a3:36:b5:b4 WPA_PTK entering state PTKSTART
wlo1: STA 8c:f5:a3:36:b5:b4 WPA: sending 1/4 msg of 4-Way Handshake
RSN: Derive PMKID using HMAC-SHA-256
RSN: Derived PMKID - hexdump(len=16): 38 09 07 ad 4f f4 af 30 7f dd a9 c8 55 76 58 52
WPA: Send EAPOL(version=3 secure=0 mic=0 ack=1 install=0 pairwise=1 kde_len=22 keyidx=0 encr=0)
WPA: Replay Counter - hexdump(len=8): 00 00 00 00 00 00 00 01
WPA: Use EAPOL-Key timeout of 100 ms (retry counter 1)
wlo1: Event EAPOL_TX_STATUS (37) received
IEEE 802.1X: 8c:f5:a3:36:b5:b4 TX status - version=2 type=0 length=4 - ack=1
wlo1: Event EAPOL_TX_STATUS (37) received
IEEE 802.1X: 8c:f5:a3:36:b5:b4 TX status - version=2 type=0 length=4 - ack=1
wlo1: Event EAPOL_TX_STATUS (37) received
IEEE 802.1X: 8c:f5:a3:36:b5:b4 TX status - version=2 type=3 length=117 - ack=1
WPA: EAPOL-Key TX status for STA 8c:f5:a3:36:b5:b4 ack=1
**WPA: Increase initial EAPOL-Key 1/4 timeout by 1000 ms because of acknowledged frame**
nl80211: Event message available
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlo1
nl80211: MLME event 59 (NL80211_CMD_FRAME) on wlo1(6e:be:22:ee:30:e3) A1=6e:be:22:ee:30:e3 A2=8c:f5:a3:36:b5:b4
nl80211: MLME event frame - hexdump(len=26): a0 00 3a 01 6e be 22 ee 30 e3 8c f5 a3 36 b5 b4 6e be 22 ee 30 e3 d0 4d 08 00
nl80211: Frame event
nl80211: RX frame da=6e:be:22:ee:30:e3 sa=8c:f5:a3:36:b5:b4 bssid=6e:be:22:ee:30:e3 freq=2412 ssi_signal=-24 fc=0xa0 seq_ctrl=0x4dd0 stype=10 (WLAN_FC_STYPE_DISASSOC) len=26
wlo1: Event RX_MGMT (18) received
mgmt::disassoc
disassocation: STA=8c:f5:a3:36:b5:b4 reason_code=8
wlo1: STA 8c:f5:a3:36:b5:b4 WPA: event 2 notification
wpa_driver_nl80211_set_key: ifindex=3 (wlo1) alg=0 addr=0x7db500 key_idx=0 set_tx=1 seq_len=0 key_len=0
   addr=8c:f5:a3:36:b5:b4
WPA: 8c:f5:a3:36:b5:b4 WPA_PTK entering state DISCONNECTED
WPA: 8c:f5:a3:36:b5:b4 WPA_PTK entering state INITIALIZE
wpa_driver_nl80211_set_key: ifindex=3 (wlo1) alg=0 addr=0x7db500 key_idx=0 set_tx=1 seq_len=0 key_len=0
   addr=8c:f5:a3:36:b5:b4
wlo1: STA 8c:f5:a3:36:b5:b4 IEEE 802.11: disassociated
EAP: Server state machine removed
nl80211: sta_remove -> DEL_STATION wlo1 8c:f5:a3:36:b5:b4 --> 0 (Success)
wlo1: STA 8c:f5:a3:36:b5:b4 MLME: MLME-DISASSOCIATE.indication(8c:f5:a3:36:b5:b4, 8)
wlo1: STA 8c:f5:a3:36:b5:b4 MLME: MLME-DELETEKEYS.request(8c:f5:a3:36:b5:b4)
wpa_driver_nl80211_set_key: ifindex=3 (wlo1) alg=0 addr=0x7db500 key_idx=0 set_tx=1 seq_len=0 key_len=0
   addr=8c:f5:a3:36:b5:b4
```

metzenseifner, 2017-12-09:
I can attest to @ptaking, using `disassoc_low_ack = '0'` in `/etc/config/wireless` does make the station stay connected to the access point better. I had to add this to all wireless interfaces.

My routers:

Linksys 

- WRT1900ACS running LEDE Reboot 17.01.2 r3435-65eec8bd5f / LuCI lede-17.01 branch (git-17.152.82987-7f6fc16)

- 2x TP-Link TL-WR1043N/ND v1 running OpenWrt Chaos Calmer 15.05.1 / LuCI 15.05-149-g0d8bbd2 Release (git-15.363.78009-956be55)

x2souffxw, 2017-12-26:
I'm experiencing this bug on my tp-link archer c5 v1 with` OpenWrt Chaos Calmer 15.05.1 / LuCI 15.05-149-g0d8bbd2 Release (git-15.363.78009-956be55)`

My laptop is disconnecting even every 5-6 minutes!

I have set:
```
uci set wireless.@wifi-iface[0].disassoc_low_ack=0
uci set wireless.@wifi-iface[1].disassoc_low_ack=0
uci commit wireless
wifi
```
on both interfaces (2,4 and 5 GHz), however there is no change.

closed 2022-02-09 by aparcar