openwrt/openwrt #469: Iran Timezone for OpenWRT

peylight, 2017-05-14:
Hi

I have OpenWRT CHAOS CALMER (15.05.1, r48532) in my RaspberryPI
I can't see any time zone for my country (Iran)
So i decided to tell this manually way to add Iran timezone to luci:
1-  SSH to openwrt and open `/usr/lib/lua/luci/sys/zoneinfo/tzdata.lua` by a text editor like vi
2- After this line:
`        { 'Asia/Tbilisi', 'GET-4' },                          `
Add this:
`        { 'Asia/Tehran', 'IRST-3:30IRDT,J80/0,J264/0' },      `
2- Then open `/usr/lib/lua/luci/sys/zoneinfo/tzoffset.lua` by text editor and add this lines into the OFFSET brackets {}
`        irst  =  12600, -- IRST`
`        irdt  =  16200, -- IRDT`

Best Regards,
peylight

amitprakash0801, 2017-06-06:
Trying to reproduce and resolve the issue, will update

peylight, 2017-06-06:
@amitprakash0801 
Thank you for your answer. but actually the problem should be from openwrt / luci for web interface.
I check the openwrt / luci but there are "Asia/Tehran" : 
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/sys/zoneinfo/tzdata.lua
But why there aren't in the image of Raspberrypi?
Who make the images? and why delete the Iran timezone?

amitprakash0801, 2017-06-12:
Hello,
 I tried to reproduce the issue on latest build and same problem i am also facing, country zone Iran is missing. I checked the luci web interface, in build itself some thing has went wrong and required country zone is not coming as u also mentioned. I am just checking the team responsible for making the build.

Regards,
Amit Prakash

closed 2022-02-09 by aparcar