openwrt/openwrt #472: TL-WR1043ND-v4 LAN/WAN stops working

lacamester, 2017-05-17:
Model	TP-Link TL-WR1043N/ND v4
Firmware Version	OpenWrt Designated Driver 50104 / LuCI Master (git-17.130.58552-d04f667) 
Kernel Version	4.4.14

Interface freeze seems to occur randomly, but may depend on the traffic and number of connections handled by the device. The LAN/WAN phy links are still establised, but no traffic is forwarded on the copper ports. The device is still accessible from wifi, but the logs don't show anything except failed pppoe connection requests after that. The network functionality can be restored after issuing "/etc/init.d/network restart".
The factory default TP-Link firmware has the same issue!

Several people observed the bug in OpenWRT forks: LEDE, and GLUON

https://forum.lede-project.org/t/lan-stops-working-every-now-and-then/2648/3
https://github.com/freifunk-gluon/gluon/issues/1101

cipiricus, 2017-08-28:
If you have same issue with default firmware that means your router can have hardware issue

danielpetre, 2017-08-29:
stumbled upon the same issue. LEDE compiled from source. Was making a speedtest and suddenly router froze. In my ssh logread -f i only saw that pppoe-wan dropped and nothing else.

jpeter01, 2017-10-02:
Hi! Exactly the same problem. I think this is software related, because if I used it with the stock firmware with default settings, the froze sometimes, but just rarely. After flashing it with OpenWrt (and LEDE too), it freezes every time, when the network load gets high. It is always reproducible with high speed torrent usage (with many connections, ~15+ connections and 7+ MB/sec should do it).

lacamester, 2017-10-02:
Hi! 
The thing is, since I moved to the latest TP-link firmware nicknamed "Portugal" those freezes occure very rarely (1-2 times in 6 months); so I can live with that.
The problem could be sw. related (I heard tp-link uses almost the same kernel and modules). If it's hw. related, then the whole series could be "defective by design", as the replacement units some people get suffer from the same sympthoms as I heard...

masterhapero, 2017-10-08:
Had similar issues with OpenWRT AA and BB versions on TL-WR703N. Tried different versions and stuff. Now I am trying to disable serial console, as it might throttle the kernel when outputting something over serial just 115.2kbps. Try add kernel.printk = 0 to /etc/sysctl.conf as the instructions in below document says. This should disable console output over serial, unless you are debugging or know you use this, you don't need this.

Appriciate feedback if this happens to help for you.

Instructions for terminating console: https://wiki.openwrt.org/doc/recipes/terminate.console.on.serial

jpeter01, 2017-10-10:
Thank you! I tried it with disabled serial console, but it didn't helped, so I moved back to the factory firmware from TP-Link. Now it is stable with a 100Mb/sec connection. I will test it later with a 1000 Mb/sec connection and high load NAT.

kindacute, 2017-10-18:
Hi all,
I have exactly the same problem, updated firmware to OpenWrt Designated Driver 50130 and my wi-fi goes down from time to time. I think it happens more often during high load (youtube/download). No entries in kernel.log, dmesg shows no new entries. Previously my 1043 was working on some old openwrt firmware (backfire? ) and it worked for about 3 years without any issues. Maybe I can provide some debug information? Let me know if I can help.
[image]

rotanid, 2017-10-19:
@kindacute if your router worked for years, it can't be revision v4 and you may have a different issue than this one

kindacute, 2017-10-19:
@rotanid Thank you for you comment. You are totally right and I missed revision in topic name. I have version 1 but problem is the same. I will full name of my device a bit later.

rotanid, 2017-10-19:
@kindacute as far as i know the revision 1 had some serious bugs in the hardware - maybe the new rev4 has similar issues

kindacute, 2017-10-20:
@rotanid Sir, I have tl-wr1043nd ver. 1.8. Possibly it has lot of bugs but it worked fine for a years and works fine on 15.05.1. The problem observed only on trunk version

jpeter01, 2017-10-20:
@rotanid It doesn't seems to be a hardware bug (I'm talking from v4), because after I went back to the factory firmware from TP-Link, the freezing of the copper ports are gone (I tested it with heavy load). And when I was using OpenWrt DD, the freezing occurred every time as the router got a heavy load (it took less than a minute).
I also tested another router (Asus WL-500g Premium v2) with OpenWRT CC, (because the 1043 v4 was useless), and it produced the same, as it got a high load, it froze (I dind't checked, if only the copper ports) in less than a minute. I went back to the factory firmware from Asus, and the router worked flawless under high load. 2 different type of router are producing the same kind of failure with OpenWrt, and are working fine with the factory firmware, that doesn't seems like a hardware bug.

Dorwido, 2017-10-21:
I did setup yesterday evening the revsion 4 router, installed straight LEDE on it and only changed the lan ip to 192.168.0.2 and setup wlan to match the old one so devices can login without changing password.
The router only 1 lan cable is connected which is conntected to a fritz box 6360 cable with unitymedia branding for the internet.
Only 3 devices use the wlan of the router, mobile phone (android 8.0), amazon dot echo revision 2 and newest amazon firetv stick.
Was then watching something with the firetv stick and pressed at some point pause and when I came back and pressed play it didnt work, figured then restarting router helped to make it work again,
This happenend then again the exact same way with pressing pause etc, after I gave up for the evening.

Searched then today for the problem and found this post and its like for the other the wlan works, can connect to router over it etc, but the lan ports are dead cant connect to the other router nor other devices connected via cable.

Will try out some stuff to see if I can figure more out which could be helpfull.

lacamester, 2017-10-21:
After reading that many people have the same problem with that particular router I was thinking about how we can debug that?
My questions are:
- Is there a "debug" kernel/eth driver/module which logs more events? (as the default one does not show anything)
- Can we dump/view the state of the eth. interface/copper ports after the problem occurs?

What do you think?

rotanid, 2017-10-21:
for all LEDE-Users, please use the LEDE bugtracker for this issue:
https://bugs.lede-project.org/index.php?do=details&task_id=762

closed 2022-02-09 by aparcar