openwrt/openwrt #484: jshn parsing is broken

bernd-wechner, 2017-05-29:
[status.zip](https://github.com/openwrt/openwrt/files/1035849/status.zip)

I gave up and went with python for this because jshn just can't parse the output of devstatus correctly somehow. A surprise and I cast a glance at the C source but couldn't spot anything and would rather work around it for my need than attempt a fix. So will log it here for diagnosis.

I'll attach three files (zipped together) to illustrate. They are:

`status.raw - produced by: devstatus eth1 > status.raw`

This is our input and how it's generated isn't really important as the parsing of it is what is broken.

`status.var - produced with: tmp="$(cat status.raw)"; echo $tmp > status.var`

and finally:

`status.parsed - produced with: jshn -r "$tmp"`

Now mostly the parsing is fine actually and the issue shows up in the one value I was looking at oddly enough, which is the rx_bytes element of the statistics element. Check it out in the attached files:

```
status.raw: 5140995254
status.var: 5140995254
status.parsed: 2147483647
```

Go figure. I cant tell wgat version of jshn I have but if it helps:

```
# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='Chaos Calmer'
DISTRIB_REVISION='r47055'
DISTRIB_CODENAME='omnia'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_DESCRIPTION='OpenWrt omnia 15.05'
DISTRIB_TAINTS='busybox'
```

PS - Trac is a braindead. I tried filing this here:

https://dev.openwrt.org/newticket

but an hour of effort (and I'm only still here bothering with this because I have a 6 week old infant asleep on my lap that will rouse and be blaring the moment I get up and I really wanted I guess to help openwrt be even greater!) later and not for love or money trying with numerous usernames, proxies, from another IP and more could I convince Trac I was not spam. It has an insane and counterproductive hair-trigger. Thankfully I found the wiki:

https://dev.openwrt.org/wiki

suggesting issues can be filed here too!

jow-, 2017-05-29:
The value `2147483647` is exactly `0x7FFFFFFF`, the highest value that can be represented by a signed 32bit integer. The `jshn` utility is not suitable to process 64bit values (such as counters) with full precision on 32bit platforms.

You can try `jsonfilter` to extact single values: `devstatus eth1 | jsonfilter -e '@.statistics.rx_bytes'` but you'll need a version >= 2016-06-01 to properly handle 64bit values.

bernd-wechner, 2017-05-29:
Thanks for the observation and tips. I had in fact suspected it was a clipping issue much like that but led myself astray when I checked the TX values:

Raw:494397486
Var:494397486
Parsed: 494397486

And given they seemed about the same size, concluded it might not be clipping. Only with your astute observation that 2147483647 is in fact max of a signed 32 bit int did I double check and notice of course the tx values are an order of magnitude (one digit) lower and fall beneath the clipping value. 

To be honest, I still think jshn is broken in this space as I can't see a reason for it to interpret the values at all, but that may be my blindness to other applications. I notice for example that it converts true to 1 and I didn't like that one jot, but find the python json module doesn't, it just passes it as text without interpretation.

Given jshn reads text and produces text I am left wondering what application demanded interpreting the values except that its output does include types for some application i'm clearly not familiar with. Still my sense of pedantry perhaps thinks jshn would do well with two features in this space:

1) bombing with overflows by default rather than clipping and a command line parameter to explicitly request clipping. 

2) a command line parameter to request text handling only. 

But then I admit I have looked and experimented and find I still don't know what -n -i do, and Iw as just using jshn because it's used by jshn.sh which was used in the example at bottom of this page:

https://wiki.openwrt.org/doc/techref/ubus

But in all honest given python-light is native on my openwrt router (and readily available for openwrt) I'm not sure what advantage sh processing has anyhow and I met my immediate need with this:

```
#!/usr/bin/python
import json
import subprocess

p=2 # Decimal precision for humanized byte counts

def byte_fmt(num, precision=1, suffix='B'):
    for unit in ['','Ki','Mi','Gi','Ti','Pi','Ei','Zi']:
        if abs(num) < 1024.0:
            return ("%3."+str(precision)+"f %s%s") % (num, unit, suffix)
        num /= 1024.0
    return ("%."+str(precision)+"f %s%s") % (num, 'Yi', suffix)

wan_devices = subprocess.check_output(["uci", "get", "network.wan.ifname"]).strip().split()

print "Interface: wan"

for dev in wan_devices:
    wan_status = json.loads(subprocess.check_output(["devstatus", dev]))

    if len(wan_devices) > 1:
        print

    print "Device: %s" % dev
    print "Up: %s" % wan_status['up']
    print "Carrier: %s" % wan_status['carrier']
    print "Speed: %s" % wan_status['speed']
    print "Received bytes: ", byte_fmt(wan_status['statistics']['rx_bytes'], precision=p)
    print "Transmitted bytes: ", byte_fmt(wan_status['statistics']['tx_bytes'], precision=p)

```
and in all honest all I have left till I'm happy is to work out what a speed of 1000F means :-)

closed 2022-02-09 by aparcar