openwrt/openwrt #485: OpenSSL version is outdated, how to upgrade to OpenSSL 1.1.0?

PearCar, 2017-05-30:
The latest Openssl package for Openwrt is for Openssl 1.0.2h, as you can see here: https://github.com/openwrt-mirror/openwrt/tree/master/package/libs/openssl

OpenSSL 1.1.0 has a lot of changes compared to OpenSSL 1.0.2, so it isn't possible to make a few changes to the Makefile in order to compile OpenSSL 1.1.0 for OpenWrt.

How can I install OpenSSL 1.1.0 on an OpenWRT router?

wigyori, 2017-05-30:
If you can, please hold on until the merge with LEDE is done (in the next few weeks), I assume the base packages will get a nice upgrade. Currently LEDE also has 1.0.2k in it.

If you're looking for bugfixes and/or can't wait, I'll upgrade the openssl package to the latest 1.0.2l, but not to 1.1.x just yet. If you explicitly need 1.1.0, I suggest to try building it in your own tree - can help to set it up.

PearCar, 2017-05-30:
@wigyori I specially need OpenSSL 1.1.x for things such as support for [X25519](https://www.openssl.org/news/openssl-1.1.0-notes.html) in OpenSSL 1.1.x 
Could you please help me to set up my own tree for OpenSSL 1.1.x?
Since there are large amount of code changes in and the Openwrt patch files (e.g. change perl path, disable engines, change libeay.num, disable doc, etc., etc.) for OpenSSL 1.0.2 are referring to lines that no longer exist in OpenSSl 1.1.x, so these patchfiles have to be removed?
 The makefile would cause errors such as "target linux-mips-openwrt doesn't exist!" The Openwrt-SDK compiled fine for other Openwrt packages.

And other errors, such as:

> ./libcrypto.so: warning: gethostbyname is obsolescent, use getnameinfo() instead.
> ./libcrypto.so: undefined reference to `getcontext'
> ./libcrypto.so: undefined reference to `setcontext'
> ./libcrypto.so: undefined reference to `makecontext'
> collect2: error: ld returned 1 exit status
> make[6]: *** [link_app.gnu] Error 1
>

PearCar, 2017-05-30:
@wigyori @diizzyy @ffainelli @hauke @nbd168 @mkrkn @neheb I specially need OpenSSL 1.1.x for things such as support for [X25519](https://www.openssl.org/news/openssl-1.1.0-notes.html) in OpenSSL 1.1.x 
Could you please help me to set up my own tree for OpenSSL 1.1.x?
Since there are large amount of code changes and the Openwrt patch files (e.g. change perl path, disable engines, change libeay.num, disable doc, etc., etc.) for OpenSSL 1.0.2 are referring to lines that no longer exist in OpenSSL 1.1.x, so these patchfiles have to be removed?
 The makefile would cause errors such as "target linux-mips-openwrt doesn't exist!" The Openwrt-SDK compiled fine for other Openwrt packages.

And other errors, such as:

> ./libcrypto.so: warning: gethostbyname is obsolescent, use getnameinfo() instead.
> ./libcrypto.so: undefined reference to `getcontext'
> ./libcrypto.so: undefined reference to `setcontext' [...]

zorun, 2017-10-26:
FYI I have a working port of openssl 1.1.0 in LEDE, I will submit it to LEDE in the next few days (when it builds on all (or most) targets).

closed 2022-02-09 by aparcar