openwrt/openwrt #495: Ath9k SPI MMC microSD access can freeze machine for many seconds at a time

gardners, 2017-06-15:
The SPI MMC kernel driver for OpenWRT CC at least, is problematic on an Ath9k device.

Our specific situation is using a Core Domino module, with a microSD card connected to the SPI bus.  Chip select for the microSD card appears to be working fine, in that the module boots, and the SD card mounts, and can be accessed, albeit slowly, which is a common problem with SD cards via SPI interface.  (The CS line is actually obtained by inverting the CS line for the SPI flash that the Ath9k boots from).  This is for the Serval Mesh Extender.  Source is at: 
https://github.com/servalproject/openwrt/tree/MeshExtender2.0

The key commits for incorporating microSD card using the SPI MMC driver are as follows, and should work on other hardware:

1cfb77ccaf4fd6a5ef345ba599534a5ae26114df
3133e6ed935bc4d7705f72e639519e39cafa711d
034daf1f4c63c90b079e40da3f8529288ddc1a79

The problem that we see is that if the SD card is being heavily accessed, the Ath9k system becomes totally unresponsive, sometimes for seconds at a time.  

My suspicion is that the SPI MMC driver attempts to process its entire IO queue all at once with interrupts disabled, without regard for how long this may take.  Given typical SD card write speeds of <100KB/sec over SPI, this can easily extend to several seconds of delay. 

Having the kernel unresponsive for seconds at a time on a regular basis is of course suboptimal, and causes the wifi AP to drop out, for example.

A solution to this problem would be very welcome.  

My uneducated guess is that briefly yielding after each write to the microSD card would likely solve the problem.

Deoptim, 2017-08-24:
Hello.
And in the log you have a recording error?

First orange massage: https://wiki.openwrt.org/toh/tp-link/tl-mr3420/deep.mmc.hack
Conflict with these patches.

gardners, 2017-08-24:
There are no write errors -- it just takes a long time to write, during which time the CPU is completely non-responsive.

I don't recall seeing the information in the orange box on the deep.mmc.hack page when I implemented this.  I can certainly try removing those patches and rebuilding.  I'll let you know how it performs.

A somewhat related problem I am seeing is that uboot can't flash properly when the SD card is present, even though it's CS is deasserted.  

But one thing at a time: I'll delete those patches and try building again, and let you know the result.

gardners, 2017-08-25:
With those patches deleted, I now get the following kernel panic:

                                                                                
  _____     ____    __  __   _____   _   _    ____                              
 |  __ \   / __ \  |  \/  | |_   _| | \ | |  / __ \                             
 | |  | | | |  | | | \  / |   | |   |  \| | | |  | |                            
 | |  | | | |  | | | |\/| |   | |   | . ` | | |  | |                            
 | |__| | | |__| | | |  | |  _| |_  | |\  | | |__| |                            
 |_____/   \____/  |_|  |_| |_____| |_| \_|  \____/                             
                                                                                
Autobooting in: 2 s (type 'gl' to run U-Boot console)                           
                                                                                
Device calibrated. Booting ...                                                  
Booting image at: 0x9F050000                                                    
                                                                                
   Image name:   MIPS OpenWrt Linux-3.18.45                                     
   Created:      2017-08-25   1:12:28 UTC                                       
   Image type:   MIPS Linux Kernel Image (lzma compressed)                      
   Data size:    1160872 Bytes = 1.1 MB                                         
   Load address: 0x80060000                                                     
   Entry point:  0x80060000                                                     
                                                                                
Uncompressing kernel image... OK!                                               
Starting kernel...                                                              
                                                                                
[    0.000000] Linux version 3.18.45 (paul@F96NG92-L) (gcc version 4.8.3 (OpenW7
[    0.000000] bootconsole [early0] enabled                                     
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)                           
[    0.000000] SoC: Atheros AR9330 rev 1                                        
[    0.000000] Determined physical RAM map:                                     
[    0.000000]  memory: 04000000 @ 00000000 (usable)                            
[    0.000000] Initrd not found or empty - disabling initrd                     
[    0.000000] Zone ranges:                                                     
[    0.000000]   Normal   [mem 0x00000000-0x03ffffff]                           
[    0.000000] Movable zone start for each node                                 
[    0.000000] Early memory node ranges                                         
[    0.000000]   node   0: [mem 0x00000000-0x03ffffff]                          
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x03ffffff]                 
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.  
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32s
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pa6
[    0.000000] Kernel command line:  board=GL-AR150 mtdparts=spi0.0:256k(u-bootd
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)              
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)    
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)     
[    0.000000] Writing ErrCtl register=00000000                                 
[    0.000000] Readback ErrCtl register=00000000                                
[    0.000000] Memory: 60880K/65536K available (2533K kernel code, 149K rwdata,)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1       
[    0.000000] NR_IRQS:51                                                       
[    0.000000] Clocks: CPU:400.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:25.0z
[    0.000000] Calibrating delay loop... 265.42 BogoMIPS (lpj=1327104)          
[    0.080000] pid_max: default: 32768 minimum: 301                             
[    0.080000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)      
[    0.090000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes) 
[    0.100000] NET: Registered protocol family 16                               
[    0.100000] MIPS: machine is GL AR150                                        
[    0.590000] Switched to clocksource MIPS                                     
[    0.590000] NET: Registered protocol family 2                                
[    0.600000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)  
[    0.600000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)         
[    0.610000] TCP: Hash tables configured (established 1024 bind 1024)         
[    0.610000] TCP: reno registered                                             
[    0.620000] UDP hash table entries: 256 (order: 0, 4096 bytes)               
[    0.620000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)          
[    0.630000] NET: Registered protocol family 1                                
[    0.630000] futex hash table entries: 256 (order: -1, 3072 bytes)            
[    0.650000] squashfs: version 4.0 (2009/01/31) Phillip Lougher               
[    0.650000] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORI.
[... cut ...]
[   76.470000] [<801eb4d0>] ath79_spi_txrx_mode0+0x94/0x10c                     
[   76.470000] [<801ebaf8>] bitbang_txrx_8+0x78/0xbc                            
[   76.470000] [<801ebd70>] spi_bitbang_transfer_one+0x12c/0x21c                
[   76.470000] [<801e9e48>] __spi_pump_messages+0x404/0x464                     
[   76.470000] [<801ea584>] __spi_sync+0xfc/0x13c                               
[   76.470000] [<8301128c>] 0x8301128c                                          
...

and continues until the machine reboots spontaneously after about 300 seconds.

Paul.

Deoptim, 2017-08-25:
I see, thanks for the try.

For the first time I see this [in this link](https://github.com/servalproject/openwrt/commit/3133e6ed935bc4d7705f72e639519e39cafa711d) :

> // PGS 20170201 - Fictional CS line for 2nd SPI device
> // (used for microSD card). In reality, we are simply
> // inverting CS0 and feeding that to the microSD slot,
> // however, I don't know how to specify that here.
> #define TL_MR3X20_GPIO_CS1_MMC 0

However, how did someone change the polarity of CS line, if it is not changed in the "spi_board_info" section?

How to change the polarity of CS line: http://elinux.org/BeagleBoard/SPI#Chip_Select_polarity

Deoptim, 2017-08-25:
Although, in theory, it will work under the condition that there is an inverting circuit(CS1>for flash and CS1>(inverting circuit)>SD-card).
But here's the thing, how about the delay between transactions - they must be certain!

gardners, 2017-08-25:
Hello,

What we have is a NOT gate on the CS line to the internal flash, which is then used to feed the CS line to the MMC/microSD card.

I agree that there is a potential race condition here, where the delay of the NOT gate may cause a problem, however, given that the maximum SPI clock speed is ~30MHz, any problem here would persist for only a clock tick or so at most, and it is hard to see how this would cause any real problems.

Now, as for telling the kernel that it all it needs to do is deassert the CS line to the internal SPI flash, if you have suggestions as to how to better do that than what I have done, this would be very welcome.

Paul.

Deoptim, 2017-08-27:
> Now, as for telling the kernel that it all it needs to do is deassert the CS line to the internal SPI flash, if you have suggestions as to how to better do that than what I have done, this would be very welcome.

```
static struct ath79_spi_controller_data ath79_spi1_cdata = {
	.cs_type = ATH79_SPI_CS_TYPE_INTERNAL,
	.cs_line = 0,
	.is_flash = false,
};

static struct spi_board_info ath79_spi_info[] __initdata = {
	{
		.bus_num	= 0,
		.chip_select	= 1,
		.max_speed_hz	= 25000000,
		.modalias	= "mmc_spi",
		.platform_data	= &ath79_mmc_pdata,
		.controller_data = &ath79_spi1_cdata,
		.mode		= SPI_CS_HIGH,
	}
};
```

Works for me. Tested on Chaos Calmer Brunch/

closed 2022-02-09 by aparcar