openwrt/openwrt #497: build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:279:21: error: dereferencing pointer to incomplete type

CoppaMan, 2017-06-20:
Today I tried building the image for my Lamobo R1 with `make`, which failed in `build_dir/host/u-boot-2014.10/lib/rsa/rsa-sign.c:279:21`:
</br>
error: dereferencing pointer to incomplete type 'RSA {aka struct rsa_st}'
  if (BN_num_bits(key->e) > 64)
                     ^~
scripts/Makefile.host:134: recipe for target 'tools/lib/rsa/rsa-sign.o' failed
make[5]: *** [tools/lib/rsa/rsa-sign.o] Error 1
Makefile:1195: recipe for target 'tools-only' failed
make[4]: *** [tools-only] Error 2
make[4]: Leaving directory '/home/dario/openwrt/build_dir/host/u-boot-2014.10'
Makefile:46: recipe for target '/home/dario/openwrt/build_dir/host/u-boot-2014.10/.built' failed
make[3]: *** [/home/dario/openwrt/build_dir/host/u-boot-2014.10/.built] Error 2
make[3]: Leaving directory '/home/dario/openwrt/tools/mkimage'
tools/Makefile:134: recipe for target 'tools/mkimage/compile' failed
make[2]: *** [tools/mkimage/compile] Error 2
make[2]: Leaving directory '/home/dario/openwrt'
tools/Makefile:133: recipe for target '/home/dario/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/stamp/.tools_install_yynyynyyynyyyyyyynyyynyyyyyyyyyyyyyyynnyynynnyyynnnyy' failed
make[1]: *** [/home/dario/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/stamp/.tools_install_yynyynyyynyyyyyyynyyynyyyyyyyyyyyyyyynnyynynnyyynnnyy] Error 2
make[1]: Leaving directory '/home/dario/openwrt'
/home/dario/openwrt/include/toplevel.mk:192: die Regel für Ziel „world“ scheiterte
make: *** [world] Fehler 2
<br/>
For compatibility reasons I'm building on a fresh install of Debian 9 in Virtualbox where all required packages (according to `make menuconfig`) are installed.

CoppaMan, 2017-06-22:
The development package of openssl, libssl-dev, was missing because there was no check for it in the make file. Closing

closed 2017-06-22 by CoppaMan

ThRichard, 2017-08-22:
I'm building on a Debian 9 and I get the same issue.

rsa-sign.c:279:21: error: dereferencing pointer to incomplete type 'RSA {aka struct rsa_st}'
  if (BN_num_bits(key->e) > 64)
                     ^~
scripts/Makefile.host:134: recipe for target 'tools/lib/rsa/rsa-sign.o' failed

I checked, the package libssl-dev is present

N-Mi, 2017-09-21:
Same problem here, and libssl-dev is installed :
```
$ LANG=C dpkg -l libssl-dev
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                                          Version                             Architecture                        Description
+++-=============================================================-===================================-===================================-================================================================================================================================
ii  libssl-dev:amd64                                              1.1.0f-3                            amd64                               Secure Sockets Layer toolkit - development files
```

smortex, 2017-11-11:
See #394 regarding building with recent OpenSSL.