openwrt/openwrt #500: as: unrecognized option '-EB'

liaoyangyang, 2017-06-27:
when i compile openwrt in the default conf， I got the following error， hope someone can help me
I use make -j1 V=s to see error clearly, hopping the following info can be used

+    gcc  version 5.4.0
+    platform ubuntu16.04 64bit
+    kernel 4.10.10

```bash
$ make -j1 V=s
.........
make[5]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/build-x86_64-linux-gnu/libcpp'
make[5]: Entering directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=x86_64-linux-gnu-gcc" "CFLAGS=-O2 -I/work/liao/workspace/temp/openwrt/staging_dir/host/include -I/work/liao/workspace/temp/openwrt/staging_dir/host/usr/inclu
de " "CXXFLAGS=-g -O2" "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable 
-Wno-error=unused-result -msoft-float" "INSTALL=/usr/bin/install -c" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "INSTALL_SCRIPT=/usr/bin/install -c" "LDF
LAGS=" "LIBCFLAGS=-O2 -I/work/liao/workspace/temp/openwrt/staging_dir/host/include -I/work/liao/workspace/temp/openwrt/staging_dir/host/usr/include " "LIBCFLAGS_FOR_TARGET=-Os -pipe -mno-bra
nch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float" "MAKE=make" "MAKEINFO=/work/liao/workspac
e/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0/missing makeinfo --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTE
ST=runtest" "RUNTESTFLAGS=" "exec_prefix=/work/liao/workspace/temp/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16" "infodir=/work/liao/workspace/temp/openwrt/staging_dir/toolc
hain-mips_34kc_gcc-5.3.0_musl-1.1.16/share/info" "libdir=/work/liao/workspace/temp/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/lib" "prefix=/work/liao/workspace/temp/openwr
t/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16" "tooldir=/work/liao/workspace/temp/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/mips-openwrt-linux-musl" "AR=x86_64-
linux-gnu-ar" "AS=x86_64-linux-gnu-as" "CC=x86_64-linux-gnu-gcc" "CXX=x86_64-linux-gnu-g++" "LD=x86_64-linux-gnu-ld" "LIBCFLAGS=-O2 -I/work/liao/workspace/temp/openwrt/staging_dir/host/inclu
de -I/work/liao/workspace/temp/openwrt/staging_dir/host/usr/include " "NM=x86_64-linux-gnu-nm" "PICFLAG=" "RANLIB=x86_64-linux-gnu-ranlib" "DESTDIR=" DO=all multi-do # make
make[5]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/zlib'
make[5]: Entering directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/libbacktrace'
make  all-am
make[6]: Entering directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/libbacktrace'
true  DO=all multi-do # make
make[6]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/libbacktrace'
make[5]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/libbacktrace'
make[5]: Entering directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[5]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/libcpp'
make[5]: Entering directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/libdecnumber'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/libdecnumber'
make[5]: Entering directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/fixincludes'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/fixincludes'
make[5]: Entering directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/gcc'
x86_64-linux-gnu-g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-f
ormat-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/work/liao/workspace/temp/openwrt/build_dir/toolchain-m
ips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0/gcc -I/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0/gcc/. -I/work/liao/workspace/temp/openwrt/build
_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0/gcc/../include -I/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0/gcc/../libcpp/includ
e -I/work/liao/workspace/temp/openwrt/staging_dir/host/include -I/work/liao/workspace/temp/openwrt/staging_dir/host/include -I/work/liao/workspace/temp/openwrt/staging_dir/host/include  -I/w
ork/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0/gcc/../libdecnumber -I/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_
musl-1.1.16/gcc-5.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0/gcc/../libbacktrace   -o ins
n-automata.o -MT insn-automata.o -MMD -MP -MF ./.deps/insn-automata.TPo insn-automata.c
as: unrecognized option '-EB'
Makefile:1065: recipe for target 'insn-automata.o' failed
make[5]: *** [insn-automata.o] Error 1
make[5]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/gcc'
Makefile:4101: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal'
Makefile:46: recipe for target '/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/.built' failed
make[3]: *** [/work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/.built] Error 2
make[3]: Leaving directory '/work/liao/workspace/temp/openwrt/toolchain/gcc/minimal'
toolchain/Makefile:81: recipe for target 'toolchain/gcc/minimal/compile' failed
make[2]: *** [toolchain/gcc/minimal/compile] Error 2
make[2]: Leaving directory '/work/liao/workspace/temp/openwrt'
toolchain/Makefile:80: recipe for target '/work/liao/workspace/temp/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/stamp/.toolchain_install' failed
make[1]: *** [/work/liao/workspace/temp/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/stamp/.toolchain_install] Error 2
make[1]: Leaving directory '/work/liao/workspace/temp/openwrt'
/work/liao/workspace/temp/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2
```

hubenchang0515, 2017-09-16:
I meet the same problem;

liaoyangyang, 2017-09-23:
@hubenchang0515 看你是也是中国的，我就直接用汉语了，囧！！！

我这边已经解决这个错误了，这几天正好需要编译固件，潜心搞了一下，现在描述一下，不知道情况是否一致

英文不好，如果有爱心人事可以帮忙翻译下

原因出在gcc上面，我的电脑的gcc会默认读取当前目录的specs，而出错的文件夹下有一个specs，导致编译时会默认读取这个配置，这个配置应该是为mips编译器准备的。

验证如下

1.进入一个没有specs文件的文件，执行gcc -v会出现Using built-in specs.

```bash
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.4' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 
```
2.进入出错的文件执行

```bash
gcc -v
Reading specs from ./specs
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.4' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc driver version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) executing gcc version 5.3.0
```

可以看出读取当前的文件夹的specs，并且你在这个文件夹创建一个简单的c程序，也会报同样的错误

目前还没有查到怎么关闭gcc的这种行为，不过我用笨办法解决了，如果你们查到了，请告诉我，解决办法如下：

```bash
cd /work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/gcc
# 备份specs
cp specs specs_bak
#生成系统目前的specs
x86_64-linux-gnu-g++ -dumpspecs > specs
```

继续执行编译，你会发现可以继续编译了，但是引入了新的错误，大概是找不到各种头文件

```bash
checking size of double... 8
checking size of long double... 8
checking for inttypes.h... no
checking for stdint.h... no
checking for stdlib.h... no
checking for ftw.h... no
checking for unistd.h... no
checking for sys/stat.h... no
checking for sys/types.h... no
checking for string.h... no
checking for strings.h... no
checking for memory.h... no
checking for grep that handles long lines and -e... xxx/openwrt/staging_dir/host/bin/grep
checking for egrep... xxx/openwrt/staging_dir/host/bin/grep -E
checking for ANSI C header files... no
checking whether decimal floating point is supported... no
checking whether fixed-point is supported... (cached) no
checking whether to use setjmp/longjmp exceptions... unknown
configure: error: unable to detect exception model
Makefile:10906: recipe for target 'configure-target-libgcc' failed
make[4]: *** [configure-target-libgcc] Error 1

```

(猜测)这是因为修改了specs文件，导致mips编译器编译时查询了错误的环境变量，这时候切回去就好了

```bash
cd  /work/liao/workspace/temp/openwrt/build_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/gcc-5.3.0-minimal/gcc
cp specs_bak specs
```
然后继续执行，又能编译了。

提示，大概会出现3次这种错误，分别是编译`gcc-5.3.0-minimal`，`gcc-5.3.0-initial`，`gcc-5.3.0-final`，每次进入出错的文件夹，执行上述操作即可，然后就一路畅通了

closed 2022-02-09 by aparcar