openwrt/openwrt #507: Compiling tools/automake failed

SEVENID, 2017-07-20:
```
make[3]: Entering directory '/home/user/openwrt/tools/automake'
(cd /home/user/openwrt/build_dir/host/automake-1.15; AUTOM4TE=/home/user/openwrt/staging_dir/host/bin/autom4te AUTOCONF=/home/user/openwrt/staging_dir/host/bin/autoconf AUTOMAKE=/home/user/openwrt/staging_dir/host/bin/automake ACLOCAL=/home/user/openwrt/staging_dir/host/bin/aclocal AUTOHEADER=/home/user/openwrt/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/user/openwrt/staging_dir/host/bin/libtoolize LIBTOOL=/home/user/openwrt/staging_dir/host/bin/libtool M4=/home/user/openwrt/staging_dir/host/bin/m4 AUTOPOINT=true STAGING_DIR="" ./bootstrap.sh)
Unescaped left brace in regex is illegal here in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at ./bin/automake.tmp line 3938.
make[3]: *** [Makefile:51: /home/user/openwrt/build_dir/host/automake-1.15/.configured] Error 255
make[3]: Leaving directory '/home/user/openwrt/tools/automake'
make[2]: *** [tools/Makefile:135: tools/automake/compile] Error 2
make[2]: Leaving directory '/home/user/openwrt'
make[1]: *** [tools/Makefile:133: /home/user/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/stamp/.tools_install_yynyynyyynyyyyyyynyyynyyyyyyyyyyyyyyynnyynynnyyynnnyy] Error 2
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:194: world] Error 2
```
Target: Allwinner A1x/A20/A3x
Profile: Cubieboard2
(builer) Make 4.2.1, Automake 1.15.1

DocMAX, 2017-07-29:
updates?

xcellerator, 2017-08-01:
Can confirm same issue here too.

laenco, 2017-08-09:
https://git.lede-project.org/?p=source.git;a=commit;h=92c80f38cff3c20388f9ac13d5196f2745aeaf77

kfl62, 2017-08-28:
After applying path  I have new error in tools/mkimage
https://pastebin.com/LAmqcVyL

GitCT, 2017-09-10:
sudo apt-get install libssl1.0-dev ;-)

HugoPoi, 2017-10-09:
This issue occur when using perl >= 5.22 for building. Currently automake has been fixed but openwrt use an older version 1.15 not the 1.15.1.

Fast fix
copy the patch https://git.lede-project.org/?p=source.git;a=blob_plain;f=tools/automake/patches/010-automake-port-to-Perl-5.22-and-later.patch;h=31b9273d547145e5ecbeaef20a1e82cc9292fdc2;hb=92c80f38cff3c20388f9ac13d5196f2745aeaf77 directly in `tools/automake/patches`

The proper way is to upgrade automake in openwrt with 1.15.1 but we need to rewrite all patches !

HugoPoi, 2017-10-21:
This issue occur when using perl >= 5.22 for building. Currently automake has been fixed but openwrt use an older version 1.15 not the 1.15.1.

Fast fix
copy the patch https://git.lede-project.org/?p=source.git;a=blob_plain;f=tools/automake/patches/010-automake-port-to-Perl-5.22-and-later.patch;h=31b9273d547145e5ecbeaef20a1e82cc9292fdc2;hb=92c80f38cff3c20388f9ac13d5196f2745aeaf77 directly in `tools/automake/patches`

The proper way is to upgrade automake in openwrt with 1.15.1 but we need to rewrite all patches !
I encounter many other issue for trying to build under archlinux, is better to build on tested system like debian.

Iolop, 2017-12-14:
The same.
Follow HugoPoi 's advice ,and it do work

closed 2022-02-09 by aparcar