openwrt/openwrt #509: error: dereferencing pointer to incomplete type

DocMAX, 2017-07-23:
```
make[6]: Entering directory '/mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/host/cmake-3.6.1'
[ 10%] Building C object Utilities/cmcurl/lib/CMakeFiles/cmcurl.dir/vtls/openssl.c.o
/mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c: In function 'get_cert_chain':
/mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c:2482:13: error: dereferencing pointer to incomplete type 'X509 {aka struct x509_st}'
     cinf = x->cert_info;
             ^~
/mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c:2484:30: error: dereferencing pointer to incomplete type 'X509_CINF {aka struct x509_cinf_st}'
     j = asn1_object_dump(cinf->signature->algorithm, bufp, CERTBUFFERSIZE);
                              ^~
/mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/host/cmake-3.6.1/Utilities/cmcurl/lib/vtls/openssl.c:2510:20: error: dereferencing pointer to incomplete type 'EVP_PKEY {aka struct evp_pkey_st '
       switch(pubkey->type) {
                    ^~

```

kfl62, 2017-08-28:
Same problem. Running Archlinux

xcellerator, 2017-11-25:
Can confirm. Same problem running Arch Linux too. Any updates/patches to fix this?

closed 2022-02-09 by aparcar