openwrt/openwrt #518: VFS: Unable to mount root fs on MX25L25635E

ReleaseBug, 2017-08-08:
Hi，

I am working on a Macronix MX25L25635E (32MB) flash chip platform.
And error occured at kernel booting :"Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)"

I have noticed issue #430 has the same flash chip and the same error.

Is it a flash driver problem ?  Is there a solution ? Thx~

===========SOME  INFO ===================
Here is my dts:
&spi0 {
	status = "okay";

	m25p80@0 {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "jedec,spi-nor";
		reg = <0>;
		linux,modalias = "m25p80", "mx25l25635e";
		spi-max-frequency = <10000000>;

		partition@0 {
			label = "u-boot";
			reg = <0x0 0x30000>;
			read-only;
		};

		partition@30000 {
			label = "u-boot-env";
			reg = <0x30000 0x10000>;
			read-only;
		};

		factory: partition@40000 {
			label = "factory";
			reg = <0x40000 0x10000>;
			read-only;
		};

		partition@50000 {
			label = "firmware";
			reg = <0x50000 0x1fb0000>;
		};
	};
};

Here is my boot log, **it seems the root filesystem is not found in the "firmware" mtd section**:

Boot and Recovery Environment for Embedded Devices
DRAM: 128MB
Platform: MediaTek MT7620A ver 2, eco 6
Board: Youku YK1
Clocks: CPU: 580MHz, Bus: 193MHz
Flash: Macronix MX25L25635E (32MB) on rt2880-spi
rt2880-eth: Using MAC address 54:36:9b:08:4e:86
eth0: MediaTek MT7620A built-in 5-port 10/100M switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 0   

Trying to boot firmware from 0x00050000 in flash bank 0 ...
Reading data into memory ...
U-Boot firmware image header detected.
    Image Name:   MIPS OpenWrt Linux-4.4.14
    Data Size:    1289415 Bytes
    Load Address: 80000000
    Entry Point:  80000000
Uncompressing data (LZMA) ... done.
Flushing cache ... done.

Starting kernel at 0x80000000...

[    0.000000] Linux version 4.4.14 (ubuntu@ubuntu-virtual-machine) (gcc version 5.3.0 (OpenWrt GCC 5.3.0 unknown) ) #2 Mon Jul 31 18:08:58 UTC 2017
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is YOUKU YK1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00076e9b
[    0.000000] Readback ErrCtl register=00076e9b
[    0.000000] Memory: 125748K/131072K available (2912K kernel code, 140K rwdata, 700K rodata, 144K init, 195K bss, 5324K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:256
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
[    0.000000] systick: enable autosleep mode
[    0.000000] systick: running - mult: 214748, shift: 32
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
[    0.000011] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
[    0.015451] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.080607] pid_max: default: 32768 minimum: 301
[    0.089933] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.102953] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.123136] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.142788] pinctrl core: initialized pinctrl subsystem
[    0.153752] NET: Registered protocol family 16
[    0.184244] rt2880_gpio 10000600.gpio: registering 24 gpios
[    0.195268] rt2880_gpio 10000600.gpio: registering 24 irq handlers
[    0.207696] rt2880_gpio 10000638.gpio: registering 16 gpios
[    0.218679] rt2880_gpio 10000638.gpio: registering 16 irq handlers
[    0.231118] rt2880_gpio 10000660.gpio: registering 32 gpios
[    0.242101] rt2880_gpio 10000660.gpio: registering 32 irq handlers
[    0.254524] rt2880_gpio 10000688.gpio: registering 1 gpios
[    0.265329] rt2880_gpio 10000688.gpio: registering 1 irq handlers
[    0.279019] clocksource: Switched to clocksource systick
[    0.291003] NET: Registered protocol family 2
[    0.300463] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.314234] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.326827] TCP: Hash tables configured (established 1024 bind 1024)
[    0.339561] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.351082] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.363814] NET: Registered protocol family 1
[    0.372903] rt-timer 10000100.timer: maximum frequency is 1220Hz
[    0.385793] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.417119] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.428646] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.450918] io scheduler noop registered
[    0.458573] io scheduler deadline registered (default)
[    0.469130] ralink-usb-phy usbphy: invalid resource
[    0.479424] Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled
[    0.493286] console [ttyS0] disabled
[    0.500334] 10000c00.uartlite: ttyS0 at MMIO 0x10000c00 (irq = 20, base_baud = 2500000) is a Palmchip BK-3103
[    0.520005] console [ttyS0] enabled
[    0.520005] console [ttyS0] enabled
[    0.533817] bootconsole [early0] disabled
[    0.533817] bootconsole [early0] disabled
[    0.554719] spi spi0.0: force spi mode3
[    0.562962] m25p80 spi0.0: mx25l25635e (32768 Kbytes)
[    0.573096] 4 ofpart partitions found on MTD device spi0.0
[    0.584028] Creating 4 MTD partitions on "spi0.0":
[    0.593580] 0x000000000000-0x000000030000 : "u-boot"
[    0.605301] 0x000000030000-0x000000040000 : "u-boot-env"
[    0.617807] 0x000000040000-0x000000050000 : "factory"
[    0.629865] 0x000000050000-0x000002000000 : "firmware"
[    0.897419] rt2880-pinmux pinctrl: pin io40 already requested by pinctrl; cannot claim for 10100000.ethernet
[    0.917066] rt2880-pinmux pinctrl: pin-40 (10100000.ethernet) status -22
[    0.930435] rt2880-pinmux pinctrl: could not request pin 40 (io40) from group ephy  on device rt2880-pinmux
[    0.949857] mtk_soc_eth 10100000.ethernet: Error applying setting, reverse things back
[    0.968131] gsw: setting port4 to ephy mode
[    0.976728] mtk_soc_eth 10100000.ethernet: loaded mt7620 driver
[    0.989387] mtk_soc_eth 10100000.ethernet eth0: mediatek frame engine at 0xb0100000, irq 5
[    1.006455] rt2880_wdt 10000120.watchdog: Initialized
[    1.018170] NET: Registered protocol family 10
[    1.030841] NET: Registered protocol family 17
[    1.039861] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[    1.065029] 8021q: 802.1Q VLAN Support v1.8
[    1.076569] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    1.091545] Please append a correct "root=" boot option; here are the available partitions:
[    1.108198] 1f00             192 mtdblock0  (driver?)
[    1.118273] 1f01              64 mtdblock1  (driver?)
[    1.128348] 1f02              64 mtdblock2  (driver?)
[    1.138421] 1f03           32448 mtdblock3  (driver?)
[    1.148495] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    1.167417] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

gsj0791, 2017-09-11:
update your version to latest or remove patch file target/linux/generic/patchs-*/*m25p80-mx-disable-software-protection.patch
enjoy！

ReleaseBug, 2017-09-17:
Thx for your help!

And bad news: my board  is just broken, it can't be used.
Because，i replaced its 128MB RAM chip to 256MB.  Some bad thing happened，it's just broken.

if some one have tested this fix method, please tell me.

zhejunli, 2017-12-08:
te

zhejunli, 2017-12-08:
I have seen the Macronix MX25L25635E doesn't not support a full command set.  In the official website, it is not recommended for new design. I found 35F provides a full flash CMD set compared with 35E.

Most flash chips >16MB supports 4BAddr commands, which the 35E, eventhough it is 32MB, doesn't support. This may cause a problem in m25p80.c to driver it properly, especially to the >16MB region.

Hope this helps.

closed 2022-02-09 by aparcar