openwrt/openwrt #519: opkg segfaults

Dlay2017, 2017-08-09:
On this machine: 

Model | TP-Link TL-WR1043N/ND v4
Firmware Version | OpenWrt Designated Driver 50108 / LuCI Master (git-17.214.30899-bf407f0)
Kernel Version | 4.4.14

opkg segfaults

root@LocalNet:~# opkg list
Segmentation fault

opkg update works

The kernel.log says:

[ 1836.680761] do_page_fault(): sending SIGSEGV to opkg for invalid read access from 00000000
[ 1836.689380] epc = 7748e9a0 in libc.so[7741c000+91000]
[ 1836.694798] ra  = 0040a01f in opkg[400000+18000]

Any suggestions ?

rootninja, 2017-08-31:
Model	Linksys E4200v2 / EA4500
Firmware Version	OpenWrt Designated Driver 50108 / LuCI Master (git-17.234.69785-72b90c0)
Kernel Version	4.4.14

openwrt# opkg list |grep luci-app-sqm
Segmentation fault

Luci simply does not provide any results when searching. This also includes clicking on the letters to see available upgrades/updates available both of which display nothing below. Interestingly the logs do not display any info when the Segmentation fault is returned however in Luci I receive the following in the logs: 

Thu Aug 31 02:08:58 2017 daemon.err uhttpd[1966]: Segmentation fault
Thu Aug 31 02:09:03 2017 daemon.err uhttpd[1966]: Segmentation fault

I was previously running OpenWrt Designated Driver 50107 / LuCI Master (git-17.165.70928-dd6cb31) which also behaved the same prompting me to attempt to upgrade

I am however able to install the package if I know the exact name of every package required ie 

openwrt# opkg install luci-app-sqm

korakinos, 2017-09-05:
Have you tried 'opkg upgrade opkg'? I believe I had the same problem and was able to resolve it this way.

Dlay2017, 2017-09-06:
for sure i tried that.. no effect.. :-(

petzi4, 2017-09-07:
Same Problem here with Model
TP-Link Archer C7 v2.0
on  DESIGNATED DRIVER (Bleeding Edge, 50108)

daper, 2017-09-14:
Hi, I've been having the same issue with TP-Link TL-WR1043N/ND v4. And I've also traced the error to this repository "http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages" of /etc/opkg/distfeeds.conf. You can avoid getting segfault error by disabling the repo... but you'll miss all packages there... still investigating

Dlay2017, 2017-09-14:
yes i can confirm that too, when you comment out the line in distfeeds.conf you can use opkg normally..

rootninja, 2017-09-19:
confirmed here as well by disabling repo

mach327, 2017-11-18:
I narrowed this down to python3-idna. 
Specifically, it looks like the Description field:
```
Description:  
 .
 (Variant for Python3)
```
Is invalid. Entering some text on the same line as the field header `Description:` avoided the segfault in opkg when processing the package lists.

This gets set by `feeds/packages/lang/python/python-idna/Makefile`. 
It's a typo where `$(call define ...)` was used instead of `$(call ...)`, which leaves a blank line in description instead of including the parent package's description, leading opkg to poorly parse the package list and segfault.
```
define Package/python3-idna/description
$(call define Package/python-idna/description)
.
(Variant for Python3)
endef
```

I'll send the packages repo a pull request momentarily to fix it.

closed 2017-12-05 by jow-