openwrt/openwrt #523: OpenWRT support for Mikrotik RB493GAH

AlverGant, 2017-08-22:
Hi,

I'm not sure if RB493GAH is supported because I am experiencing difficulties regarding NAND flash partitioning so I'm asking it here.

I am able to TFTP boot a Chaos Calmer compiled initramfs.elf binary but when I try to erase kernel and rootfs partitions, thousands of bad erase block appears leaving no room to permanent installation.
My flash chip is Samsumg K9F1208U0C which is a 64MB 8bit device, different from the most common 128MB versions.

Here is my proc/mtd parts which I do think is wrong
dev:    size   erasesize  name
mtd0: 0000b000 00001000 "routerboot"
mtd1: 00001000 00001000 "hard_config"
mtd2: 00002000 00001000 "bios"
mtd3: 00001000 00001000 "soft_config"
mtd4: 00040000 00004000 "booter"
mtd5: 003c0000 00004000 "kernel"
mtd6: 03c00000 00004000 "rootfs"

Unfortunately, now I don't have space available even to do a rollback to RouterOS via netinstall, leaving a significantly powerfull device almost useless.

If somebody can give me a hint that would be awesome.

Bellow the console output during TFTP boot

RouterBOOT booter 3.24

RouterBoard 493G

CPU frequency: 680 MHz
  Memory size: 256 MiB
    NAND size:  64 MiB

Press any key within 2 seconds to enter setup..
trying dhcp protocol... OK
resolved mac address EC:08:6B:EC:3F:A2
Gateway: 10.61.34.254
transfer started .............................................. transfer ok, time=2.75s

loading kernel from nand... kernel not found
trying dhcp protocol... OK
resolved mac address EC:08:6B:EC:3F:A2
Gateway: 10.61.34.254
transfer started .............................................. transfer ok, time=2.74s
setting up elf image... OK
jumping to kernel code
[    0.000000] Linux version 3.18.45 (lede@lede) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r49389) ) #4 Sun Aug 13 23:50:06 BRT 2017
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Atheros AR7161 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x0fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x0fffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x0fffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line:  console=ttyS0,115200 parts=1 boot_part_size=4194304 gpio=4023 HZ=340000000 mem=256M kmac=D4:CA:6D:34:0A:29 board=493G ver=3.24 hw_opt=00000000 boot=1 mlc=9 rootfstype=yaffs noinitrd
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 253312K/262144K available (2436K kernel code, 153K rwdata, 516K rodata, 2836K init, 184K bss, 8832K reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[... 120 lines cut ...]
[   41.100000] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   41.110000] br-lan: port 2(eth1.1) entered forwarding state
[   41.120000] br-lan: port 2(eth1.1) entered forwarding state
[   41.120000] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.2: link becomes ready
[   43.080000] Atheros AR8216/AR8236/AR8316 ag71xx-mdio.0:00: Port 5 is up
[   43.090000] br-lan: port 1(eth0) entered forwarding state
[   43.120000] br-lan: port 2(eth1.1) entered forwarding state
[   80.370000] random: nonblocking pool is initialized

BusyBox v1.23.2 (2017-08-02 15:12:00 BRT) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 CHAOS CALMER (Chaos Calmer, r49389)
 -----------------------------------------------------
  * 1 1/2 oz Gin            Shake with a glassful
  * 1/4 oz Triple Sec       of broken ice and pour
  * 3/4 oz Lime Juice       unstrained into a goblet.
  * 1 1/2 oz Orange Juice
  * 1 tsp. Grenadine Syrup
 -----------------------------------------------------

AlverGant, 2017-08-25:
For the record, I was able to revert to RouterOS with the following process:
Recompile Chaos Calmer with a modified MTD tool that allowed me to forcelly rewrite all NAND blocks including false marked bad blocks (mtd erase -N /dev/mtd6). That alone freed some MBytes just the necessary to do a RouterOS Netinstall.
Once RouterOS got installed it imediately reclaim all NAND storage with 0.1% bad blocks.
Subsequently I was able to finally install OpenWRT through a TFTP and finally wget2nand.
This time without the micro-sd card installed, I had no problems

betelgeux1, 2017-09-27:
Were you able to get the ETH0 switch routing?

closed 2022-02-09 by aparcar